jblance/mpp-solar #412: json_mqtt output

sante85, 2023-10-19:
Hi @rossandrews hi have this config in attachment and the output is this

ID/status/serial_number/value 92932208100911
mppsolar {"ac_input_voltage": 223.0, "ac_input_frequency": 50.0, "ac_output_voltage": 223.0, "ac_output_frequency": 50.0, "ac_output_apparent_power": 0, "ac_output_active_power": 0, "ac_output_load": 0, "bus_voltage": 351, "battery_voltage": 0.0, "battery_charging_current": 0, "battery_capacity": 0, "inverter_heat_sink_temperature": 39, "pv1_input_current": 0.0, "pv1_input_voltage": 0.0, "battery_voltage_from_scc": 0.0, "battery_discharge_current": 0, "is_sbu_priority_version_added": 0, "is_configuration_changed": 0, "is_scc_firmware_updated": 0, "is_load_on": 1, "is_battery_voltage_to_steady_while_charging": 0, "is_charging_on": 0, "is_scc_charging_on": 0, "is_ac_charging_on": 0, "battery_voltage_offset_for_fans_on": 0, "eeprom_version": 0, "pv1_charging_power": 0, "is_charging_to_float": 0, "is_switched_on": 1, "is_dustproof_installed": 0, "time_until_the_end_of_absorb_charging": 1, "time_until_the_end_of_float_charging": 15, "scc_flag": "SCC not communicating?", "allowscconflag": "00", "chargeaveragecurrent": "00", "scc_pwm_temperature": 31, "inverter_temperature": 35, "battery_temperature": 39, "transformer_temperature": 32, "gpio13": 2, "fan_lock_status": "Not locked", "not_used": "000", "fan_pwm_speed": 30, "scc_charge_power": 0, "parallel_warning": "0000", "sync_frequency": 0.0, "inverter_charge_status": "nocharging", "unknown_value_in_response_18": "b'0'", "unknown_value_in_response_19": "b'060'", "unknown_value_in_response_20": "b'030'", "unknown_value_in_response_21": "b'060'", "unknown_value_in_response_22": "b'030'", "unknown_value_in_response_23": "b'58.40'", "unknown_value_in_response_24": "b'000'", "unknown_value_in_response_25": "b'120'", "unknown_value_in_response_26": "b'0'", "unknown_value_in_response_27": "b'0000'"}
**mppsolar** {"pv2_input_current": 0.0, "pv2_input_voltage": 0.0, "pv2_charging_power": 0}
**mppsolar** {"ac_input_voltage": 230.0, "ac_input_current": 34.7, "ac_output_voltage": 230.0, "ac_output_frequency": 50.0, "ac_output_current": 34.7, "ac_output_apparent_power": 8000, "ac_output_active_power": 8000, "battery_voltage": 48.0, "battery_recharge_voltage": 46.0, "battery_under_voltage": 44.0, "battery_bulk_charge_voltage": 56.4, "battery_float_charge_voltage": 54.0, "battery_type": "User", "max_ac_charging_current": 2, "max_charging_current": 60, "input_voltage_range": "Appliance", "output_source_priority": "Utility Solar Battery", "charger_source_priority": "Solar + Utility", "max_parallel_units": 9, "machine_type": "Off Grid", "topology": "transformerless", "output_mode": "single machine output", "battery_redischarge_voltage": 54.0, "pv_ok_condition": "As long as one unit of inverters has connect PV, parallel system will consider PV OK", "pv_power_balance": "PV input max power will be the sum of the max charged power and loads power", "max_charging_time_for_cv_stage": 480, "operation_logic": "Automatic mode", "max_discharging_current": 0, "buzzer": "enabled", "lcd_reset_to_default": "enabled", "lcd_backlight": "enabled", "primary_source_interrupt_alarm": "enabled", "record_fault_code": "enabled", "overload_bypass": "disabled", "unknown_j": "disabled", "overload_restart": "disabled", "over_temperature_restart": "disabled"}

Why mqtt topic exposed is mppsolar and not that i have configured in tag ?
Eg.

[92932208100911_config]
port = /dev/ttyUSB0
protocol = PI30
command = get_settings
[mppsolar.conf.txt](https://github.com/jblance/mpp-solar/files/13039610/mppsolar.conf.txt)

**tag = 92932208100911/config**
outputs = json_mqtt

jblance, 2023-10-19:
try running with debug to get more info on what is happening
looks to me the tags with `/` arent working correctly

sante85, 2023-10-20:
With debug
mppsolar -p /dev/ttyUSB0 -c QPIGS -T tag_test/ddd -D -o json_mqtt

2023-10-20 12:56:41,858:INFO:__init__:main@215: Solar Device Command Utility, version: 0.16.10
2023-10-20 12:56:41,859:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2023-10-20 12:56:41,859:DEBUG:__init__:main@250: MqttBroker name: localhost, port: 1883, user: None
2023-10-20 12:56:41,860:DEBUG:__init__:main@252: udp port 5555
2023-10-20 12:56:41,860:DEBUG:__init__:main@254: Using Postgres None
2023-10-20 12:56:41,860:DEBUG:__init__:main@257: Using Mongo None with mppsolar
2023-10-20 12:56:41,861:INFO:__init__:main@358: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
2023-10-20 12:56:41,867:DEBUG:__init__:main@362: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2023-10-20 12:56:41,867:DEBUG:device:__init__@34: __init__ args ()
2023-10-20 12:56:41,868:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x75fa99d0>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-10-20 12:56:41,868:DEBUG:__init__:get_port_type@69: port matches ttyusb
2023-10-20 12:56:41,868:INFO:__init__:get_port@111: Using serialio for communications
2023-10-20 12:56:41,882:DEBUG:__init__:get_protocol@13: Protocol PI30
2023-10-20 12:56:42,574:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0x75fa9710>, protocol PI30 protocol handler
2023-10-20 12:56:42,574:DEBUG:__init__:main@407: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x761ae9b0>, 'QPIGS', 'tag_test/ddd', 'json_mqtt', None, None)]
2023-10-20 12:56:42,575:INFO:__init__:main@412: Looping 1 commands
2023-10-20 12:56:42,575:INFO:__init__:main@422: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x75fa9710>, protocol: PI30 protocol handler for command: QPIGS, tag: tag_test/ddd, outputs: json_mqtt
2023-10-20 12:56:42,575:INFO:device:run_command@51: Running command QPIGS
2023-10-20 12:56:42,575:INFO:abstractprotocol:get_full_command@61: Using protocol b'PI30' with 44 commands
2023-10-20 12:56:42,576:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'QPIGS'
2023-10-20 12:56:42,576:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0xb7 0xa9 0xb7a9
2023-10-20 12:56:42,576:DEBUG:abstractprotocol:get_full_command@67: full command: b'QPIGS\xb7\xa9\r'
2023-10-20 12:56:42,577:INFO:device:run_command@81: full command b'QPIGS\xb7\xa9\r' for command QPIGS
2023-10-20 12:56:42,577:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'QPIGS'
2023-10-20 12:56:42,577:DEBUG:abstractprotocol:get_command_defn@73: Found command QPIGS in protocol b'PI30'
2023-10-20 12:56:42,577:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 2400
2023-10-20 12:56:42,580:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2023-10-20 12:56:44,076:DEBUG:serialio:send_and_receive@30: serial response was: b'(222.5 50.0 222.5 50.0 0000 0000 000 350 00.00 000 000 0039 \x82r\x82\x02\x82\x82\x82r\x82\x02\x82\x82r\x82\x82\x02\x82\x82\x82\x82\x82\x02\x82\x82\x82\x8a\x82\x82\x82\x82\x02\x82\x82\x02\x82\x82\x02\x82\x82\x82\x82\x82\x02\x82\x8a\x82\xea\xfbj'
2023-10-20 12:56:44,078:DEBUG:device:run_command@98: Send and Receive Response b'(222.5 50.0 222.5 50.0 0000 0000 000 350 00.00 000 000 0039 \x82r\x82\x02\x82\x82\x82r\x82\x02\x82\x82r\x82\x82\x02\x82\x82\x82\x82\x82\x02\x82\x82\x82\x8a\x82\x82\x82\x82\x02\x82\x82\x02\x82\x82\x02\x82\x82\x82\x82\x82\x02\x82\x8a\x82\xea\xfbj'
2023-10-20 12:56:44,078:INFO:abstractprotocol:decode@246: response passed to decode: b'(222.5 50.0 222.5 50.0 0000 0000 000 350 00.00 000 000 0039 \x82r\x82\x02\x82\x82\x82r\x82\x02\x82\x82r\x82\x82\x02\x82\x82\x82\x82\x82\x02\x82\x82\x82\x8a\x82\x82\x82\x82\x02\x82\x82\x02\x82\x82\x02\x82\x82\x82\x82\x82\x02\x82\x8a\x82\xea\xfbj'
2023-10-20 12:56:44,078:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'QPIGS'
2023-10-20 12:56:44,079:DEBUG:abstractprotocol:get_command_defn@73: Found command QPIGS in protocol b'PI30'
2023-10-20 12:56:44,079:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'(222.5 50.0 222.5 50.0 0000 0000 000 350 00.00 000 000 0039 \x82r\x82\x02\x82\x82\x82r\x82\x02\x82\x82r\x82\x82\x02\x82\x82\x82\x82\x82\x02\x82\x82\x82\x8a\x82\x82\x82\x82\x02\x82\x82\x02\x82\x82\x02\x82\x82\x82\x82\x82\x02\x82\x8a\x82'
2023-10-20 12:56:44,080:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x26 0xe9 0x26e9
2023-10-20 12:56:44,081:INFO:abstractprotocol:decode@261: validity check fail: {'validity check': ['Error: Invalid response CRCs', '']}
2023-10-20 12:56:44,081:INFO:device:run_command@116: Decoded response {'_command': 'QPIGS', '_command_description': 'General Status Parameters inquiry', 'validity check': ['Error: Invalid response CRCs', '']}
2023-10-20 12:56:44,082:DEBUG:__init__:main@426: results: {'_command': 'QPIGS', '_command_description': 'General Status Parameters inquiry', 'validity check': ['Error: Invalid response CRCs', '']}
2023-10-20 12:56:44,082:INFO:__init__:get_output@40: attempting to create output processor: json_mqtt
2023-10-20 12:56:44,112:DEBUG:json_mqtt:__init__@19: __init__: kwargs {}
2023-10-20 12:56:44,113:DEBUG:__init__:main@432: Using output filter: None
[... cut ...]
2023-10-20 12:56:44,115:DEBUG:mqttbrokerc:publish@146: Publishing '{"validity_check": "Error: Invalid response CRCs"}' to 'mppsolar'
2023-10-20 12:56:44,116:DEBUG:mqttbrokerc:publish@152: Not connected, connecting
2023-10-20 12:56:44,116:DEBUG:mqttbrokerc:connect@88: Connecting to localhost on port 1883
2023-10-20 12:56:44,116:DEBUG:mqttbrokerc:connect@95: No mqtt authentication used
2023-10-20 12:56:44,138:DEBUG:mqttbrokerc:on_connect@64: MqttBroker connection returned result: 0 Connection successful
2023-10-20 12:56:46,139:DEBUG:__init__:main@457: Not daemon, so not looping

sante85, 2023-10-20:
the output is wrong

sante85, 2023-10-20:
here not consider to apply tag variable
json_mqtt.*py

if tag is None:
            tag = cmd
        output = to_json(data, keep_case, excl_filter, filter)
        payload = js.dumps(output)
        msg = {
            "topic": topic,
            "payload": payload,
        }
        msgs.append(msg)
        return msgs

jblance, 2023-10-22:
covered by PR #414

closed 2023-10-22 by jblance