jblance/mpp-solar #415: JK BMS BLE Not Working - AttributeError: 'str' object has no attribute 'get_data_value'

luckylinux, 2023-10-23:
I have 2 x JK BMS that I just set up. One is HW v10 the other HW v11.

BMS Model should be B2A24S20P.

I tried to connect using BLE to both, but all I get is errors.

Tried both JK02 and JK04 as protocol, both fail.

Any idea what is going on ?

This is a 16s setup by the way (each BMS manages 16 cells). 

`(mppsolar) root@SolarMonitor:~# jkbms -p C8:47:8C:E5:98:96 -P JK02 -c getCellData
Command: getCellData - BLE Cell Data inquiry
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/opt/venv/mppsolar/bin/jkbms", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/__init__.py", line 434, in main
    op.output(
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/outputs/screen.py", line 81, in output
    maxP = get_max_response_length(displayData)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/helpers.py", line 100, in get_max_response_length
    if len(response.get_data_value()) > _max_length:
           ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'get_data_value'
(mppsolar) root@SolarMonitor:~# jkbms -p C8:47:8C:E5:98:96 -P JK04 -c getCellData
Command: getCellData - BLE Cell Data inquiry
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/opt/venv/mppsolar/bin/jkbms", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/__init__.py", line 434, in main
    op.output(
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/outputs/screen.py", line 81, in output
    maxP = get_max_response_length(displayData)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/helpers.py", line 100, in get_max_response_length
    if len(response.get_data_value()) > _max_length:
           ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'get_data_value'
(mppsolar) root@SolarMonitor:~# jkbms -p C8:47:8C:EC:1E:60 -P JK04 -c getCellData
Command: getCellData - BLE Cell Data inquiry
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/opt/venv/mppsolar/bin/jkbms", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/__init__.py", line 434, in main
    op.output(
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/outputs/screen.py", line 81, in output
    maxP = get_max_response_length(displayData)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/helpers.py", line 100, in get_max_response_length
    if len(response.get_data_value()) > _max_length:
           ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'get_data_value'
(mppsolar) root@SolarMonitor:~# jkbms -p C8:47:8C:EC:1E:60 -P JK02 -c getCellData
Command: getCellData - BLE Cell Data inquiry
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/opt/venv/mppsolar/bin/jkbms", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/__init__.py", line 434, in main
    op.output(
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/outputs/screen.py", line 81, in output
    maxP = get_max_response_length(displayData)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/helpers.py", line 100, in get_max_response_length
    if len(response.get_data_value()) > _max_length:
           ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'get_data_value'
(mppsolar) root@SolarMonitor:~# jkbms -p C8:47:8C:EC:1E:60 -P JK10 -c getCellData
2023-10-23 04:40:44,481:ERROR:__init__:get_protocol@21: No module found for protocol jk10
2023-10-23 04:40:44,482:ERROR:device:run_command@54: Attempted to run command with no protocol defined
Command: Unknown command - No description found
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/opt/venv/mppsolar/bin/jkbms", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/__init__.py", line 434, in main
    op.output(
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/outputs/screen.py", line 81, in output
    maxP = get_max_response_length(displayData)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/helpers.py", line 100, in get_max_response_length
    if len(response.get_data_value()) > _max_length:
           ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'get_data_value'
(mppsolar) root@SolarMonitor:~# jkbms -p C8:47:8C:EC:1E:60 -P JK11 -c getCellData
2023-10-23 04:40:49,232:ERROR:__init__:get_protocol@21: No module found for protocol jk11
2023-10-23 04:40:49,233:ERROR:device:run_command@54: Attempted to run command with no protocol defined
Command: Unknown command - No description found
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/opt/venv/mppsolar/bin/jkbms", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/__init__.py", line 434, in main
    op.output(
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/outputs/screen.py", line 81, in output
    maxP = get_max_response_length(displayData)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/mppsolar/src/mppsolar/mppsolar/helpers.py", line 100, in get_max_response_length
    if len(response.get_data_value()) > _max_length:
           ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'get_data_value'
`

luckylinux, 2023-10-23:
Is this issue due to a too-recent version of Python ? I have Python 3.11.2 installed on Raspberry Pi 3B (32 bit).

jblance, 2023-10-23:
can you post a debug run

luckylinux, 2023-10-24:
Alright ... I tried to follow the Troubleshooting guide.

```
(mppsolar) root@SolarMonitor:~# hcitool -i hci1 lescan
LE Scan ...
C8:47:8C:E5:98:96 (unknown)
C8:47:8C:E5:98:96 BAT02
C8:47:8C:EC:1E:60 (unknown)
C8:47:8C:EC:1E:60 BAT01
```

Keep in mind that BAT01 is supposedly HW 11.x and BAT02 is supposedly HW 10.x.
JKv11 seems to be the protocol for HW 11.x, but what am I supposed to use with HW 10.x ? JK04 ? In the wiki JK04 seems to be for (much) older hardware versions.

**Is there a way to specify which Interface to use with the jkbms command** ? I have 2 USB BLE adapters that will provide much better reception compared to the built-in bluetooth antenna of the Raspberry PI, which is sitting in a Metal Case. So Antenna transmit/reception will be greatly reduced if jkbms is using the internal BLE by default. I guess I could blacklist the driver for the Linux kernel if all else fails ...

BAT01 HW 11.x with -P JKv11:
```
(mppsolar) root@SolarMonitor:~# mpp-solar -p C8:47:8C:EC:1E:60 -P JKv11 -D
2023-10-24 05:34:49,375:INFO:__init__:main@216: Solar Device Command Utility, version: 0.16.12-dev, python version: 3.11.2
2023-10-24 05:34:49,375:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2023-10-24 05:34:49,376:DEBUG:__init__:main@251: MqttBroker name: localhost, port: 1883, user: None
2023-10-24 05:34:49,376:DEBUG:__init__:main@253: udp port 5555
2023-10-24 05:34:49,376:DEBUG:__init__:main@255: Using Postgres None
2023-10-24 05:34:49,377:DEBUG:__init__:main@258: Using Mongo None with mppsolar
2023-10-24 05:34:49,377:INFO:__init__:main@359: Creating device "unnamed" (type: "mppsolar") on port "C8:47:8C:EC:1E:60 (porttype=None)" using protocol "JKv11"
2023-10-24 05:34:49,383:DEBUG:__init__:main@363: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2023-10-24 05:34:49,383:DEBUG:device:__init__@34: __init__ args ()
2023-10-24 05:34:49,383:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': 'C8:47:8C:EC:1E:60', 'protocol': 'JKv11', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x755e4f50>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-10-24 05:34:49,384:DEBUG:__init__:get_port_type@51: port matches jkble ':'
2023-10-24 05:34:49,384:INFO:__init__:get_port@105: Using jkbleio for communications
2023-10-24 05:34:49,432:DEBUG:__init__:get_protocol@13: Protocol JKv11
2023-10-24 05:34:49,442:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.jkbleio.JkBleIO object at 0x756525b0>, protocol JKBMS BLE communication protocol handler software v11.x
2023-10-24 05:34:49,443:DEBUG:__init__:main@408: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x767e4b90>, '', '', 'screen', None, None)]
2023-10-24 05:34:49,443:INFO:__init__:main@413: Looping 1 commands
2023-10-24 05:34:49,443:INFO:__init__:main@423: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.jkbleio.JkBleIO object at 0x756525b0>, protocol: JKBMS BLE communication protocol handler software v11.x for command: , tag: , outputs: screen
2023-10-24 05:34:49,444:INFO:device:run_command@51: Running command 
2023-10-24 05:34:49,444:INFO:jkabstractprotocol:get_full_command@74: Using protocol b'JKv11' with 3 commands
2023-10-24 05:34:49,444:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2023-10-24 05:34:49,444:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getCellData'
2023-10-24 05:34:49,444:DEBUG:abstractprotocol:get_command_defn@73: Found command getCellData in protocol b'JKv11'
2023-10-24 05:34:49,445:DEBUG:jkabstractprotocol:get_full_command@79: self._command = getCellData
2023-10-24 05:34:49,445:DEBUG:jkabstractprotocol:get_full_command@90: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2023-10-24 05:34:49,445:DEBUG:jkabstractprotocol:get_full_command@98: cmd with command code: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2023-10-24 05:34:49,445:DEBUG:jkabstractprotocol:get_full_command@100: cmd with crc: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2023-10-24 05:34:49,446:INFO:device:run_command@81: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2023-10-24 05:34:49,446:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2023-10-24 05:34:49,446:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getCellData'
2023-10-24 05:34:49,446:DEBUG:abstractprotocol:get_command_defn@73: Found command getCellData in protocol b'JKv11'
2023-10-24 05:34:49,446:INFO:jkabstractprotocol:get_full_command@74: Using protocol b'JKv11' with 3 commands
2023-10-24 05:34:49,447:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2023-10-24 05:34:49,447:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getCellData'
2023-10-24 05:34:49,447:DEBUG:abstractprotocol:get_command_defn@73: Found command getCellData in protocol b'JKv11'
2023-10-24 05:34:49,447:DEBUG:jkabstractprotocol:get_full_command@79: self._command = getCellData
2023-10-24 05:34:49,447:DEBUG:jkabstractprotocol:get_full_command@90: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
[... cut ...]
AttributeError: 'str' object has no attribute 'get_data_value'

```

Looking at thee Debug information it seems to be a BLE communication issue:
```
2023-10-24 05:36:50,163:WARNING:jkbleio:ble_connect@61: Cannot connect to mac C8:47:8C:EC:1E:60 - exceeded 3 attempts

2023-10-24 05:34:16,563:WARNING:jkbleio:ble_connect@61: Cannot connect to mac C8:47:8C:E5:98:96 - exceeded 3 attempts

2023-10-24 05:30:54,323:WARNING:jkbleio:ble_connect@61: Cannot connect to mac C8:47:8C:E5:98:96 - exceeded 3 attempts
```

luckylinux, 2023-10-24:
Now I disabled onboard bluetooth of Raspberry Pi3 and I am getting somewhere ...

Add to /boot/config.txt
```
# Disable Bluetooth
dtoverlay=disable-bt
```

Attached are the DEBUG logs for:
- BAT01 HW 11.x with -P JKv11
- BAT02 HW 10.x with -P JK04
- BAT02 HW 10.x with -P JKv11
- 
[BAT02_HW10_JKv11_Communication_Working.txt](https://github.com/jblance/mpp-solar/files/13110566/BAT02_HW10_JKv11_Communication_Working.txt)
[BAT02_HW10_JK04_Communication_Working.txt](https://github.com/jblance/mpp-solar/files/13110567/BAT02_HW10_JK04_Communication_Working.txt)
[BAT01_HW11_JKv11_Communication_Working.txt](https://github.com/jblance/mpp-solar/files/13110568/BAT01_HW11_JKv11_Communication_Working.txt)

luckylinux, 2023-11-06:
Any update on this ?

jblance, 2023-11-22:
this bug should have been fixed in version 0.16.13

closed 2024-05-07 by jblance