jblance/mpp-solar #421: how can I update mpp-solar

ts1cos, 2023-11-05:
I'm on version 0.14.12 and would like to update mpp-solar but I don't know how.

Saentist, 2023-11-05:
https://github.com/jblance/mpp-solar/wiki#installation

ts1cos, 2023-11-05:
root@mpp-solar:~# sudo pip install mppsolar
Requirement already satisfied: mppsolar in /usr/local/lib/python3.10/dist-packages (0.14.12)
Requirement already satisfied: PyYAML in /usr/lib/python3/dist-packages (from mppsolar) (5.4.1)
Requirement already satisfied: paho-mqtt in /usr/local/lib/python3.10/dist-packages (from mppsolar) (1.6.1)
Requirement already satisfied: pyserial in /usr/local/lib/python3.10/dist-packages (from mppsolar) (3.5)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

Saentist, 2023-11-05:
```bash
mpp-solar -v
Solar Device Command Utility, version: 0.16.11-dev, python version: 3.8.10
```

jblance, 2023-11-22:
@ts1cos did you get updated?

ts1cos, 2023-11-23:
yes, I had to uninstall it and reinstall it and then it worked

Saentist, 2023-11-23:
> yes, I had to uninstall it and reinstall it and then it worked

How you uninstall it I can't

jblance, 2023-11-26:
> > yes, I had to uninstall it and reinstall it and then it worked
> 
> How you uninstall it I can't

@Saentist - How did you install?
if you used pip, then `pip uninstall mppsolar` (or `sudo pip uninstall mppsolar`) should do it

closed 2023-11-26 by jblance

Saentist, 2023-11-27:
@jblance 

```bash
root@ET1610PT:~# pip uninstall mppsolar
WARNING: Skipping mppsolar as it is not installed.
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

root@ET1610PT:~# pip3 uninstall mppsolar
WARNING: Skipping mppsolar as it is not installed.
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

root@ET1610PT:~# mpp-solar -v
Solar Device Command Utility, version: 0.16.11-dev, python version: 3.8.10
```

ts1cos, 2023-11-27:
pip uninstall mpp-solar

Saentist, 2023-11-27:
> pip uninstall mpp-solar

nothing different

```bash 
root@ET1610PT:~# pip3 uninstall mpp-solar
WARNING: Skipping mpp-solar as it is not installed.
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

root@ET1610PT:~# pip uninstall mpp-solar
WARNING: Skipping mpp-solar as it is not installed.
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
```

ts1cos, 2023-11-27:
sudo pip uninstall mpp-solar or login as root

jblance, 2023-11-27:
How did you install it?

Saentist, 2023-11-27:
> How did you install it?
> […](#)

With PIP i guess

```bash
# whereis mpp-solar
mpp-solar: /usr/local/bin/mpp-solar
```

jblance, 2023-11-27:
If installed with pip, pip should be able to uninstall

Otherwise you'll need to delete the actual files
`find / -name mppsolar` as root to see where (or look at the script pointed
to by the whereis command)

It's a pain, which I why I recommend pip installs now

Saentist, 2023-11-27:
```bash
# find / -name mppsola
/root/src/mppsolar
/root/src/mppsolar/mppsolar
/opt/INV/mpp-solar/build/lib/mppsolar     #cloned repo
/opt/INV/mpp-solar/src/mppsolar    #cloned repo
/opt/INV/mpp-solar/src/mppsolar/mppsolar   #cloned repo
/opt/INV/mpp-solar/mppsolar #cloned repo
/opt/INV/src/mppsolar
/opt/INV/src/mppsolar/tests/mppsolar
/opt/INV/src/mppsolar/mppsolar
/opt/src/mppsolar
/opt/src/mppsolar/mppsolar
/usr/lib/python3.8/site-packages/mppsolar-0.13.6-py3.8.egg/mppsolar

# find / -name mpp-solar
/opt/INV/mpp-solar #cloned repo
/usr/local/bin/mpp-solar

```
To remove them all or?

jblance, 2023-11-27:
If you are trying to clean up, yes
Lots of src repos there so I assume at some point you have installed from
src which can get messy fast
If you are trying things out I suggest venv or docker to keep it isolated

Saentist, 2023-11-27:
deleted all but as before same error with paho-mqtt https://github.com/jblance/mpp-solar/issues/397

```python
~# pip install mppsolar
Collecting mppsolar
  Downloading mppsolar-0.16.13-py3-none-any.whl.metadata (2.0 kB)
Collecting PyYAML>=6 (from mppsolar)
  Using cached PyYAML-6.0.1-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (2.1 kB)
Collecting fastapi>=0.68.0 (from mppsolar)
  Using cached fastapi-0.104.1-py3-none-any.whl.metadata (24 kB)
Collecting fastapi-mqtt>=1.1 (from mppsolar)
  Using cached fastapi_mqtt-2.0.0-py3-none-any.whl.metadata (5.1 kB)
Collecting jinja2>=3.0.0 (from mppsolar)
  Using cached Jinja2-3.1.2-py3-none-any.whl (133 kB)
Collecting paho-mqtt>=1.6 (from mppsolar)
  Using cached paho-mqtt-1.6.1.tar.gz (99 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      ERROR: Can not execute `setup.py` since setuptools is not available in the build environment.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

jblance, 2023-11-27:
Try deleting the cached paho tar file

jblance, 2023-11-28:
great that you've figured it out
another reason to use venv or docker

Saentist, 2023-11-29:
> great that you've figured it out another reason to use venv or docker

Venv complicated configuration for end user https://www.jetbrains.com/help/pycharm/creating-virtual-environment.html
Docker 1.5gb extra space just to run it slower running extra memory usage

Native is best solution but python version selection when there is more then one is problematical.

multilectical, 2023-12-01:
Hmmm, I just attempted to upgrade and ended doing a dist upgrade on raspbian, but now can only seem to get mppsolar running via venv which works. But can't figure out how to get the service started up again. Any direction here would be super appreciated.

I edited the path to where mpp seems to run in venv (/home/onion/venv/mppsolar/bin/mpp-solar) , but it's not working. Anyone have a good idea how to make this happen? 

here's my service file. I think I need to prefix python under execstart possibly???

```
onion@raspberrypi:~ $ cat /etc/xdg/systemd/user/mpp-solar.service

# systemd unit file for the MPP-Solar Service
#
# needs to go to /etc/systemd/user/*

[Unit]
# Human readable name of the unit
Description=MPP Solar Service

[Service]
# Command to execute when the service is started
ExecStart=/home/onion/venv/mppsolar/bin/python /home/onion/venv/mppsolar/bin/mpp-solar -C /etc/mpp-solar/mpp-solar.conf --daemon

# Disable Python's buffering of STDOUT and STDERR, so that output from the
# service shows up immediately in systemd's logs
Environment=PYTHONUNBUFFERED=1

# Automatically restart the service if it crashes
Restart=always
WatchdogSec=60

# Our service will notify systemd once it is up and running
Type=notify

# Use a dedicated user to run our service
# User=mpp_service

[Install]
# Tell systemd to automatically start this service when the system boots
# (assuming the service is enabled)
WantedBy=default.target

```

here's what it tells me went wrong:

```
onion@raspberrypi:~ $ systemctl --user status mpp-solar.service
× mpp-solar.service - MPP Solar Service
     Loaded: loaded (/etc/xdg/systemd/user/mpp-solar.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Fri 2023-12-01 12:31:15 EST; 3min 30s ago
    Process: 873 ExecStart=/home/onion/venv/mppsolar/bin/python /home/onion/venv/mppsolar/bin/mpp-solar -C /etc/mpp-solar/mpp-solar.conf ->
   Main PID: 873 (code=exited, status=1/FAILURE)
        CPU: 321ms
```

but when I try it directly it works:

```
onion@raspberrypi:~ $ /home/onion/venv/mppsolar/bin/python /home/onion/venv/mppsolar/bin/mpp-solar -P PI41 -o screen -c QPIGS -p /dev/hidraw0
Command: QPIGS - General Status Parameters inquiry LV5048
--------------------------------------------------------------------------------
Parameter                      Value          	Unit
l1_ac_input_voltage            121.6          	V   
l1_ac_input_frequency          60.0           	Hz  
l1_ac_output_voltage           121.6          	V   
l1_ac_output_frequency         60.0           	Hz  
l1_ac_output_apparent_power    85             	VA  
l1_ac_output_active_power      66             	W   
l1_ac_output_load              3              	%   
bus_voltage                    201            	V   
battery_voltage                50.7           	V   
battery_charging_current       4              	A   
battery_capacity               25             	%   
inverter_heat_sink_temperature 20             	Deg_C
pv_input_current_for_battery   4.0            	A   
pv_input_voltage               105.7          	V   
battery_voltage_from_scc       50.86          	V   
battery_discharge_current      0              	A   
is_l1_scc_ok                   1              	bool
is_l1_ac_charging_on           0              	bool
is_l1_scc_charging_on          1              	bool
is_battery_over                0              	bool
is_battery_under               0              	bool
is_l1_line_not_ok              1              	bool
is_load_on                     0              	bool
rsv1                           0              	A   
rsv2                           0              	A   
pv_input_power                 208            	W   
is_charging_to_float           0              	bool
is_switched_on                 1              	bool
is_reserved                    0              	bool
--------------------------------------------------------------------------------
```

multilectical, 2023-12-04:
I got it working. Just in case it can help others while updating, what I did that I think fixed things after updating to the virtual environment (this: https://github.com/jblance/mpp-solar/wiki/Installation#using-a-virtual-environment ) :

```
source venv/mppsolar/bin/activate // enter the new virtual env
pip install systemd-python // install python systemd which wasn't yet installed in the virtual env
```

I also made put a file with the following where the previous mpp-solar used to live at /usr/local/bin/mpp-solar

```
#!/bin/bash
/home/onion/venv/mppsolar/bin/python /home/onion/venv/mppsolar/bin/mpp-solar
```

That just calls the virtual env version without having to enter that environment on the cli and might be enough to make the service work without the above step.