jblance/mpp-solar #423: crash after update

knocki12, 2023-11-12:
Hello, I did an update to mppsolar using the command
pip install --upgrade mppsolar --root-user-action=ignore
After installing the new version, I got the following error:
````
mppsolar
Command: QPI - Protocol ID inquiry
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/bin/mppsolar", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/bin/src/mppsolar/mppsolar/__init__.py", line 434, in main
    op.output(
  File "/usr/local/bin/src/mppsolar/mppsolar/outputs/screen.py", line 81, in output
    maxP = get_max_response_length(displayData)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/bin/src/mppsolar/mppsolar/helpers.py", line 100, in get_max_response_length
    if len(response.get_data_value()) > _max_length:
           ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'get_data_value'

mpp-solar -v 
Solar Device Command Utility, version: 0.16.12-dev, python version: 3.11.2
```

jblance, 2023-11-12:
opps, thats a bug introduced from the powermon work
hopefully fixed in 0.16.13

knocki12, 2023-11-13:
Everything works, thanks