jblance/mpp-solar #443: No module found for output processor postgresql Error: No module named 'mppsolar.outputs.postgresql'

adinetech, 2023-12-16:
I run `mppsolar -p <mac-address> -P JK02 -c getCellData` to get the Cell Information.

[image]

I have tried `mppsolar -p <mac-address> -P JK02 -c getCellData -o postgresql --postgres_url postgresql://adine:passwordhere@192.168.0.12/battery` and It returns ```2023-12-16 09:31:28,793:CRITICAL:__init__:get_output@48: No module found for output processor postgresql Error: No module named 'mppsolar.outputs.postgresql'```

Initially I installed it by running `pip install mppsolar[ble]` and then `pip install mppsolar[pgsql]` and I have also tried `pip install mppsolar[ble,pgsql]`

I aim to export data directly to a database and then use Grafana to visualize it. I didn't have much luck with MQTT and Telegraf. Any help would be appreciated!

BoboTiG, 2023-12-16:
I am on it, I've fixed the name and missed the class name renaming.

BoboTiG, 2023-12-16:
Which mppsolar version is it?

adinetech, 2023-12-16:
It is `0.16.14`

```pi@raspberrypi:~$ mppsolar -v
Solar Device Command Utility, version: 0.16.14, python version: 3.10.12

BoboTiG, 2023-12-16:
Do you have more luck with `-o postgres`?
You can list of available outputs using `mpp-solar -o help` also.

adinetech, 2023-12-17:
No luck with that too.

```
pi@raspberrypi:~$ mppsolar -o help
Command: outputs help - List available output modules
--------------------------------------------------------------------------------
Parameter             Value             Unit
baseoutput            the base class for the output processors, not used directly           
boxdraw               outputs the results to standard out in a table formatted with line art boxes      
domoticz_autodiscover outputs the to the supplied mqtt broker in hass format: eg "homeassistant/sensor/mpp_{tag}_{key}/state"            
domoticz_mqtt         outputs the to the supplied mqtt broker in hass format: eg "homeassistant/sensor/mpp_{tag}_{key}/state"            
hass_mqtt             outputs the to the supplied mqtt broker in hass format: eg "homeassistant/sensor/mpp_{tag}_{key}/state"            
hassd_mqtt            outputs the to the supplied mqtt broker in hass format: eg "homeassistant/sensor/mpp_{tag}_{key}/state"            
influx2_mqtt          outputs the to the supplied mqtt broker: eg mpp-solar,command={tag} max_charger_range=120.0        
influx_mqtt           outputs the to the supplied mqtt broker: eg {tag}, {tag},setting=total_ac_output_apparent_power value=1577.0,unit="VA"             
json                  outputs the results to standard out in json format            
json_mqtt             outputs all the results to the supplied mqtt broker in a single message formatted as json: eg      
json_udp              outputs all the results to tcp UDP datagram packet in JSON format     
json_units            outputs the results (including units) to standard out in json format          
mongo*                ERROR: No module named 'pymongo'      
mqtt                  outputs the results to the supplied mqtt broker: eg {tag}/status/total_output_active_power/value 1250      
postgres*             ERROR: No module named 'psycopg2'     
prom                  outputs Node exporter prometheus format to standard out       
raw                   outputs the raw results to standard out       
screen                [the default output module] outputs the results to standard out in a slightly formatted way        
simple                outputs 'param=value' only to standard out            
simpleunits           outputs 'param=value(unit)' only to standard out      
tag_mqtt              outputs the to the supplied mqtt broker using the supplied tag as the topic: eg {tag}/max_charger_range 120.0      
value                 outputs the values only to standard out 
```

I installed psycopg2 by running `sudo apt install python3-psycopg2`

Seems like it's fine now, I have some database connection error, have to get that sorted.

```
Traceback (most recent call last):
  File "/home/pi/.local/bin/mppsolar", line 8, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python3.10/site-packages/mppsolar/__init__.py", line 434, in main
    op.output(
  File "/home/pi/.local/lib/python3.10/site-packages/mppsolar/outputs/postgres.py", line 50, in output
    conn = psycopg2.connect(postgres_url)
  File "/usr/lib/python3/dist-packages/psycopg2/__init__.py", line 122, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
psycopg2.OperationalError: could not translate host name "adine" to address: Temporary failure in name resolution
```

adinetech, 2023-12-17:
I was able to access the database with user adine and my password `psql -h localhost -U adine -d battery`

It doesn't work.
```
pi@raspberrypi:~$ mppsolar -p c8:47:8c:e2:2d:a2 -P JK02 -c getCellData -o postgres --postgres_url postgresql://adine:<pass-here@localhost/battery

Traceback (most recent call last):
  File "/home/pi/.local/bin/mppsolar", line 8, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python3.10/site-packages/mppsolar/__init__.py", line 434, in main
    op.output(
  File "/home/pi/.local/lib/python3.10/site-packages/mppsolar/outputs/postgres.py", line 50, in output
    conn = psycopg2.connect(postgres_url)
  File "/usr/lib/python3/dist-packages/psycopg2/__init__.py", line 122, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
psycopg2.OperationalError: could not translate host name "adine" to address: Temporary failure in name resolution
```

BoboTiG, 2023-12-17:
The URL you gave is `postgresql://adine:PASSWORD@localhost/battery`, it should have a port number specified, right? The port where psql is listening to, something like `postgresql://adine:PASSWORD@localhost:PORT/battery`, maybe?

adinetech, 2023-12-17:
I have tried that, and it returns the same error as mentioned previously.

adinetech, 2023-12-17:
I was able to create the Linux user adine and set up proper permissions to get this partially working. However, I encountered this error;

```
pi@raspberrypi:~$ mppsolar -p c8:47:8c:e2:2d:a2 -P JK02 -c getCellData -o postgres --postgres_url postgresql://adine:PASSWORD@localhost:5432/battery
2023-12-17 15:52:50,547:ERROR:postgres:output@78: Postgres error relation "mppsolar" does not exist
LINE 1: insert into mppsolar (command,data, updated) values ('getCel...
```

adinetech, 2023-12-17:
I manually created a table and it works now (I need to check whether I was able to get the data on the Grafana).
```
CREATE TABLE mppsolar (
    id SERIAL PRIMARY KEY,
    command VARCHAR(255),
    data JSONB,
    updated TIMESTAMP DEFAULT CURRENT_TIMESTAMP
);
```
```
pi@raspberrypi:~$ mppsolar -p c8:47:8c:e2:2d:a2 -P JK02 -c getCellData -o postgres --postgres_url postgresql://adine:PASSWORD@localhost:5432/battery
pi@raspberrypi:~$ 
```
@BoboTiG How to run this command automatically every x seconds/minutes? BoboTiG, do you have any suggestions?

BoboTiG, 2023-12-17:
> @BoboTiG How to run this command automatically every x seconds/minutes? BoboTiG, do you have any suggestions?

Yup, I'll share a solution when I'll be back home.

adinetech, 2023-12-17:
It's all set! I used crontab to repeat the data acquisition process every minute, then forked and customized https://github.com/ch-reck/jkbms-influxdb-grafana for my specific needs. I'm excited to write a blog post/create a repo soon.

[image]

closed 2023-12-17 by jblance