jblance/mpp-solar #445: Regarding to JK BMS CAN Data reading

Sanket01-ai, 2023-12-18:
Hello dear excellencies. I have a BMS of JK B2A24S15P Model. But I am facing some problems to read the CAN Data . Please guide me regarding to my problem.
The code is as following -

#include "Arduino.h"
#include "mcp2515.h"
const struct MCP2515::TXBn_REGS MCP2515::TXB[MCP2515::N_TXBUFFERS] = {
    {MCP_TXB0CTRL, MCP_TXB0SIDH, MCP_TXB0DATA},
    {MCP_TXB1CTRL, MCP_TXB1SIDH, MCP_TXB1DATA},
    {MCP_TXB2CTRL, MCP_TXB2SIDH, MCP_TXB2DATA}
};
const struct MCP2515::RXBn_REGS MCP2515::RXB[N_RXBUFFERS] = {
    {MCP_RXB0CTRL, MCP_RXB0SIDH, MCP_RXB0DATA, CANINTF_RX0IF},
    {MCP_RXB1CTRL, MCP_RXB1SIDH, MCP_RXB1DATA, CANINTF_RX1IF}
};
MCP2515::MCP2515(const uint8_t _CS, const uint32_t _SPI_CLOCK, SPIClass * _SPI)
{
    if (_SPI != nullptr) {
        SPIn = _SPI;
    }
    else {
        SPIn = &SPI;
        SPIn->begin();
    }
    SPICS = _CS;
    SPI_CLOCK = _SPI_CLOCK;
    pinMode(SPICS, OUTPUT);
    endSPI();
    digitalWrite(SPICS, HIGH);
}

void MCP2515::startSPI() {
    SPIn->beginTransaction(SPISettings(SPI_CLOCK, MSBFIRST, SPI_MODE0));
    digitalWrite(SPICS, LOW);
}
void MCP2515::endSPI() {
    digitalWrite(SPICS, HIGH);
    SPIn->endTransaction();
}
MCP2515::ERROR MCP2515::reset(void)
{
    startSPI();
    SPIn->transfer(INSTRUCTION_RESET);
    endSPI();
    delay(10);
    uint8_t zeros[14];
    memset(zeros, 0, sizeof(zeros));
    setRegisters(MCP_TXB0CTRL, zeros, 14);
    setRegisters(MCP_TXB1CTRL, zeros, 14);
    setRegisters(MCP_TXB2CTRL, zeros, 14);
    setRegister(MCP_RXB0CTRL, 0);
    setRegister(MCP_RXB1CTRL, 0);
    setRegister(MCP_CANINTE, CANINTF_RX0IF | CANINTF_RX1IF | CANINTF_ERRIF | CANINTF_MERRF);
    // receives all valid messages using either Standard or Extended Identifiers that
    // meet filter criteria. RXF0 is applied for RXB0, RXF1 is applied for RXB1
    modifyRegister(MCP_RXB0CTRL,
                   RXBnCTRL_RXM_MASK | RXB0CTRL_BUKT | RXB0CTRL_FILHIT_MASK,
                   RXBnCTRL_RXM_STDEXT | RXB0CTRL_BUKT | RXB0CTRL_FILHIT);
    modifyRegister(MCP_RXB1CTRL,
                   RXBnCTRL_RXM_MASK | RXB1CTRL_FILHIT_MASK,
                   RXBnCTRL_RXM_STDEXT | RXB1CTRL_FILHIT);
    // clear filters and masks
    // do not filter any standard frames for RXF0 used by RXB0
    // do not filter any extended frames for RXF1 used by RXB1
    RXF filters[] = {RXF0, RXF1, RXF2, RXF3, RXF4, RXF5};
    for (int i=0; i<6; i++) {
        bool ext = (i == 1);
        ERROR result = setFilter(filters[i], ext, 0);
        if (result != ERROR_OK) {
            return result;
        }
[... 551 lines cut ...]
    clearInterrupts();
    //modifyRegister(MCP_CANINTF, CANINTF_ERRIF, 0);
  }
  
}
void MCP2515::clearMERR()
{
  //modifyRegister(MCP_EFLG, EFLG_RX0OVR | EFLG_RX1OVR, 0);
  //clearInterrupts();
  modifyRegister(MCP_CANINTF, CANINTF_MERRF, 0);
}
void MCP2515::clearERRIF()
{
    //modifyRegister(MCP_EFLG, EFLG_RX0OVR | EFLG_RX1OVR, 0);
    //clearInterrupts();
    modifyRegister(MCP_CANINTF, CANINTF_ERRIF, 0);
}
uint8_t MCP2515::errorCountRX(void)                             
{
    return readRegister(MCP_REC);
}
uint8_t MCP2515::errorCountTX(void)                             
{
    return readRegister(MCP_TEC);
}

jblance, 2023-12-20:
not related to this project

closed 2023-12-20 by jblance

Sanket01-ai, 2023-12-21:
> not related to this project

Do you mean the Code which i mentioned above is not related to the CAN related fot JK BMS

jblance, 2023-12-21:
No I mean that the code is no related to this project - this is a python project that comunicates with a few devices (including some JKBMS) 
This is not the JKBMS vendor support project

Sanket01-ai, 2023-12-22:
Okay thank you for your clarification....But sir If you have JK B2A24S15P file please help me.