jblance/mpp-solar #453: Command ACCT0200-0500 was rejected

eMindFfM, 2024-01-02:
Hi All,

I'm running a stange error. I can query and also send command over the command line an my raspbery PI 5.

(mppsolar) pi@Pi5Test:~ $ mpp-solar -p /dev/hidraw0 -P PI17 -c EDB1
Command: EDB1 - Enable/disable AC charge battery
--------------------------------------------------------------------------------
Parameter Value                 Unit
edb       ACK
--------------------------------------------------------------------------------

But when I try to an AC charge time I'm getting an error. Something seems o go wrong :-(
Please find attached the execution in debugmode.

(mppsolar) pi@Pi5Test:~ $ mpp-solar -p /dev/hidraw0 -P PI17 -c ACCT0200-0500 -D
2024-01-02 22:41:55,660:INFO:__init__:main@205: Solar Device Command Utility, version: 0.16.17-dev, python version: 3.11.2
2024-01-02 22:41:55,660:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-01-02 22:41:55,660:DEBUG:__init__:main@240: MqttBroker name: localhost, port: 1883, user: None
2024-01-02 22:41:55,660:DEBUG:__init__:main@242: udp port 5555
2024-01-02 22:41:55,661:DEBUG:__init__:main@244: Using Postgres None
2024-01-02 22:41:55,661:DEBUG:__init__:main@247: Using Mongo None with mppsolar
2024-01-02 22:41:55,661:INFO:__init__:main@358: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2024-01-02 22:41:55,662:DEBUG:__init__:main@362: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2024-01-02 22:41:55,662:DEBUG:device:__init__@34: __init__ args ()
2024-01-02 22:41:55,662:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7fff4769f5d0>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway'}
2024-01-02 22:41:55,662:DEBUG:__init__:get_port_type@39: port matches hidraw
2024-01-02 22:41:55,662:INFO:__init__:get_port@93: Using hidrawio for communications
2024-01-02 22:41:55,662:DEBUG:__init__:get_protocol@13: Protocol PI17
2024-01-02 22:41:55,767:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0x7fff477177d0>, protocol PI17 protocol handler
2024-01-02 22:41:55,767:DEBUG:__init__:main@408: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7fff47717350>, 'ACCT0200-0500', 'ACCT0200-0500', 'screen', None, None)]
2024-01-02 22:41:55,767:INFO:__init__:main@413: Looping 1 commands
2024-01-02 22:41:55,767:INFO:__init__:main@423: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x7fff477177d0>, protocol: PI17 protocol handler for command: ACCT0200-0500, tag: ACCT0200-0500, outputs: screen
2024-01-02 22:41:55,767:INFO:device:run_command@51: Running command ACCT0200-0500
2024-01-02 22:41:55,767:INFO:pi17:get_full_command@890: Using protocol b'PI17' with 48 commands
2024-01-02 22:41:55,767:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'ACCT0200-0500'
2024-01-02 22:41:55,767:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EY
2024-01-02 22:41:55,767:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EM
2024-01-02 22:41:55,767:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: ED
2024-01-02 22:41:55,768:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EH
2024-01-02 22:41:55,768:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: LON
2024-01-02 22:41:55,768:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PA
2024-01-02 22:41:55,768:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PB
2024-01-02 22:41:55,768:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PC
2024-01-02 22:41:55,768:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PD
2024-01-02 22:41:55,768:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PE
2024-01-02 22:41:55,768:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PF
2024-01-02 22:41:55,768:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: DAT
2024-01-02 22:41:55,768:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: LST
2024-01-02 22:41:55,768:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDA
2024-01-02 22:41:55,768:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDB
2024-01-02 22:41:55,768:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDC
2024-01-02 22:41:55,768:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDD
2024-01-02 22:41:55,768:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDE
2024-01-02 22:41:55,770:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDF
2024-01-02 22:41:55,770:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDG
2024-01-02 22:41:55,770:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: BT
2024-01-02 22:41:55,770:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: MCHGV
2024-01-02 22:41:55,770:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: ACCT
2024-01-02 22:41:55,770:DEBUG:abstractprotocol:get_command_defn@81: Matched: ACCT0200-0500 to: ACCT value: 0200-0500
2024-01-02 22:41:55,770:DEBUG:pi17:get_full_command@931: _pre_cmd: b'^S014ACCT0200-0500'
2024-01-02 22:41:55,770:DEBUG:pi17:get_full_command@938: full command: b'^S014ACCT0200-0500\r'
2024-01-02 22:41:55,770:INFO:device:run_command@81: full command b'^S014ACCT0200-0500\r' for command ACCT0200-0500
2024-01-02 22:41:55,770:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'ACCT0200-0500'
2024-01-02 22:41:55,770:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EY
2024-01-02 22:41:55,770:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EM
2024-01-02 22:41:55,770:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: ED
2024-01-02 22:41:55,770:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EH
2024-01-02 22:41:55,770:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: LON
2024-01-02 22:41:55,770:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PA
2024-01-02 22:41:55,770:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PB
[... 44 lines cut ...]
2024-01-02 22:41:56,386:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDF
2024-01-02 22:41:56,386:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDG
2024-01-02 22:41:56,387:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: BT
2024-01-02 22:41:56,387:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: MCHGV
2024-01-02 22:41:56,387:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: ACCT
2024-01-02 22:41:56,387:DEBUG:abstractprotocol:get_command_defn@81: Matched: ACCT0200-0500 to: ACCT value: 0200-0500
2024-01-02 22:41:56,387:INFO:abstractprotocol:decode@280: Processing response of type DEFAULT
2024-01-02 22:41:56,387:DEBUG:abstractprotocol:decode@284: trimmed and split responses: ['NAK']
2024-01-02 22:41:56,387:INFO:abstractprotocol:decode@291: Processing DEFAULT type responses
2024-01-02 22:41:56,387:DEBUG:abstractprotocol:decode@308: result NAK, key Command execution, resp_format ['ack', 'Command execution', {'NAK': 'Failed', 'ACK': 'Successful'}]
2024-01-02 22:41:56,387:INFO:device:run_command@116: Decoded response {'_command': 'ACCT0200-0500', '_command_description': 'Set AC charge time range', 'raw_response': ['^0\x1bã\r', ''], 'WARNING0': ['Command ACCT0200-0500 was rejected', '']}
2024-01-02 22:41:56,387:DEBUG:__init__:main@427: results: {'_command': 'ACCT0200-0500', '_command_description': 'Set AC charge time range', 'raw_response': ['^0\x1bã\r', ''], 'WARNING0': ['Command ACCT0200-0500 was rejected', '']}
2024-01-02 22:41:56,387:INFO:__init__:get_output@40: attempting to create output processor: screen
2024-01-02 22:41:56,387:DEBUG:screen:__init__@15: processor.screen __init__ kwargs {}
2024-01-02 22:41:56,387:DEBUG:__init__:main@433: Using output filter: None
2024-01-02 22:41:56,387:INFO:screen:output@21: Using output processor: screen
[... cut ...] se': False}
2024-01-02 22:41:56,387:DEBUG:screen:output@66: displayData: {'warning0': ['Command ACCT0200-0500 was rejected', '']}
Command: ACCT0200-0500 - Set AC charge time range
--------------------------------------------------------------------------------
Parameter Value                 Unit
warning0  Command ACCT0200-0500 was rejected
--------------------------------------------------------------------------------

2024-01-02 22:41:56,388:DEBUG:__init__:main@459: Not daemon, so not looping

Saentist, 2024-01-02:
please use code button when paste "<>"

eMindFfM, 2024-01-03:
No problem! Thnaks for the note. Any ideas so far?

jblance, 2024-01-07:
Its either the 'chunking' ie the breaking up into multiple frames to send the command (which makes it hard to do much with) or the command isnt compatible with your inverter, theres a doc 'PI17 - Infini-Solar WP LV6K_3P 10-15K and TWIN Series protocol 20220921.pdf' that has 
[image]
which suggests a much larger command (which isnt implemented so far)

jblance, 2024-01-07:
can you post the results of  `mpp-solar -p /dev/hidraw0 -P PI17 --getDeviceId`

eMindFfM, 2024-01-08:
`(mppsolar) pi@Pi5Test:~ $ mpp-solar -p /dev/hidraw0 -P PI17 --getDeviceId
Command: Get Device ID - Generate a device id
--------------------------------------------------------------------------------
Parameter Value                 Unit
deviceid  Command PI was rejected:050
--------------------------------------------------------------------------------

(mppsolar) pi@Pi5Test:~ $  mpp-solar -p /dev/hidraw0 -P PI17 -c MD
Command: MD - Device Model inquiry
--------------------------------------------------------------------------------
Parameter                         Value                 Unit
machine_number                    Infini-Solar 10KW/3P
output_rated_va                   5000                  kW
output_power_factor               99                    pf
ac_input_phase_number             1                     number
ac_output_phase_number            1                     number
norminal_ac_output_voltage        2300                  0.1V
norminal_ac_input_voltage         2300                  0.1V
battery_piece_number              04                    ea
battery_standard_voltage_per_unit 120                   0.1V
--------------------------------------------------------------------------------
`

jblance, 2024-01-08:
with the latest git code, can you try a command like:
`$ mpp-solar -p /dev/hidraw0 -P pi17infini -c ACCT2200,2230,2300,2330 -D`

eMindFfM, 2024-01-09:
Great I seems to work :-)
I'll do some more testing, but it looks good!

`````
(mppsolar) pi@Pi5Test:~ $ pip install -e "git+https://github.com/jblance/mpp-solar.git#egg=mppsolar"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining mppsolar from git+https://github.com/jblance/mpp-solar.git#egg=mppsolar
  Updating ./venv/mppsolar/src/mppsolar clone
  Running command git fetch -q --tags
  Running command git reset --hard -q e8d405cd6e5d087bf7fabb0ea0474fa15d7ed7aa
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Requirement already satisfied: paho-mqtt>=1.6 in ./venv/mppsolar/lib/python3.11/site-packages (from mppsolar) (1.6.1)
Requirement already satisfied: pydantic>=1.8.0 in ./venv/mppsolar/lib/python3.11/site-packages (from mppsolar) (2.5.3)
Requirement already satisfied: pyserial>3 in ./venv/mppsolar/lib/python3.11/site-packages (from mppsolar) (3.5)
Requirement already satisfied: annotated-types>=0.4.0 in ./venv/mppsolar/lib/python3.11/site-packages (from pydantic>=1.8.0->mppsolar) (0.6.0)
Requirement already satisfied: pydantic-core==2.14.6 in ./venv/mppsolar/lib/python3.11/site-packages (from pydantic>=1.8.0->mppsolar) (2.14.6)
Requirement already satisfied: typing-extensions>=4.6.1 in ./venv/mppsolar/lib/python3.11/site-packages (from pydantic>=1.8.0->mppsolar) (4.9.0)
Building wheels for collected packages: mppsolar
  Building editable for mppsolar (pyproject.toml) ... done
  Created wheel for mppsolar: filename=mppsolar-0.16.18-py3-none-any.whl size=3127 sha256=32c7c71baf40f86a782245cda9b83bc1f216f9d9958e56f8eddb939b658c566b
  Stored in directory: /tmp/pip-ephem-wheel-cache-t357_zt4/wheels/f3/e3/f7/ec3b637e8c6a348defc0d5b6431d99d22b59b381a71d9ecc5c
Successfully built mppsolar
Installing collected packages: mppsolar
  Attempting uninstall: mppsolar
    Found existing installation: mppsolar 0.16.17
    Uninstalling mppsolar-0.16.17:
      Successfully uninstalled mppsolar-0.16.17
Successfully installed mppsolar-0.16.18
(mppsolar) pi@Pi5Test:~ $ mpp-solar -p /dev/hidraw0 -P PI17 -c HECS
Command: HECS - Query energy control status
--------------------------------------------------------------------------------
Parameter                                              Value            Unit
solar_energy_distribution_priority                     Load-Battery-Grid
solar_charge_battery                                   enabled
ac_charge_battery                                      disabled
feed_power_to_utility                                  enabled
battery_discharge_to_loads_when_solar_input_normal     enabled
battery_discharge_to_loads_when_solar_input_loss       enabled
battery_discharge_to_feed_grid_when_solar_input_normal disabled
battery_discharge_to_feed_grid_when_solar_input_loss   disabled
reserved                                               disabled
--------------------------------------------------------------------------------

(mppsolar) pi@Pi5Test:~ $ mpp-solar -p /dev/hidraw0 -P pi17infini -c ACCT2200,2230,2300,2330 -D
2024-01-09 23:03:13,834:INFO:__init__:main@205: Solar Device Command Utility, version: 0.16.18-dev, python version: 3.11.2
2024-01-09 23:03:13,834:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-01-09 23:03:13,835:DEBUG:__init__:main@240: MqttBroker name: localhost, port: 1883, user: None
2024-01-09 23:03:13,835:DEBUG:__init__:main@242: udp port 5555
2024-01-09 23:03:13,835:DEBUG:__init__:main@244: Using Postgres None
2024-01-09 23:03:13,835:DEBUG:__init__:main@247: Using Mongo None with mppsolar
2024-01-09 23:03:13,835:INFO:__init__:main@358: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "pi17infini"
2024-01-09 23:03:13,836:DEBUG:__init__:main@362: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2024-01-09 23:03:13,837:DEBUG:device:__init__@34: __init__ args ()
2024-01-09 23:03:13,837:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'pi17infini', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7fff584d6d10>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway'}
2024-01-09 23:03:13,837:DEBUG:__init__:get_port_type@39: port matches hidraw
2024-01-09 23:03:13,837:INFO:__init__:get_port@93: Using hidrawio for communications
2024-01-09 23:03:13,838:DEBUG:__init__:get_protocol@13: Protocol pi17infini
2024-01-09 23:03:13,946:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0x7fff584d7590>, protocol PI17INFINI protocol handler
2024-01-09 23:03:13,946:DEBUG:__init__:main@408: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7fff584d7410>, 'ACCT2200,2230,2300,2330', 'ACCT2200,2230,2300,2330', 'screen', None, None)]
2024-01-09 23:03:13,946:INFO:__init__:main@413: Looping 1 commands
2024-01-09 23:03:13,946:INFO:__init__:main@423: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x7fff584d7590>, protocol: PI17INFINI protocol handler for command: ACCT2200,2230,2300,2330, tag: ACCT2200,2230,2300,2330, outputs: screen
[... cut ...] , 'push_url': 'http://localhost:9091/metrics/job/pushgateway', 'mqtt_topic': None, 'filter': None, 'excl_filter': None, 'keep_case': False}
2024-01-09 23:03:14,619:DEBUG:screen:output@66: displayData: {'acct': ['ACK', '']}
Command: ACCT2200,2230,2300,2330 - Set AC charge time range
--------------------------------------------------------------------------------
Parameter Value                 Unit
acct      ACK
--------------------------------------------------------------------------------
````

closed 2024-01-16 by jblance