jblance/mpp-solar #457: Waveshare RS232/RS485 to Ethernet Converter Question

preussal, 2024-01-27:
Hello,

I got a "Waveshare RS232/RS485 to Ethernet Converter".

I still have to see how I get the Seriel Port via the network on my Linux server.
Has anyone already had experience with it?
I could use it on my Mppsolar inverter via RS232 and make it independent of the computer.

I found nothing about it in the documentary.
Maybe someone can give me a few tips

preussal, 2024-01-27:
Apparently you can do this with ser2net.
Or directly with Pyserial

sante85, 2024-02-14:
@preussal not run for me
jkbms -p /dev/ttyUSB0  -P JK02 -c getCellData -D
2024-02-14 13:48:26,959:INFO:__init__:main@205: Solar Device Command Utility, version: 0.16.19, python version: 3.11.1
2024-02-14 13:48:26,959:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-02-14 13:48:26,959:DEBUG:__init__:main@240: MqttBroker name: localhost, port: 1883, user: None
2024-02-14 13:48:26,959:DEBUG:__init__:main@242: udp port 5555
2024-02-14 13:48:26,960:DEBUG:__init__:main@244: Using Postgres None
2024-02-14 13:48:26,960:DEBUG:__init__:main@247: Using Mongo None with mppsolar
2024-02-14 13:48:26,960:INFO:__init__:main@358: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB0 (porttype=None)" using protocol "JK02"
2024-02-14 13:48:26,963:DEBUG:__init__:main@362: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2024-02-14 13:48:26,963:DEBUG:device:__init__@34: __init__ args ()
2024-02-14 13:48:26,963:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'JK02', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0xf6f33ad0>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway'}
2024-02-14 13:48:26,964:DEBUG:__init__:get_port_type@69: port matches ttyusb
2024-02-14 13:48:26,964:INFO:__init__:get_port@111: Using serialio for communications
2024-02-14 13:48:26,971:DEBUG:__init__:get_protocol@13: Protocol JK02
2024-02-14 13:48:27,291:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0xf6e0d9f0>, protocol JK02 - JKBMS BLE 2 byte data communication protocol handler
2024-02-14 13:48:27,291:DEBUG:__init__:main@408: Commands [(<mppsolar.devices.jkbms.jkbms object at 0xf6e0d970>, 'getCellData', 'getCellData', 'screen', None, None)]
2024-02-14 13:48:27,291:INFO:__init__:main@413: Looping 1 commands
2024-02-14 13:48:27,292:INFO:__init__:main@423: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0xf6e0d9f0>, protocol: JK02 - JKBMS BLE 2 byte data communication protocol handler for command: getCellData, tag: getCellData, outputs: screen
2024-02-14 13:48:27,292:INFO:device:run_command@51: Running command getCellData
2024-02-14 13:48:27,292:INFO:jkabstractprotocol:get_full_command@74: Using protocol b'JK02' with 3 commands
2024-02-14 13:48:27,292:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2024-02-14 13:48:27,292:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getCellData'
2024-02-14 13:48:27,292:DEBUG:abstractprotocol:get_command_defn@73: Found command getCellData in protocol b'JK02'
2024-02-14 13:48:27,292:DEBUG:jkabstractprotocol:get_full_command@79: self._command = getCellData
2024-02-14 13:48:27,292:DEBUG:jkabstractprotocol:get_full_command@90: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-02-14 13:48:27,292:DEBUG:jkabstractprotocol:get_full_command@98: cmd with command code: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-02-14 13:48:27,293:DEBUG:jkabstractprotocol:get_full_command@100: cmd with crc: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2024-02-14 13:48:27,293:INFO:device:run_command@81: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2024-02-14 13:48:27,293:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2024-02-14 13:48:27,293:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getCellData'
2024-02-14 13:48:27,293:DEBUG:abstractprotocol:get_command_defn@73: Found command getCellData in protocol b'JK02'
2024-02-14 13:48:27,293:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 2400
2024-02-14 13:48:27,299:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2024-02-14 13:48:28,400:DEBUG:serialio:send_and_receive@30: serial response was: b''
2024-02-14 13:48:28,405:DEBUG:device:run_command@98: Send and Receive Response b''
2024-02-14 13:48:28,405:INFO:abstractprotocol:decode@236: response passed to decode: b''
2024-02-14 13:48:28,405:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2024-02-14 13:48:28,405:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getCellData'
2024-02-14 13:48:28,405:DEBUG:abstractprotocol:get_command_defn@73: Found command getCellData in protocol b'JK02'
2024-02-14 13:48:28,406:INFO:abstractprotocol:decode@280: Processing response of type POSITIONAL
2024-02-14 13:48:28,406:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 4, 'Header', '']
2024-02-14 13:48:28,406:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 1, 'Record_Type', '']
2024-02-14 13:48:28,406:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Int', 1, 'Record_Counter', '']
2024-02-14 13:48:28,406:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['LittleHex2Short:r/1000', 2, 'Voltage_Cell01', 'V']
2024-02-14 13:48:28,406:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['LittleHex2Short:r/1000', 2, 'Voltage_Cell02', 'V']
[... cut ...]
    processed_responses = self.process_response(
                          ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/mppsolar/protocols/abstractprotocol.py", line 203, in process_response
    r = eval(format_string)
        ^^^^^^^^^^^^^^^^^^^
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.11/site-packages/mppsolar/protocols/protocol_helpers.py", line 90, in Hex2Int
    answer = hexString[0]
             ~~~~~~~~~^^^
IndexError: index out of range

jblance, 2024-02-14:
protocols jk02, jk04 and jkv11 are for bluetooth connections and arent applicable to serial connections try jkserial, jk232 or jk485

jblance, 2024-02-22:
if you are still interested, first update to version 0.16.22+ anf then try:
`mppsolar -p /dev/ttyUSB0 -P jkserial -D` 
and post the results - this will allow me to diagnose