jblance/mpp-solar #458: smart shunt fail read some fields

sante85, 2024-01-30:
Hi if i send this command, fail 
mppsolar -P VED -p /dev/ttyUSB0 -b 19200 -c vedtext --porttype vserial -D

my workaround is add a --filter option run properly, but i have a missing field.
--filter "main_or_channel_1_battery_voltage|main_or_channel_1_battery_current|instantaneous_power|consumed_amp_hours|state-of-charge|model_description"

this is a debug run trace
mppsolar -P VED -p /dev/ttyUSB0 -b 19200 -c vedtext --porttype vserial -D
2024-01-30 11:17:15,747:INFO:__init__:main@205: Solar Device Command Utility, version: 0.16.18, python version: 3.11.1
2024-01-30 11:17:15,747:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-01-30 11:17:15,748:DEBUG:__init__:main@240: MqttBroker name: localhost, port: 1883, user: None
2024-01-30 11:17:15,748:DEBUG:__init__:main@242: udp port 5555
2024-01-30 11:17:15,748:DEBUG:__init__:main@244: Using Postgres None
2024-01-30 11:17:15,749:DEBUG:__init__:main@247: Using Mongo None with mppsolar
2024-01-30 11:17:15,749:INFO:__init__:main@358: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=vserial)" using protocol "VED"
2024-01-30 11:17:15,755:DEBUG:__init__:main@362: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2024-01-30 11:17:15,755:DEBUG:device:__init__@34: __init__ args ()
2024-01-30 11:17:15,756:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'VED', 'baud': 19200, 'porttype': 'vserial', 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x768ab790>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway'}
2024-01-30 11:17:15,756:INFO:__init__:get_port@81: Port overide - using port 'vserial'
2024-01-30 11:17:15,756:DEBUG:__init__:get_port_type@63: port matches vserial
2024-01-30 11:17:15,756:INFO:__init__:get_port@123: Using vserialio for communications
2024-01-30 11:17:15,770:DEBUG:__init__:get_protocol@13: Protocol VED
2024-01-30 11:17:17,045:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.vserialio.VSerialIO object at 0x75fd9830>, protocol VED protocol handler for Victron direct SmartShunts
2024-01-30 11:17:17,045:DEBUG:__init__:main@408: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x75fd9a70>, 'vedtext', 'vedtext', 'screen', None, None)]
2024-01-30 11:17:17,046:INFO:__init__:main@413: Looping 1 commands
2024-01-30 11:17:17,047:INFO:__init__:main@423: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.vserialio.VSerialIO object at 0x75fd9830>, protocol: VED protocol handler for Victron direct SmartShunts for command: vedtext, tag: vedtext, outputs: screen
2024-01-30 11:17:17,047:INFO:device:run_command@51: Running command vedtext
2024-01-30 11:17:17,047:INFO:ved:get_full_command@143: Using protocol b'VED' with 2 commands
2024-01-30 11:17:17,048:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'vedtext'
2024-01-30 11:17:17,048:DEBUG:abstractprotocol:get_command_defn@73: Found command vedtext in protocol b'VED'
2024-01-30 11:17:17,048:DEBUG:ved:get_full_command@164: command is VEDTEXT type so returning VEDTEXT
2024-01-30 11:17:17,049:INFO:device:run_command@81: full command VEDTEXT for command vedtext
2024-01-30 11:17:17,049:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'vedtext'
2024-01-30 11:17:17,049:DEBUG:abstractprotocol:get_command_defn@73: Found command vedtext in protocol b'VED'
2024-01-30 11:17:17,050:DEBUG:vserialio:send_and_receive@29: port /dev/ttyUSB0, baudrate 19200, records 30
2024-01-30 11:17:17,068:DEBUG:vserialio:send_and_receive@46: vserial response was: b'\x008\t16783\r\n'
2024-01-30 11:17:17,483:DEBUG:vserialio:send_and_receive@46: vserial response was: b'Checksum\t\x16\r\n'
2024-01-30 11:17:17,485:DEBUG:vserialio:send_and_receive@46: vserial response was: b'PID\t0xA389\r\n'
2024-01-30 11:17:17,500:DEBUG:vserialio:send_and_receive@46: vserial response was: b'V\t53509\r\n'
2024-01-30 11:17:17,501:DEBUG:vserialio:send_and_receive@46: vserial response was: b'I\t23866\r\n'
2024-01-30 11:17:17,503:DEBUG:vserialio:send_and_receive@46: vserial response was: b'P\t1277\r\n'
2024-01-30 11:17:17,515:DEBUG:vserialio:send_and_receive@46: vserial response was: b'CE\t-131099\r\n'
2024-01-30 11:17:17,516:DEBUG:vserialio:send_and_receive@46: vserial response was: b'SOC\t548\r\n'
2024-01-30 11:17:17,517:DEBUG:vserialio:send_and_receive@46: vserial response was: b'TTG\t-1\r\n'
2024-01-30 11:17:17,531:DEBUG:vserialio:send_and_receive@46: vserial response was: b'Alarm\tOFF\r\n'
2024-01-30 11:17:17,532:DEBUG:vserialio:send_and_receive@46: vserial response was: b'AR\t0\r\n'
2024-01-30 11:17:17,547:DEBUG:vserialio:send_and_receive@46: vserial response was: b'BMV\tSmartShunt 500A/50mV\r\n'
2024-01-30 11:17:17,549:DEBUG:vserialio:send_and_receive@46: vserial response was: b'FW\t0414\r\n'
2024-01-30 11:17:17,550:DEBUG:vserialio:send_and_receive@46: vserial response was: b'MON\t0\r\n'
2024-01-30 11:17:17,979:DEBUG:vserialio:send_and_receive@46: vserial response was: b'Checksum\t\xdf\r\n'
2024-01-30 11:17:17,995:DEBUG:vserialio:send_and_receive@46: vserial response was: b'H1\t-276495\r\n'
2024-01-30 11:17:17,997:DEBUG:vserialio:send_and_receive@46: vserial response was: b'H2\t-238319\r\n'
2024-01-30 11:17:18,011:DEBUG:vserialio:send_and_receive@46: vserial response was: b'H3\t-136881\r\n'
2024-01-30 11:17:18,012:DEBUG:vserialio:send_and_receive@46: vserial response was: b'H4\t3\r\n'
2024-01-30 11:17:18,013:DEBUG:vserialio:send_and_receive@46: vserial response was: b'H5\t0\r\n'
2024-01-30 11:17:18,015:DEBUG:vserialio:send_and_receive@46: vserial response was: b'H6\t-2994109\r\n'
2024-01-30 11:17:18,028:DEBUG:vserialio:send_and_receive@46: vserial response was: b'H7\t50422\r\n'
2024-01-30 11:17:18,029:DEBUG:vserialio:send_and_receive@46: vserial response was: b'H8\t55353\r\n'
2024-01-30 11:17:18,031:DEBUG:vserialio:send_and_receive@46: vserial response was: b'H9\t163633\r\n'
2024-01-30 11:17:18,043:DEBUG:vserialio:send_and_receive@46: vserial response was: b'H10\t2\r\n'
2024-01-30 11:17:18,044:DEBUG:vserialio:send_and_receive@46: vserial response was: b'H11\t0\r\n'
2024-01-30 11:17:18,045:DEBUG:vserialio:send_and_receive@46: vserial response was: b'H12\t0\r\n'
2024-01-30 11:17:18,046:DEBUG:vserialio:send_and_receive@46: vserial response was: b'H15\t0\r\n'
2024-01-30 11:17:18,059:DEBUG:vserialio:send_and_receive@46: vserial response was: b'H16\t0\r\n'
2024-01-30 11:17:18,061:DEBUG:vserialio:send_and_receive@46: vserial response was: b'H17\t15636\r\n'
2024-01-30 11:17:18,062:DEBUG:device:run_command@98: Send and Receive Response b'\x008\t16783\r\nChecksum\t\x16\r\nPID\t0xA389\r\nV\t53509\r\nI\t23866\r\nP\t1277\r\nCE\t-131099\r\nSOC\t548\r\nTTG\t-1\r\nAlarm\tOFF\r\nAR\t0\r\nBMV\tSmartShunt 500A/50mV\r\nFW\t0414\r\nMON\t0\r\nChecksum\t\xdf\r\nH1\t-276495\r\nH2\t-238319\r\nH3\t-136881\r\nH4\t3\r\nH5\t0\r\nH6\t-2994109\r\nH7\t50422\r\nH8\t55353\r\nH9\t163633\r\nH10\t2\r\nH11\t0\r\nH12\t0\r\nH15\t0\r\nH16\t0\r\nH17\t15636\r\n'
2024-01-30 11:17:18,063:INFO:abstractprotocol:decode@236: response passed to decode: b'\x008\t16783\r\nChecksum\t\x16\r\nPID\t0xA389\r\nV\t53509\r\nI\t23866\r\nP\t1277\r\nCE\t-131099\r\nSOC\t548\r\nTTG\t-1\r\nAlarm\tOFF\r\nAR\t0\r\nBMV\tSmartShunt 500A/50mV\r\nFW\t0414\r\nMON\t0\r\nChecksum\t\xdf\r\nH1\t-276495\r\nH2\t-238319\r\nH3\t-136881\r\nH4\t3\r\nH5\t0\r\nH6\t-2994109\r\nH7\t50422\r\nH8\t55353\r\nH9\t163633\r\nH10\t2\r\nH11\t0\r\nH12\t0\r\nH15\t0\r\nH16\t0\r\nH17\t15636\r\n'
2024-01-30 11:17:18,063:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'vedtext'
2024-01-30 11:17:18,064:DEBUG:abstractprotocol:get_command_defn@73: Found command vedtext in protocol b'VED'
2024-01-30 11:17:18,065:INFO:abstractprotocol:decode@280: Processing response of type KEYED
[... 113 lines cut ...]
2024-01-30 11:17:18,119:DEBUG:abstractprotocol:process_response@117: Got template r/100 for Amount of discharged energy b'15636'
2024-01-30 11:17:18,120:DEBUG:abstractprotocol:process_response@118: Processing data_type: float for data_name: Amount of discharged energy, raw_value b'15636'
2024-01-30 11:17:18,120:DEBUG:abstractprotocol:process_response@201: Processing format string float(raw_value)
2024-01-30 11:17:18,121:DEBUG:abstractprotocol:decode@415: Processing KEYED type responses
[... cut ...]
Traceback (most recent call last):
  File "/usr/local/bin/mppsolar", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/site-packages/mppsolar/__init__.py", line 434, in main
    op.output(
  File "/usr/local/lib/python3.11/site-packages/mppsolar/outputs/screen.py", line 85, in output
    print(f"{pad(key,maxP+1)}{value:<15}\t{unit:<4}")
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: unsupported format string passed to bytes.__format__

sante85, 2024-01-30:
@jblance can verify issue? thanks

jblance, 2024-01-31:
> ```python
> 2024-01-30 11:17:17,050:DEBUG:vserialio:send_and_receive@29: port /dev/ttyUSB0, baudrate 19200, records 30
> 2024-01-30 11:17:17,068:DEBUG:vserialio:send_and_receive@46: vserial response was: b'\x008\t16783\r\n'
> ```

I think this is the problem, the vedtext 'command' is just to listen to what the device is broadcasting and the first result can be incomplete (like in this example)

jblance, 2024-01-31:
Added test to VED protocol and I can reproduce the problem
Latest version should have a bug fix included (v 0.16.19 or later)