jblance/mpp-solar #459: PI17 - BATS command prompts error

stefanK74, 2024-01-31:
Hi,

Since a couple of days the BATS command doesn´t seem to work anymore and prompts the below response.
BATS was working previously. Other commands (PS, QS, ET) still work.

root@raspberrypi:/home/pi# mpp-solar -P PI17 -o screen -c BATS -p /dev/hidraw0
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/__init__.py", line 408, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 340, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 288, in decode
    msgs[key] = [resp_format[2][int(result)], ""]
ValueError: invalid literal for int() with base 10: ''

Any help is appreciated!
Best!

jblance, 2024-01-31:
Can you post a debug

stefanK74, 2024-02-02:
Here the debug: 

root@raspberrypi:/home/pi# root@raspberrypi:/home/pi# mpp-solar -p /dev/hidraw0 -c BATS -P PI17 -o screen -D
2024-02-02 11:35:01,905:INFO:__init__:main@218: Solar Device Command Utility, version: 0.12.5, tweak jkbleio for getInfo and to return first 300 bytes
2024-02-02 11:35:01,906:DEBUG:__init__:main@232: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xf6ee6f58>, _isConnected=False)
2024-02-02 11:35:01,906:DEBUG:__init__:main@234: udp port 5555
2024-02-02 11:35:01,906:DEBUG:__init__:main@237: Using Mongo None with mppsolar
2024-02-02 11:35:01,906:INFO:__init__:main@337: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2024-02-02 11:35:01,910:DEBUG:__init__:main@341: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2024-02-02 11:35:01,910:DEBUG:device:__init__@33: __init__ args ()
2024-02-02 11:35:01,910:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xf6ee6f58>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2024-02-02 11:35:01,911:DEBUG:__init__:get_port_type@38: port matches hidraw
2024-02-02 11:35:01,911:INFO:__init__:get_port@89: Using hidrawio for communications
2024-02-02 11:35:01,913:DEBUG:__init__:get_protocol@12: Protocol PI17
2024-02-02 11:35:01,916:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0xf68a3610>, protocol <mppsolar.protocols.pi17.pi17 object at 0xf6889f70>
2024-02-02 11:35:01,916:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xf68a3580>, 'BATS', 'BATS', 'screen', None, None)]
2024-02-02 11:35:01,916:INFO:__init__:main@395: Looping 1 commands
2024-02-02 11:35:01,917:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0xf68a3610>, protocol: <mppsolar.protocols.pi17.pi17 object at 0xf6889f70> for command: BATS, tag: BATS, outputs: screen
2024-02-02 11:35:01,917:INFO:device:run_command@274: Running command BATS
2024-02-02 11:35:01,917:INFO:pi17:get_full_command@902: Using protocol b'PI17' with 47 commands
2024-02-02 11:35:01,917:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'BATS'
2024-02-02 11:35:01,917:DEBUG:abstractprotocol:get_command_defn@62: Found command BATS in protocol b'PI17'
2024-02-02 11:35:01,917:DEBUG:pi17:get_full_command@917: _pre_cmd: b'^P005BATS'
2024-02-02 11:35:01,918:DEBUG:pi17:get_full_command@918: _prefix: ^P005
2024-02-02 11:35:01,918:DEBUG:pi17:get_full_command@925: full command: b'^P005BATS\r'
2024-02-02 11:35:01,918:INFO:device:run_command@303: full command b'^P005BATS\r' for command BATS
2024-02-02 11:35:01,918:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'BATS'
2024-02-02 11:35:01,918:DEBUG:abstractprotocol:get_command_defn@62: Found command BATS in protocol b'PI17'
2024-02-02 11:35:01,975:DEBUG:hidrawio:send_and_receive@29: length of to_send: 10
2024-02-02 11:35:01,976:DEBUG:hidrawio:send_and_receive@40: 2 chunk send
2024-02-02 11:35:04,586:DEBUG:hidrawio:send_and_receive@69: usb response was: b'^D0761500,0550,0543,0000,060,0530,0490,0536,0490,0536,0,,,1,0480,000,0050,0200c,\r'
2024-02-02 11:35:04,587:DEBUG:device:run_command@322: Send and Receive Response b'^D0761500,0550,0543,0000,060,0530,0490,0536,0490,0536,0,,,1,0480,000,0050,0200c,\r'
2024-02-02 11:35:04,588:INFO:abstractprotocol:decode@196: response passed to decode: b'^D0761500,0550,0543,0000,060,0530,0490,0536,0490,0536,0,,,1,0480,000,0050,0200c,\r'
2024-02-02 11:35:04,588:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'BATS'
2024-02-02 11:35:04,588:DEBUG:abstractprotocol:get_command_defn@62: Found command BATS in protocol b'PI17'
2024-02-02 11:35:04,589:INFO:abstractprotocol:decode@236: Processing response of type DEFAULT
2024-02-02 11:35:04,589:DEBUG:abstractprotocol:decode@240: trimmed and split responses: [b'1500', b'0550', b'0543', b'0000', b'060', b'0530', b'0490', b'0536', b'0490', b'0536', b'0', b'', b'', b'1', b'0480', b'000', b'0050', b'0200c', b'']
2024-02-02 11:35:04,589:INFO:abstractprotocol:decode@247: Processing DEFAULT type responses
2024-02-02 11:35:04,590:DEBUG:abstractprotocol:decode@278: Error resolving 0200c as int
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/__init__.py", line 408, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 340, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/protocols/abstractprotocol.py", line 288, in decode
    msgs[key] = [resp_format[2][int(result)], ""]
ValueError: invalid literal for int() with base 10: ''
root@raspberrypi:/home/pi#

Thanks!

stefanK74, 2024-02-04:
Can be closed, it seems as there was an issue with the inverter (Easun15k).
When I changed and applied float voltage using SolarPower, it seems to have fixed a glitch and I was able to read BATS without any problems and without any changes to the mpp-solar script/configuration.

closed 2024-02-04 by stefanK74