jblance/mpp-solar #460: Does mpp-solar works with new JK-BMS (using RS485 internal bus or CAN bus)?

txubelaxu, 2024-02-03:
This is the new JK BMS: https://www.jkbms.com/product/jk-pb2a16s15p-jk-pb2a16s-15p/

It has a comms board with 5 ports:

1 RS485 ((RJ45) to be connected to Inverter
1 CAN (RJ45) to be connected to Inverter
1 RS232 (RJ11)
2 RS485 (RJ45) for paralling other (same) BMS

jblance, 2024-02-14:
it might work with serial rs232/rs485 - I dont have this device so i cant test, if you try and post debug results then fixes can be made
it doesnt speak CAN yet

txubelaxu, 2024-02-17:
Well, I will give it a try but:

1. In my RS485 network there is a MASTER yet: one of the JK BMS (the one with address 0x00). Does mpp-solar's jkserial protocol work in a "listening" mode?
2. I made a small python script to capture the traffic. This is the script and the result:

```
import serial
import sys

def read_serial_port(serial_port, baud_rate):
    with serial.Serial(serial_port, baud_rate, timeout=0) as ser:
        while True:
            # Read all available bytes from the serial port
            data = ser.read_all()

            # If there is data, process and display in hexadecimal format
            if data:
                hex_data = data.hex()
                print(hex_data, end='\n')
                sys.stdout.flush()

        # If there is no more data, exit the loop
        sys.stdout.flush()

if __name__ == "__main__":
    # Serial port configuration (adjust the port and baud rate based on your setup)
    serial_port = '/dev/ttyUSB0'  # Change this to the serial port you are using
    baud_rate = 115200  # Change this to the baud rate of your device

    read_serial_port(serial_port, baud_rate)
```

This is the received data if MASTER, CLIENT and "listener script" are connected:
```
55aaeb900200120d120d110d110d110d110d110d110d110d110d110d110d110d110d120d100d0000000000000000000000000000000000000000000000000000000000000000ffff0000110d01000002530050004f004a004d004b004d004d0053004e004d004a004c004d00520051000000000000000000000000000000000000000000000000000000000000000000dd000000000014d100000000000000000000c700cd000000000000000064aa9e040080a3040000000000b30c00006400000048a3290001010000000000000000000000000000ff00010000009a030000000060543f4000000000e81400000001010100060000f2765c0000000000dd00
c700ce009a03453f09007f0000008051010000000000000000000000000000feff7fdd2f0101b0070000008c001016200001059a
55aaeb900100ac0d0000280a00005a0a0000240e0000780d000005000000790d0000500a00007a0d0000160d0000c409000050c30000030000003c000000102700002c0100003c00000005000000d0070000bc02000058020000bc0200005802000038ffffff9cffffffe8030000200300001000000001000000010000000100000080a30400dc0500007a0d0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000500000060e3160010023c3218feffffffbfe9010200000000f50010161e00016456
011016200001020000d6f1
55aaeb900205120d120d120d120d120d120d120d120d120d110d120d120d120d110d120d110d0000000000000000000000000000000000000000000000000000000000000000ffff0000120d0100000a4d004b004d004b004d004a004c0049004c004b004e004c0050004e0050004e000000000000000000000000000000000000000000000000000000000000000000ce00000000001fd100000000000000000000c000c7000000000000000064aa9e040080a3040000000000df04000064000000451f0c0001010000000000000000000000000000ff00010000009303000000000bc23f4000000000e914000000010101000600009c6d5c0000000000ce00
bf00c1008d03553e0900090000008051010000000000000000000000000000feff7fdd2f0101b00700000017
011016200001044b
0110161e0001020000d22f
55aaeb900105ac0d0000280a00005a0a00001a0e0000780d000005000000790d0000500a00007a0d0000160d0000c4090000a8610000030000003c000000400d03002c0100003c00000005000000d0070000bc02000058020000bc0200005802000038ffffff9cffffffe8030000200300001000000001000000010000000100000080a30400dc0500007a0d0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000010000000500000060e3160010023c3218feffffffbfe901020000000000
0110161e00016587
021016200001020000c201
031016200001020000cf91
041016200001020000e9a1
051016200001020000e431
061016200001020000f0c1
071016200001020000fd51
081016200001020000bca1
091016200001020000b131
0a1016200001020000a5c1
0b1016200001020000a851
0c10162000010200008e61
0d101620000102000083f1
0e10162000010200009701
0f10162000010200009a91
55aaeb900200120d120d120d120d110d110d110d110d120d110d120d110d120d110d110d110d0000000000000000000000000000000000000000000000000000000000000000ffff0000120d01000003530050004f004a004d004b004d004d0053004e004d004a004c004d00520051000000000000000000000000000000000000000000000000000000000000000000dd00000000001ad100000000000000000000c600cd000000000000000064aa9e040080a3040000000000b30c0000640000004fa3290001010000000000000000000000000000ff00010000009a030000000060543f4000000000e914000000010101000600003a775c0000000000dd00
c700ce009a034c3f09007f0000008051010000000000000000000000000000feff7fdd2f0101b007000000f0001016200001059a
55aaeb900100ac0d0000280a00005a0a0000240e0000780d000005000000790d0000500a00007a0d0000160d0000c409000050c30000030000003c000000102700002c0100003c00000005000000d0070000bc02000058020000bc0200005802000038ffffff9cffffffe8030000200300001000000001000000010000000100000080a30400dc0500007a0d0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000500000060e3160010023c3218feffffffbfe9010200000000f50010161e00016456
011016200001020000d6f1
[... cut ...]
55aaeb900100ac0d0000280a00005a0a0000240e0000780d000005000000790d0000500a00007a0d0000160d0000c409000050c30000030000003c000000102700002c0100003c00000005000000d0070000bc02000058020000bc0200005802000038ffffff9cffffffe8030000200300001000000001000000010000000100000080a30400dc0500007a0d0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000500000060e3160010023c3218feffffffbfe9010200000000f50010161e00016456

```

jblance, 2024-02-18:
this looks a lot like the jk02_32 protocol decode im work with in #275, running a 'response' (from 55aaeb till next instance of 55aaeb) through the decode Im working on (in utils/jk/py current;y) gives:
```
Container: 
    header = b'U\xaa\xeb\x90' (total 4)
    Record_Type = 2
    Record_Counter = 0
    cell_voltage_array = ListContainer: 
        3346
        3346
        3345
        3345
        3345
        3345
        3345
        3345
        3346
        3345
        3345
        3345
        3345
        3345
        3345
        3345
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
    discard1 = b'\xff\xff\x00\x00' (total 4)
    Average_Cell_Voltage = 3345
    Delta_Cell_Voltage = 1
    Current_Balancer = 768
    cell_resistance_array = ListContainer: 
        83
        80
        79
        74
        77
        75
        77
        77
        83
        78
        77
        74
        76
        77
        82
        81
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
        0
    mos_temp = 221
    discard3 = b'\x00\x00\x00\x00' (total 4)
    battery_voltage = 53527
    battery_power = 0
    battery_current = 0
    T2 = 199
    T3 = 205
    balance_current = 0
    discard9 = b'\x00\x00\x00' (total 3)
    Percent_Remain = 100
    Capacity_Remain = 302762
    Nominal_Capacity = 304000
    Cycle_Count = 0
    Cycle_Capacity = 3251
    discard7 = b'd\x00\x00\x00' (total 4)
    uptime = 2728790
    discard8 = b'\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'... (truncated, total 24)
    rest = b'\x03\x00\x00\x00\x00`T?@\x00\x00\x00\x00\xe8\x14\x00'... (truncated, total 87)
```
note most numbers need /1000 (except temperatures which are /10) from the raw decodes above

jblance, 2024-02-18:
any feedback, or script output with the correspondin app screenshot would be a huge help in getting the decode correct

txubelaxu, 2024-02-19:
https://github.com/jblance/mpp-solar/pull/465
[image]
[image]
[image]
[image]
[image]
[image]
[image]

txubelaxu, 2024-02-20:
MASTER ALONE IN THE RS485 NETWORK: (MASTER HAS ADDRESS 0x00)
```
MASTER ASKING TO ADDRESS 0x01: 011016200001020000d6f1
MASTER ASKING TO ADDRESS 0x02: 021016200001020000c201
......
MASTER ASKING TO ADDRESS 0x0e: 0e10162000010200009701
MASTER ASKING TO ADDRESS 0x0f: 0f10162000010200009a91
MASTER BROADCASTING ITS STATUS:
55aaeb900200120d120d120d110d110d120d110d110d120d110d110d110d120d120d110d110d0000000000000000000000000000000000000000000000000000000000000000ffff0000120d01000002530050004f004a004d004b004d004d0053004e004d004a004c004d00520051000000000000000000000000000000000000000000000000000000000000000000df000000000019d100000000000000000000c800ce0000000000000000642e9e040080a30400000000002f0d0000640000006a8b2a0001010000000000000000000000000000ff00010000009a030000000060543f4000000000e814000000010101000600004488650000000000df00
c900d0009a036b270a007f0000008051010000000000000000000000000000feff7fdd2f0101b00700000029001016200001059a
55aaeb900100ac0d0000280a00005a0a0000240e0000780d000005000000790d0000500a00007a0d0000160d0000c409000050c30000030000003c000000102700002c0100003c00000005000000d0070000bc02000058020000bc0200005802000038ffffff9cffffffe8030000200300001000000001000000010000000100000080a30400dc0500007a0d0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000500000060e3160010023c3218feffffffbfe9010200000000f50010161e00016456
011016200001020000d6f1
021016200001020000c201
031016200001020000cf91
041016200001020000e9a1
051016200001020000e431
061016200001020000f0c1
071016200001020000fd51
081016200001020000bca1
091016200001020000b131
0a1016200001020000a5c1
0b1016200001020000a851
0c10162000010200008e61
0d101620000102000083f1
0e10162000010200009701
0f10162000010200009a91
55aaeb900200120d120d120d110d120d110d120d110d110d110d110d110d110d120d120d110d0000000000000000000000000000000000000000000000000000000000000000ffff0000120d01000002530050004f004a004d004b004d004d0053004e004d004a004c004d00520051000000000000000000000000000000000000000000000000000000000000000000df000000000019d100000000000000000000c800ce0000000000000000642e9e040080a30400000000002f0d000064000000708b2a0001010000000000000000000000000000ff00010000009a030000000060543f4000000000e814000000010101000600008888650000000000df00
c900d0009a0371270a007f0000008051010000000000000000000000000000feff7fdd2f0101b00700000079001016200001059a
55aaeb900100ac0d0000280a00005a0a0000240e0000780d000005000000790d0000500a00007a0d0000160d0000c409000050c30000030000003c000000102700002c0100003c00000005000000d0070000bc02000058020000bc0200005802000038ffffff9cffffffe8030000200300001000000001000000010000000100000080a30400dc0500007a0d0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000500000060e3160010023c3218feffffffbfe9010200000000f50010161e00016456

```

MASTER AND 1 CLIENT IN THE RS485 NETWORK: (MASTER HAS ADDRESS 0x00, CLIENT HAS ADDRESS 0x01)
```
MASTER ASKS TO CLIENT 0x01 FOR INFO: 011016200001020000d6f1
CLIENT RESPONSE: 55aaeb900205120d120d120d120d120d120d120d120d120d110d120d120d120d110d120d110d0000000000000000000000000000000000000000000000000000000000000000ffff0000120d0100000a4d004b004d004b004d004a004c0049004c004b004e004c0050004e0050004e000000000000000000000000000000000000000000000000000000000000000000ce00000000001fd100000000000000000000c000c7000000000000000064aa9e040080a3040000000000df04000064000000451f0c0001010000000000000000000000000000ff00010000009303000000000bc23f4000000000e914000000010101000600009c6d5c0000000000ce00
bf00c1008d03553e0900090000008051010000000000000000000000000000feff7fdd2f0101b00700000017
CLIENT ANOTHER RESPONSE: 011016200001044b
MASTER ASKS TO CLIENT 0x01 FOR INFO: 0110161e0001020000d22f
55aaeb900105ac0d0000280a00005a0a00001a0e0000780d000005000000790d0000500a00007a0d0000160d0000c4090000a8610000030000003c000000400d03002c0100003c00000005000000d0070000bc02000058020000bc0200005802000038ffffff9cffffffe8030000200300001000000001000000010000000100000080a30400dc0500007a0d0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000010000000500000060e3160010023c3218feffffffbfe901020000000000
CLIENT ANOTHER RESPONSE: 0110161e00016587
021016200001020000c201
031016200001020000cf91
041016200001020000e9a1
051016200001020000e431
061016200001020000f0c1
071016200001020000fd51
081016200001020000bca1
091016200001020000b131
0a1016200001020000a5c1
0b1016200001020000a851
0c10162000010200008e61
0d101620000102000083f1
0e10162000010200009701
0f10162000010200009a91
55aaeb900200120d120d120d120d110d110d110d110d120d110d120d110d120d110d110d110d0000000000000000000000000000000000000000000000000000000000000000ffff0000120d01000003530050004f004a004d004b004d004d0053004e004d004a004c004d00520051000000000000000000000000000000000000000000000000000000000000000000dd00000000001ad100000000000000000000c600cd000000000000000064aa9e040080a3040000000000b30c0000640000004fa3290001010000000000000000000000000000ff00010000009a030000000060543f4000000000e914000000010101000600003a775c0000000000dd00
[... cut ...] 00000000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000010000000500000060e3160010023c3218feffffffbfe901020000000000
0110161e00016587
021016200001020000c201
031016200001020000cf91
041016200001020000e9a1
051016200001020000e431
061016200001020000f0c1
071016200001020000fd51
081016200001020000bca1
091016200001020000b131
0a1016200001020000a5c1
0b1016200001020000a851
0c10162000010200008e61
0d101620000102000083f1
0e10162000010200009701
0f10162000010200009a91
```

jblance, 2024-02-22:
latest verision (in v 0.16.23+) of jk02_32 decodes the responses, not sure if it will work with the serial ports but worth a try
post a debug if there are errors
note: still a bunch of data that i cant guess what it relates to - the jk.py script in utils provides the current decode and is the easiest way to try additional decodes

txubelaxu, 2024-02-22:
jk02_32_definition fields depend on what is the value for Record_Type.
The fields in actual jk02_32_definition are OK when Record_Type=2. STATUS
When Record_Type=1 the info is relating to the CONFIG values of the BMS, so field names are wrong for this case.

I can confirm too that the starting "01" "02" "03" is the address of the client. So, 011016200001020000d6f1 means MASTER is ASKING to client addressed 0x01 for information. I changed my client to address 0x03 and client answered when MASTER sent 031016200001020000cf91 message and there were no response to 011016200001020000d6f1.

txubelaxu, 2024-02-22:
```
Container: 
    header = b'U\xaa\xeb\x90' (total 4)
    Record_Type = 1    <<<<<<<<<<<<<<<<<<<<<< CONFIG INFORMATION
    Record_Counter = 5
    cell_voltage_array = ListContainer: 
        3500                   <<<<<<<<<<<<<<<<<<<<<< VOLTAGE SMART SLEEP
        0
        2600                   <<<<<<<<<<<<<<<<<<<<<< CELL UNDER VOLTAGE PROTECTION (UVP)
        0
        2650                   <<<<<<<<<<<<<<<<<<<<<< CELL UNDER VOLTAGE PROTECTION RECOVERY (UVPR)
        0
        3610                   <<<<<<<<<<<<<<<<<<<<<< CELL OVER VOLTAGE PROTECTION (OVP)
        0
        3448                   <<<<<<<<<<<<<<<<<<<<<< CELL OVER VOLTAGE PROTECTION RECOVERY (OVPR)
        0
        5                         <<<<<<<<<<<<<<<<<<<<<< BALANCE TRIGGER VOLTAGE (??)
        0
        3449                   <<<<<<<<<<<<<<<<<<<<<< SOC-100% VOLTAGE
        0
        2640                   <<<<<<<<<<<<<<<<<<<<<< SOC-0% VOLTAGE
        0
        3450                   <<<<<<<<<<<<<<<<<<<<<< VOLTAGE CELL REQUEST CHARGE VOLTAGE (RCV) OR SMART BALANCE VOLTAGE ??????????????????
        0
        3350                   <<<<<<<<<<<<<<<<<<<<<< VOLTAGE CELL REQUEST FLOAT VOLTAGE (RFV)
        0
        2500                   <<<<<<<<<<<<<<<<<<<<<< POWER OFF VOLTAGE
        0
        25000
        0
        3                         <<<<<<<<<<<<<<<<<<<<<< CHARGE OCP DELAY(s) ?????????????
        0
        60                       <<<<<<<<<<<<<<<<<<<<<< CHARGE OCPR TIME (s) ????????????????
        0
        3392
        3
        300                     <<<<<<<<<<<<<<<<<<<<<< DISCHARGE OCP DELAY (s) ????????????????
        0
    discard1 = b'<\x00\x00\x00' (total 4)
    Average_Cell_Voltage = 5 <<<<<<<<<<<<<<<<<<<<<< 
    Delta_Cell_Voltage = 0
    Current_Balancer = 2000    <<<<<<<<<<<<<<<<<<<<<< MAX BALANCE CURRENT (mA) 
    cell_resistance_array = ListContainer: 
        0
        700                                   <<<<<<<<<<<<<<<<<<<<<< charge OTP (dºC)
        0
        600                                   <<<<<<<<<<<<<<<<<<<<<< charge OTPR (dºC)
        0
        700                                   <<<<<<<<<<<<<<<<<<<<<< discharge OTP (dºC)
        0
        600                                   <<<<<<<<<<<<<<<<<<<<<< discharge OTPR (dºC)
        0
        65336
        65535
        65436
        65535
        1000                                  <<<<<<<<<<<<<<<<<<<<<< MOS OTP (dºC)
        0
        800                                    <<<<<<<<<<<<<<<<<<<<<< MOS OTPR (dºC)
        0
        16
        0
        1
        0
        1
        0
        1
        0
        41856
        4
        1500                  <<<<<<<<<<<<<<<<<<<<<< SCP DELAY (us)
        0
        3450                  <<<<<<<<<<<<<<<<<<<<<< START BALANCE VOLT (V) ????????????????
        0
        0
    mos_temp = 0
    discard3 = b'\x00\x00\x00\x00' (total 4)
    battery_voltage = 0
    battery_power = 0
    battery_current = 0
    T2 = 0
    T3 = 0
    balance_current = 0
    discard9 = b'\x00\x00\x00' (total 3)
    Percent_Remain = 0
    Capacity_Remain = 0
    Nominal_Capacity = 0
    Cycle_Count = 0
    Cycle_Capacity = 0
    discard7 = b'\x00\x00\x00\x00' (total 4)
    uptime = 0
    discard8 = b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'... (truncated, total 24)
    rest = b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'... (truncated, total 35)

```
I am going to change some values in my BMS to confirm. :-)

txubelaxu, 2024-02-22:
```
Container: 
    header = b'U\xaa\xeb\x90' (total 4)
    Record_Type = 1    <<<<<<<<<<<<<<<<<<<<<< CONFIG INFORMATION BELOW
    Record_Counter = 5
    cell_voltage_array = ListContainer: 
        3500                   <<<<<<<<<<<<<<<<<<<<<< VOLTAGE SMART SLEEP [CONFIRMED]
        0
        2600                   <<<<<<<<<<<<<<<<<<<<<< CELL UNDER VOLTAGE PROTECTION (UVP) [CONFIRMED]
        0
        2650                   <<<<<<<<<<<<<<<<<<<<<< CELL UNDER VOLTAGE PROTECTION RECOVERY (UVPR)  [CONFIRMED]
        0
        3610                   <<<<<<<<<<<<<<<<<<<<<< CELL OVER VOLTAGE PROTECTION (OVP)  [CONFIRMED]
        0
        3448                   <<<<<<<<<<<<<<<<<<<<<< CELL OVER VOLTAGE PROTECTION RECOVERY (OVPR)  [CONFIRMED]
        0
        5                         <<<<<<<<<<<<<<<<<<<<<< BALANCE TRIGGER VOLTAGE (mA)  [CONFIRMED]
        0
        3449                   <<<<<<<<<<<<<<<<<<<<<< SOC-100% VOLTAGE  [CONFIRMED]
        0
        2640                   <<<<<<<<<<<<<<<<<<<<<< SOC-0% VOLTAGE  [CONFIRMED]
        0
        3450                   <<<<<<<<<<<<<<<<<<<<<< VOLTAGE CELL REQUEST CHARGE VOLTAGE (RCV)  [CONFIRMED]
        0
        3350                   <<<<<<<<<<<<<<<<<<<<<< VOLTAGE CELL REQUEST FLOAT VOLTAGE (RFV)  [CONFIRMED]
        0
        2500                   <<<<<<<<<<<<<<<<<<<<<< POWER OFF VOLTAGE  [CONFIRMED]
        0
        25000                 <<<<<<<<<<<<<<<<<<<<<< CONTINUED CHARGE CURRENT (mA)  [CONFIRMED]
        0
        3                         <<<<<<<<<<<<<<<<<<<<<< CHARGE OCP DELAY(s)  [CONFIRMED]
        0
        60                       <<<<<<<<<<<<<<<<<<<<<< CHARGE OCPR TIME (s)  [CONFIRMED]
        0
        3392                   <<<<<<<<<<<<<<<<<<<<<< ????????????????
        3                         <<<<<<<<<<<<<<<<<<<<<< ????????????????
        300                     <<<<<<<<<<<<<<<<<<<<<< DISCHARGE OCP DELAY (s)  [CONFIRMED]
        0
    discard1 = b'<\x00\x00\x00' (total 4)
    Average_Cell_Voltage = 5 <<<<<<<<<<<<<<<<<<<<<< SCPR Time (s) [CONFIRMED]
    Delta_Cell_Voltage = 0
    Current_Balancer = 2000    <<<<<<<<<<<<<<<<<<<<<< MAX BALANCE CURRENT (mA)  [CONFIRMED]
    cell_resistance_array = ListContainer: 
        0
        700                                   <<<<<<<<<<<<<<<<<<<<<< charge OTP (dºC) [CONFIRMED]
        0
        600                                   <<<<<<<<<<<<<<<<<<<<<< charge OTPR (dºC) [CONFIRMED]
        0
        700                                   <<<<<<<<<<<<<<<<<<<<<< discharge OTP (dºC) [CONFIRMED]
        0
        600                                   <<<<<<<<<<<<<<<<<<<<<< discharge OTPR (dºC) [CONFIRMED]
        0
        65336                               <<<<<<<<<<<<<<<<<<<<<< charge UTP [CONFIRMED] 65336<>-20.0ºC
        65535                               
        65436                               <<<<<<<<<<<<<<<<<<<<< charge UTPR [CONFIRMED] 65436<>-10.0ºC
        65535
        1000                                  <<<<<<<<<<<<<<<<<<<<<< MOS OTP (dºC) [CONFIRMED]
        0
        800                                    <<<<<<<<<<<<<<<<<<<<<< MOS OTPR (dºC) [CONFIRMED]
        0
        16                                      <<<<<<<<<<<<<<<<<<<<<< CELL COUNT. [UNCONFIRMED]                                   
        0
        1
        0
        1
        0
        1
        0
        41856
        4
        1500                  <<<<<<<<<<<<<<<<<<<<<< SCP DELAY (us)  [CONFIRMED]
        0
        3450                  <<<<<<<<<<<<<<<<<<<<<< START BALANCE VOLT (V) [CONFIRMED]
        0
        0
    mos_temp = 0
    discard3 = b'\x00\x00\x00\x00' (total 4)
    battery_voltage = 0
    battery_power = 0
    battery_current = 0
    T2 = 0
    T3 = 0
    balance_current = 0
    discard9 = b'\x00\x00\x00' (total 3)
    Percent_Remain = 0
    Capacity_Remain = 0
    Nominal_Capacity = 0
    Cycle_Count = 0
    Cycle_Capacity = 0
    discard7 = b'\x00\x00\x00\x00' (total 4)
    uptime = 0
    discard8 = b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'... (truncated, total 24)
    rest = b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'... (truncated, total 35)
```

jblance, 2024-02-22:
so the serial data is continuous? ie you dont need to send a command (and get a response)

if so then I'd guess we need to build the data into full responses and have the decode distinguish between the different responses?
The record type 1 or 2 or 3 should be doable, though I'd probably do it in powermon not mppsolar as powermon is more flexible
Are the short 'lines' (eg ``061016200001020000f0c1``) useful or just master / slave comms?

jblance, 2024-02-22:
updated jk.py - added record type 1 definition
```python
record_01 = cs.Struct(
    "Record_Counter" / cs.Byte,
    "voltage_smart_sleep" / cs.Int32ul,
    "cell_under_voltage_protection" / cs.Int32ul,
    "cell_under_voltage_protection_recovery" / cs.Int32ul,
    "cell_over_voltage_protection" / cs.Int32ul,
    "cell_over_voltage_protection_recovery" / cs.Int32ul,
    "balance_trigger_voltage" / cs.Int32ul,
    "soc_100%_voltage" / cs.Int32ul,
    "soc_0%_voltage" / cs.Int32ul,
    "cell_request_charge_voltage" / cs.Int32ul,
    "cell_request_float_voltage" / cs.Int32ul,
    "power_off_cell_voltage" / cs.Int32ul,
    "continued_charge_current" / cs.Int32ul,
    "charge_ocp_delay_sec" / cs.Int32ul,
    "charge_ocpr_time_sec" / cs.Int32ul,
    "unknown2" / cs.Int32ul,
    "discharge_ocp_delay_sec" / cs.Int32ul,
    "discharge_ocpr_time_sec" / cs.Int32ul,
    "scpr_time_sec" / cs.Int32ul,
    "max_balance_current" / cs.Int32ul,
    "charge_otp_c" / cs.Int32ul,
    "charge_otpr_c" / cs.Int32ul,
    "discharge_otp_c" / cs.Int32ul,
    "discharge_otpr_c" / cs.Int32ul,
    "charge_utp_c" / cs.Int32sl,
    "charge_utpr_c" / cs.Int32sl,
    "mos_otp_c" / cs.Int32ul,
    "mos_otpr_c" / cs.Int32ul,
    "cell_count" / cs.Int32ul,
    "unknown7" / cs.Bytes(4),
    "unknown8" / cs.Bytes(4),
    "unknown9" / cs.Bytes(4),
    "unknown10" / cs.Int32ul,
    "scp_delay_us" / cs.Int32ul,
    "start_balance_voltage" / cs.Int32ul,
    "unknown11" / cs.Bytes(8),
    "unknown12" / cs.Bytes(8),
    "unknown13" / cs.Bytes(8),
    "unknown14" / cs.Bytes(8),
    "unknown15" / cs.Bytes(8),

    "rest" / cs.GreedyBytes,
)
```
gave:
```
Container: 
    header = b'U\xaa\xeb\x90' (total 4)
    record_type = 1
    record_decode = Container: 
        Record_Counter = 0
        voltage_smart_sleep = 3500
        cell_under_voltage_protection = 2600
        cell_under_voltage_protection_recovery = 2650
        cell_over_voltage_protection = 3620
        cell_over_voltage_protection_recovery = 3448
        balance_trigger_voltage = 5
        soc_100%_voltage = 3449
        soc_0%_voltage = 2640
        cell_request_charge_voltage = 3450
        cell_request_float_voltage = 3350
        power_off_cell_voltage = 2500
        continued_charge_current = 50000
        charge_ocp_delay_sec = 3
        charge_ocpr_time_sec = 60
        unknown2 = 10000
        discharge_ocp_delay_sec = 300
        discharge_ocpr_time_sec = 60
        scpr_time_sec = 5
        max_balance_current = 2000
        charge_otp_c = 700
        charge_otpr_c = 600
        discharge_otp_c = 700
        discharge_otpr_c = 600
        charge_utp_c = -200
        charge_utpr_c = -100
        mos_otp_c = 1000
        mos_otpr_c = 800
        cell_count = 16
        unknown7 = b'\x01\x00\x00\x00' (total 4)
        unknown8 = b'\x01\x00\x00\x00' (total 4)
        unknown9 = b'\x01\x00\x00\x00' (total 4)
        unknown10 = 304000
        scp_delay_us = 1500
        start_balance_voltage = 3450
        unknown11 = b'\x00\x00\x00\x00\x00\x00\x00\x00' (total 8)
        unknown12 = b'\x00\x00\x00\x00\x00\x00\x00\x00' (total 8)
        unknown13 = b'\x00\x00\x00\x00\x00\x00\x00\x00' (total 8)
        unknown14 = b'\x00\x00\x00\x00\x00\x00\x00\x00' (total 8)
        unknown15 = b'\x00\x00\x00\x00\x00\x00\x00\x00' (total 8)
        rest = b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'... (truncated, total 126)
```

based on python construct logic:
```python
jk02_32_definition = cs.Struct(
    "header" / cs.Bytes(4),
    "record_type" / cs.Byte,
    "record_decode" / cs.Switch(cs.this.record_type, {1: record_01, 2: record_02, 3: cs.GreedyBytes}),
)
```

unknown10 might be capacity ?
unknown7-9 kind looks like flags?

txubelaxu, 2024-02-23:
confirmed: unknown10 is the capacity of the whole battery below BMS.
I had a bug in my script. The messages arrived fragmented. I supossed 2 messages where was only 1. I suppose the USB/SERIAL->RS485 device has a limit in its buffer. I have corrected in my just uploaded script. So there was missing data in my previous messages.

`so the serial data is continuous? ie you dont need to send a command (and get a response)`
It is a RS485 network where multiple JK-BMS are connected. It is used by JK to internal communicate the BMSs working in the system. So, one of the BMS in the system is the "MASTER" BMS (the one has the 0x00 RS485 address). All the other BMSs are slaves, and each one has a differente RS485 address.
The definition of RS485 does not allow multiple MASTER devices, so, we can only "listen to" the network. But the information flowing is very interesting: all the status data of each node and the config data of each node. 

`if so then I'd guess we need to build the data into full responses and have the decode distinguish between the different responses?`
Yes, each message has the RS485 address of the sender. I have detected.

`The record type 1 or 2 or 3 should be doable, though I'd probably do it in powermon not mppsolar as powermon is more flexible.`
Well. I have now 1 ESP32 (via bluetooth) for each JK-BMS using [esphome-jk-bms](https://github.com/syssi/esphome-jk-bms) So, I have 2 ESP32 working and bluetooth unavailable to use my phone to connect via bluetooth... These ESP32 are integrated in my Home Assistant system. Working well. So, my target is to get this information sniffing this info directly from RS485 network. It is how I discover mpp-solar... 
`Are the short 'lines' (eg 061016200001020000f0c1) useful or just master / slave comms?`
short lines are info from master to network. "06" is the destination RS485 address. one SLAVE only speaks if master asks to it.

It is inside script.py as well:
                 ## CONFIG (RECORD TYPE=1)
                 print('RS485 ADDRESS:' + str(result['rest'][49]))
                 print('CHARGE_SWITCH_ENABLED:' + str(result['cell_resistance_array'][19]))
                 print('DISCHARGE_SWITCH_ENABLED:' + str(result['cell_resistance_array'][21]))
                 print('BALANCE_SWITCH_ENABLED:' + str(result['cell_resistance_array'][23]))
                 print('HEATING_SWITCH_ENABLED:' + str(get_bit_value(result['rest'][61],0)))
                 print('DISABLE_TEMP_SENSOR_SWITCH_ENABLED:' + str(get_bit_value(result['rest'][61],1)))
                 print('DISPLAY_ALWAYS_ON_SWITCH_ENABLED:' + str(get_bit_value(result['rest'][61],4)))
                 print('SMART_SLEEP_ON_SWITCH_ENABLED:' + str(get_bit_value(result['rest'][61],6)))
                 print('DISABLE_PCL_MODULE_SWITCH_ENABLED:' + str(get_bit_value(result['rest'][61],7)))
                 print('TIMED_STORED_DATA_SWITCH_ENABLED:' + str(get_bit_value(result['rest'][62],0)))
                 print('CHARGING_FLOAT_MODE_SWITCH_ENABLED:' + str(get_bit_value(result['rest'][62],1)))
                 EMERGENCY_SWITCH_ENABLED     [I have not tested... lol]

                 ## STATUS (RECORD TYPE=2)
                 print("CHARGE WORKING:" + str((result['discard8'][1])))
                 print("DISCHARGE WORKING:" + str((result['discard8'][2])))
                 print("PREDISCHARGE WORKING:" + str((result['discard8'][3])))       [NOT CONFIRMED]
                 print("BALANCE WORKING:" + str((result['discard8'][4])))                   [NOT CONFIRMED]

txubelaxu, 2024-02-25:
Here is a table with those balues, I think: [esphome-jk-bms](https://github.com/syssi/esphome-jk-bms/blob/7466ae3e64a97b3b94e8937b4d5ed9eb9a9da555/docs/pb2a16s20p/JK_BMS.RS485.Modbus.v1_0.pdf)

txubelaxu, 2024-02-27:
> so the serial data is continuous? ie you dont need to send a command (and get a response)
>
> if so then I'd guess we need to build the data into full responses and have the decode distinguish between [...]

I have just tested powermon changing the content of conf.yaml with a lot of combinations with no success.
Feel free, @jblance, to ask me for testing any special powermon configuration with my system.

jrventer, 2024-03-01:
I have managed to map the Record 2 which is the 0x1200 registers I will work on the mapping of the Record1 0x1000 registers:
Below the updated struct record2 for jk.py
```
record_02 = cs.Struct(
    "Record_Counter" / cs.Byte,
    "cell_voltage_array" / cs.Array(32, cs.Int16ul),
    "cell_presence" / cs.BitStruct("cells" / cs.Array(32, cs.Enum(cs.Bit, not_present=0, present=1))),
    "CellVolAve" / cs.Int16ul,
    "CellVdifMax" / cs.Int16ul,
    "MaxVolCellNbr" / cs.Byte,
    "MinVolCellNbr" / cs.Byte,
    "cell_resistance_array" / cs.Array(32, cs.Int16ul),
    "TempMos" / cs.Int16ul,
    "CellWireResStatus" / cs.BitStruct("cells:" / cs.Array(32, cs.Enum(cs.Bit, OK=0, ALARM=1))),
    "BatVoltage" / cs.Int32ul,
    "BatWatt" / cs.Int32ul,
    "BatCurrent" / cs.Int32sl,
    "TempBat1" / cs.Int16ul,
    "TempBat2" / cs.Int16ul,
    "Alarms" / cs.BitStruct(
        AlarmWireRes=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmMosOTP=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmCellQuantity=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmCurSensorErr=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmCellOVP=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmBatOVP=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmChOCP=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmChSCP=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmChOTP=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmChUTP=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmCPUAuxCommuErr=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmCellUVP=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmBatUVP=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmDchOCP=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmDchSCP=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmDchOTP=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmChargeMOS=cs.Enum(cs.Bit, OK=0, ALARM=1),
        AlarmDischargeMOS=cs.Enum(cs.Bit, OK=0, ALARM=1),
        GPSDisconneted=cs.Enum(cs.Bit, OK=0, ALARM=1),
        ModifyPWDinTime=cs.Enum(cs.Bit, OK=0, ALARM=1),
        DischargeOnFailed=cs.Enum(cs.Bit, OK=0, ALARM=1),
        BatteryOverTemp=cs.Enum(cs.Bit, OK=0, ALARM=1),
        TemperatureSensorAnomaly=cs.Enum(cs.Bit, OK=0, ALARM=1),
        PLCModuleAnomaly=cs.Enum(cs.Bit, OK=0, ALARM=1),
        UnusedBit24=cs.Enum(cs.Bit, OK=0, ALARM=1),
        UnusedBit25=cs.Enum(cs.Bit, OK=0, ALARM=1),
        UnusedBit26=cs.Enum(cs.Bit, OK=0, ALARM=1),
        UnusedBit27=cs.Enum(cs.Bit, OK=0, ALARM=1),
        UnusedBit28=cs.Enum(cs.Bit, OK=0, ALARM=1),
        UnusedBit29=cs.Enum(cs.Bit, OK=0, ALARM=1),
        UnusedBit30=cs.Enum(cs.Bit, OK=0, ALARM=1),
        UnusedBit31=cs.Enum(cs.Bit, OK=0, ALARM=1)
    ),
    "BalanCurrent" / cs.Int16sl,
    "BalanStatus" /  cs.Enum(cs.Int8ul, off=0, charge=1, discharge=2),
    "SOCStateOfCharge" / cs.Int8ul,
    "SOCCapRemain" / cs.Int32ul,
    "SOCFullChargeCap" / cs.Int32ul,
    "SOCCycleCount" / cs.Int32ul,
    "SOCCycleCap" / cs.Int32ul,
    "SOCSOH" / cs.Int8ul,
    "Precharge" /  cs.Enum(cs.Int8ul, closed=0, open=1),
    "UserAlarm" / cs.Int16sl,
    "Runtime" / cs.Int32ul,
    "Charge" /  cs.Enum(cs.Int8ul, closed=0, open=1),
    "Discharge" /  cs.Enum(cs.Int8ul, closed=0, open=1),
    "UserAlarm2" / cs.Int16sl,
    "TimeDcOCPR" / cs.Int16ul,
    "TimeDcSCPR" / cs.Int16ul,
    "TimeCOCPR" / cs.Int16ul,
[... 251 lines cut ...]
        BatDisCurCorrect = 2.3693574824343304e-38
        VolChargCur = 0
        VolDischargCur = 256
        BatVol = 0
        HeatCurrent = 56982
        RVD = 100
        ChargerPlugged = (enum) connected 1
        SysRunTicks = 0
        TempBat3 = 285
        TempBat4 = 278
        TempBat5 = 274
        RTCTicks = 2104886157
        TimeEnterSleep = 15402963
        PCLModuleSta = (enum) closed 0
        Reseved268RVD = b'\x00\x80Q\x01\x00\x00\x00\x00' (total 8)```

Also from what I see the master sends a write register then the slave sends the data following with a write register response. So should be feasible to identify the dat for the different slaves and master.
Write Register details below:
- address 5662(0x161e) = 0x1000 configuration rw registers
- address 5664(0x1620) = 0x1200 status read only holding registers

```
2024-02-28 16:24:04,150 Master  -> ID: 1, Write Multiple registers: 0x10, Write address: 5664, Write quantity: 1, Write data: [00 00]
2024-02-28 16:24:04,550 Warning : Ignoring data: [55 aa eb 90 02 05 e9 0c e9 0c e8 0c e9 0c e9 0c e9 0c e9 0c e9 0c e9 0c e9 0c e9 0c e9 0c e9 0c e9 0c e9 0c e8 0c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 e9 0c 01 00 00 01 44 00 41 00 43 00 40 00 43 00 41 00 43 00 41 00 44 00 42 00 44 00 41 00 44 00 43 00 45 00 43 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1d 01 00 00 00 00 94 ce 00 00 37 8f 03 00 c5 ee ff ff 13 01 16 01 00 00 00 00 00 00 00 61 63 66 03 00 70 82 03 00 08 00 00 00 84 eb 1c 00 64 00 00 00 92 21 37 00 01 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff 00 01 00 00 00 97 03 00 00 0c 00 15 4f 3f 40 00 00 00 00 a8 14 00 00 00 01 01 01 00 06 00 00 4e de 64 01 00 00 00 00 1d 01 16 01 12 01 8d 03 6f 7d d3 07 eb 00 00 00 80 51 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 fe ff 7f dd 2f 01 01 b0 07 00 00 00 93]
2024-02-28 16:24:04,551 Slave   -> ID: 1, Write Multiple registers: 0x10, Write address: 5664, Write quantity: 1```

txubelaxu, 2024-03-04:
OK... more data discovered.
[image]

Frame size is 308 bytes. Counting from 0 to 307:
- CHECKSUM at position 299 
- ADDRESS at position 300

Warning! is CHECKSUM not CRC. So, function is:
```uint16_t chksum(const uint8_t data[], const uint16_t len) {
  uint16_t checksum = 0;
  for (uint16_t i = 0; i < len; i++) {
    checksum = checksum + data[i];
  }
  return checksum;
}
```

txubelaxu, 2024-03-04:
You have here: https://github.com/txubelaxu/esphome-jk-bms working code about it. It is not 100% working, because I have to improve the addressing, but that code is getting info from frames, reusing BLE code + MODBUS code

txubelaxu, 2024-03-11:
See --> https://github.com/txubelaxu/esphome-jk-bms/blob/main/components/jk_rs485_bms/README.md

txubelaxu, 2024-03-12:
"charge_ocp_delay_sec" / cs.Int32ul,
    "charge_ocpr_time_sec" / cs.Int32ul,
    "Continued Discharge Current" / cs.Int32ul,   <--------------
    "discharge_ocp_delay_sec" / cs.Int32ul,
    "discharge_ocpr_time_sec" / cs.Int32ul,

txubelaxu, 2024-04-09:
Until now, I had got ESP acting as a "sniffer". There is one user that have tested (previous version) with 1 MASTER and 6 SLAVES.

Well, I have got now the ESP working as master. https://github.com/txubelaxu/esphome-jk-bms/blob/main/components/jk_rs485_bms/README.md

So, there is only one thing to do: speaking from ESP to inverter. lol

If you need something from me for mpp-solar project, ask for it with no problem. Anything that I could help.

closed 2024-04-09 by txubelaxu