jblance/mpp-solar #461: Protocol issue for JK-B2A8S20P-H with heating port

the-duke, 2024-02-08:
I like to use a JK-B2A8S20P BMS with heating port over a serial connection. The hardware setup seems to work fine (USB-Serial-Adapter connected to the 4-pin GPS-Port). Wenn ist send 4E 57 00 13 00 00 00 00 06 03 00 00 00 00 00 00 68 00 00 01 29 hex sequence via CuteCom to the BMS i receive hex data with the custom BMS Name inside.

But the "jkbms -p /dev/ttyUSB0 -P JKV11 -b 115200 -c getInfo -D" command returns parsing errors:
```
$ jkbms -p /dev/ttyUSB0 -P JKV11 -b 115200 -c getInfo -D
2024-02-08 13:28:37,451:INFO:__init__:main@205: Solar Device Command Utility, version: 0.16.19, python version: 3.11.6
2024-02-08 13:28:37,452:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-02-08 13:28:37,452:DEBUG:__init__:main@240: MqttBroker name: localhost, port: 1883, user: None
2024-02-08 13:28:37,452:DEBUG:__init__:main@242: udp port 5555
2024-02-08 13:28:37,452:DEBUG:__init__:main@244: Using Postgres None
2024-02-08 13:28:37,452:DEBUG:__init__:main@247: Using Mongo None with mppsolar
2024-02-08 13:28:37,452:INFO:__init__:main@358: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB0 (porttype=None)" using protocol "JKV11"
2024-02-08 13:28:37,453:DEBUG:__init__:main@362: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2024-02-08 13:28:37,453:DEBUG:device:__init__@34: __init__ args ()
2024-02-08 13:28:37,453:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'JKV11', 'baud': 115200, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7f9a2c581410>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway'}
2024-02-08 13:28:37,453:DEBUG:__init__:get_port_type@69: port matches ttyusb
2024-02-08 13:28:37,453:INFO:__init__:get_port@111: Using serialio for communications
2024-02-08 13:28:37,455:DEBUG:__init__:get_protocol@13: Protocol JKV11
2024-02-08 13:28:37,539:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0x7f9a2c581e50>, protocol JKBMS BLE communication protocol handler software v11.x
2024-02-08 13:28:37,539:DEBUG:__init__:main@408: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x7f9a2c9b8e50>, 'getInfo', 'getInfo', 'screen', None, None)]
2024-02-08 13:28:37,539:INFO:__init__:main@413: Looping 1 commands
2024-02-08 13:28:37,539:INFO:__init__:main@423: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x7f9a2c581e50>, protocol: JKBMS BLE communication protocol handler software v11.x for command: getInfo, tag: getInfo, outputs: screen
2024-02-08 13:28:37,539:INFO:device:run_command@51: Running command getInfo
2024-02-08 13:28:37,539:INFO:jkabstractprotocol:get_full_command@74: Using protocol b'JKv11' with 3 commands
2024-02-08 13:28:37,539:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getInfo
2024-02-08 13:28:37,539:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getInfo'
2024-02-08 13:28:37,539:DEBUG:abstractprotocol:get_command_defn@73: Found command getInfo in protocol b'JKv11'
2024-02-08 13:28:37,539:DEBUG:jkabstractprotocol:get_full_command@79: self._command = getInfo
2024-02-08 13:28:37,539:DEBUG:jkabstractprotocol:get_full_command@90: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-02-08 13:28:37,539:DEBUG:jkabstractprotocol:get_full_command@98: cmd with command code: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-02-08 13:28:37,539:DEBUG:jkabstractprotocol:get_full_command@100: cmd with crc: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11')
2024-02-08 13:28:37,539:INFO:device:run_command@81: full command bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11') for command getInfo
2024-02-08 13:28:37,539:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getInfo
2024-02-08 13:28:37,539:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getInfo'
2024-02-08 13:28:37,539:DEBUG:abstractprotocol:get_command_defn@73: Found command getInfo in protocol b'JKv11'
2024-02-08 13:28:37,539:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 115200
2024-02-08 13:28:37,541:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2024-02-08 13:28:38,641:DEBUG:serialio:send_and_receive@30: serial response was: b''
2024-02-08 13:28:38,647:DEBUG:device:run_command@98: Send and Receive Response b''
2024-02-08 13:28:38,647:INFO:abstractprotocol:decode@236: response passed to decode: b''
2024-02-08 13:28:38,647:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getInfo
2024-02-08 13:28:38,647:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getInfo'
2024-02-08 13:28:38,647:DEBUG:abstractprotocol:get_command_defn@73: Found command getInfo in protocol b'JKv11'
2024-02-08 13:28:38,647:INFO:abstractprotocol:decode@280: Processing response of type POSITIONAL
2024-02-08 13:28:38,647:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 4, 'Header', '']
2024-02-08 13:28:38,647:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 1, 'Record Type', '']
2024-02-08 13:28:38,647:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Int', 1, 'Record Counter', '']
2024-02-08 13:28:38,647:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Ascii', 16, 'Device Model', '']
2024-02-08 13:28:38,647:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Ascii', 8, 'Hardware Version', '']
2024-02-08 13:28:38,647:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Ascii', 8, 'Software Version', '']
2024-02-08 13:28:38,647:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['uptime', 4, 'Up Time', '']
2024-02-08 13:28:38,647:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Int', 4, 'Power-on Times', '']
2024-02-08 13:28:38,647:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Ascii', 16, 'Device Name', '']
2024-02-08 13:28:38,647:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Ascii', 16, 'Device Passcode', '']
2024-02-08 13:28:38,647:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Ascii', 8, 'Manufacturing Date', '']
2024-02-08 13:28:38,647:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Ascii', 11, 'Serial Number', '']
2024-02-08 13:28:38,648:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Ascii', 5, 'Passcode', '']
2024-02-08 13:28:38,648:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Ascii', 16, 'User Data', '']
2024-02-08 13:28:38,648:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Ascii', 16, 'Setup Passcode', '']
2024-02-08 13:28:38,648:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['discard', 672, 'unknown', '']
2024-02-08 13:28:38,648:DEBUG:abstractprotocol:decode@284: trimmed and split responses: [b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'']
2024-02-08 13:28:38,648:DEBUG:abstractprotocol:decode@485: Processing POSITIONAL type responses
2024-02-08 13:28:38,648:DEBUG:abstractprotocol:decode@512: Got defn ['Hex2Str', 4, 'Header', '']
2024-02-08 13:28:38,648:DEBUG:abstractprotocol:process_response@118: Processing data_type: Hex2Str for data_name: Header, raw_value b''
2024-02-08 13:28:38,648:DEBUG:abstractprotocol:process_response@201: Processing format string Hex2Str(raw_value)
2024-02-08 13:28:38,648:DEBUG:protocol_helpers:Hex2Str@104: Hex b'' decoded to 
2024-02-08 13:28:38,648:DEBUG:abstractprotocol:decode@485: Processing POSITIONAL type responses
2024-02-08 13:28:38,648:DEBUG:abstractprotocol:decode@512: Got defn ['Hex2Str', 1, 'Record Type', '']
2024-02-08 13:28:38,648:DEBUG:abstractprotocol:process_response@118: Processing data_type: Hex2Str for data_name: Record Type, raw_value b''
2024-02-08 13:28:38,648:DEBUG:abstractprotocol:process_response@201: Processing format string Hex2Str(raw_value)
[... 11 lines cut ...]
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/theduke/.local/lib/python3.11/site-packages/mppsolar/devices/device.py", line 115, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
  File "/home/theduke/.local/lib/python3.11/site-packages/mppsolar/protocols/abstractprotocol.py", line 203, in process_response
    r = eval(format_string)
        ^^^^^^^^^^^^^^^^^^^
  File "<string>", line 1, in <module>
  File "/home/theduke/.local/lib/python3.11/site-packages/mppsolar/protocols/protocol_helpers.py", line 90, in Hex2Int
    answer = hexString[0]
             ~~~~~~~~~^^^
IndexError: index out of range

```

[image]
[image]
[image]
[image]

Are there any hints to fix the protocol for my BMS. It may be related
 to the special software version for the heat port status and heat-current?

the-duke, 2024-02-08:
Ok, i found a solution that works for me in the issue https://github.com/jblance/mpp-solar/issues/389 by editing the jkserial protocol. I deleted all other cell response messages and left only the cells 1-4. now the result data seems to be correct.

jblance, 2024-02-08:
Reopening as reminder / tracking
Solution needs a variable length decode

jblance, 2024-02-12:
> passed to decode: b''

Yeah the jkserial protocol is the one needed - but I need to work out how to make it deal with variable length
Can you post a BMS response in text form so I can use it for testing (effectively what was in the screenshot from cutecom.log or a debug run using the jkserial protocol)
Thanks

jblance, 2024-02-14:
implemented in dev branch in powermon using construct library

closed 2024-02-14 by jblance

the-duke, 2024-02-14:
Of course [cutecom.log](https://github.com/jblance/mpp-solar/files/14279229/cutecom.log)  

I check out the branch jk-variable-length to give it a try

jblance, 2024-02-14:
thanks
powermon serial probably wont work - I need to adjust it to work for jk serial connections

jblance, 2024-02-14:
might work now with latest commit in branch - can you try and post a debug