jblance/mpp-solar #477: Wont run on raspbian bullseye.

DannySkjodt1, 2024-03-09:
$ mpp-solar
Exception ignored in: <function Client.__del__ at 0x76006610>
Traceback (most recent call last):
  File "/home/danny/.local/lib/python3.9/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/home/danny/.local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/home/danny/.local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
AttributeError: 'Client' object has no attribute '_sock'
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
  File "/home/danny/.local/lib/python3.9/site-packages/mppsolar/__init__.py", line 230, in main
    mqtt_broker = MqttBroker(
  File "/home/danny/.local/lib/python3.9/site-packages/mppsolar/libs/mqttbrokerc.py", line 44, in __init__
    self.mqttc = mqtt_client.Client()
TypeError: __init__() missing 1 required positional argument: 'callback_api_version'

jblance, 2024-03-09:
What version are you using?
Later and current versions require python 3.10 or later

fableman73, 2024-03-10:
root@husb:~# python -V
Python 3.12.2
root@husb:~# python3 -V
Python 3.12.2
root@husb:~# mpp-solar -p C8:47:80:07:3E:36 -c getInfo
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 5, in <module>
    from mppsolar import main
ModuleNotFoundError: No module named 'mppsolar'

fableman73, 2024-03-10:
reinstalled using python -m pip 

Now get:

root@husb:~# mpp-solar -p C8:47:80:07:3E:36 -c getInfo
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.12/site-packages/mppsolar/__init__.py", line 426, in main
    results = _device.run_command(command=_command)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/mppsolar/devices/device.py", line 92, in run_command
    raw_response = self._port.send_and_receive(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/mppsolar/inout/jkbleio.py", line 38, in send_and_receive
    record_type = command_defn["record_type"]
                  ~~~~~~~~~~~~^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable

jblance, 2024-03-10:
add `-D` to the command to get debug output and post that

DannySkjodt1, 2024-03-13:
> What version are you using? Later and current versions require python 3.10 or later
> […](#)

Yes this can be closed, upgrading to python 3.12 fixed it.

closed 2024-03-13 by DannySkjodt1