jblance/mpp-solar #485: Can't set charge voltages on MPPSolar MPI 15K

danieltroger, 2024-04-13:
Hi, I have the 15KW variant of https://www.mppsolar.com/v3/mpi12-15kwwpip65/

I'm mostly communicating with it with the PI17 protocol and many commands work fine. This is the device ID:
```
ubuntu@ubuntu:~$ mpp-solar -p /dev/hidraw0 -n 'BPVInverter2' -T 'BPVInverter2' -P PI17 -o screen --getDeviceId
Command: Get Device ID - Generate a device id
--------------------------------------------------------------------------------
Parameter Value                 Unit
deviceid  17:061                    
--------------------------------------------------------------------------------
```

However, it doesn't respect the bulk charging voltage - it directly goes to the float charging voltage when charging. Therefore I wanted to write my own script that sets the float charging voltage to 58.4v until little current is flowing and then sets it back down to 53v. However, the command doesn't seem to work.

I've tried a multitude of protocols but can't get it working. I'm running a raspberry pi zero 2w connected via USB - would appreciate some help.

Log:
```
ubuntu@ubuntu:~$ mpp-solar -p /dev/hidraw0 -P PI17  -D  -c MCHGV0584,0530 
2024-04-13 15:30:34,414:INFO:__init__:main@205: Solar Device Command Utility, version: 0.16.29, python version: 3.10.12
2024-04-13 15:30:34,415:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-04-13 15:30:34,416:DEBUG:__init__:main@240: MqttBroker name: localhost, port: 1883, user: None
2024-04-13 15:30:34,416:DEBUG:__init__:main@242: udp port 5555
2024-04-13 15:30:34,417:DEBUG:__init__:main@244: Using Postgres None
2024-04-13 15:30:34,417:DEBUG:__init__:main@247: Using Mongo None with mppsolar
2024-04-13 15:30:34,418:INFO:__init__:main@358: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2024-04-13 15:30:34,427:DEBUG:__init__:main@362: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2024-04-13 15:30:34,428:DEBUG:device:__init__@34: __init__ args ()
2024-04-13 15:30:34,428:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0xffffa6d113f0>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway'}
2024-04-13 15:30:34,429:DEBUG:__init__:get_port_type@39: port matches hidraw
2024-04-13 15:30:34,430:INFO:__init__:get_port@93: Using hidrawio for communications
2024-04-13 15:30:34,435:DEBUG:__init__:get_protocol@13: Protocol PI17
2024-04-13 15:30:35,406:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0xffffa6d11300>, protocol PI17 protocol handler
2024-04-13 15:30:35,407:DEBUG:__init__:main@408: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xffffa6d11390>, 'MCHGV0584,0530', 'MCHGV0584,0530', 'screen', None, None)]
2024-04-13 15:30:35,408:INFO:__init__:main@413: Looping 1 commands
2024-04-13 15:30:35,408:INFO:__init__:main@423: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0xffffa6d11300>, protocol: PI17 protocol handler for command: MCHGV0584,0530, tag: MCHGV0584,0530, outputs: screen
2024-04-13 15:30:35,409:INFO:device:run_command@51: Running command MCHGV0584,0530
2024-04-13 15:30:35,410:INFO:pi17:get_full_command@896: Using protocol b'PI17' with 48 commands
2024-04-13 15:30:35,410:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'MCHGV0584,0530'
2024-04-13 15:30:35,411:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EY
2024-04-13 15:30:35,412:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EM
2024-04-13 15:30:35,414:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: ED
2024-04-13 15:30:35,415:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EH
2024-04-13 15:30:35,417:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: LON
2024-04-13 15:30:35,419:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PA
2024-04-13 15:30:35,420:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PB
2024-04-13 15:30:35,421:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PC
2024-04-13 15:30:35,423:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PD
2024-04-13 15:30:35,424:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PE
2024-04-13 15:30:35,425:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PF
2024-04-13 15:30:35,426:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: DAT
2024-04-13 15:30:35,428:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: LST
2024-04-13 15:30:35,429:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDA
2024-04-13 15:30:35,431:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDB
2024-04-13 15:30:35,432:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDC
2024-04-13 15:30:35,433:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDD
2024-04-13 15:30:35,434:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDE
2024-04-13 15:30:35,436:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDF
2024-04-13 15:30:35,437:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDG
2024-04-13 15:30:35,438:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: BT
2024-04-13 15:30:35,439:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: MCHGV
2024-04-13 15:30:35,441:DEBUG:abstractprotocol:get_command_defn@81: Matched: MCHGV0584,0530 to: MCHGV value: 0584,0530
2024-04-13 15:30:35,441:DEBUG:pi17:get_full_command@937: _pre_cmd: b'^S015MCHGV0584,0530'
2024-04-13 15:30:35,442:DEBUG:pi17:get_full_command@944: full command: b'^S015MCHGV0584,0530\r'
2024-04-13 15:30:35,442:INFO:device:run_command@81: full command b'^S015MCHGV0584,0530\r' for command MCHGV0584,0530
2024-04-13 15:30:35,443:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'MCHGV0584,0530'
2024-04-13 15:30:35,443:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EY
2024-04-13 15:30:35,444:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EM
2024-04-13 15:30:35,444:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: ED
2024-04-13 15:30:35,445:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EH
[... 46 lines cut ...]
2024-04-13 15:30:36,026:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDF
2024-04-13 15:30:36,027:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EDG
2024-04-13 15:30:36,027:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: BT
2024-04-13 15:30:36,028:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: MCHGV
2024-04-13 15:30:36,028:DEBUG:abstractprotocol:get_command_defn@81: Matched: MCHGV0584,0530 to: MCHGV value: 0584,0530
2024-04-13 15:30:36,029:INFO:abstractprotocol:decode@280: Processing response of type DEFAULT
2024-04-13 15:30:36,029:DEBUG:abstractprotocol:decode@284: trimmed and split responses: ['NAK']
2024-04-13 15:30:36,030:INFO:abstractprotocol:decode@291: Processing DEFAULT type responses
2024-04-13 15:30:36,030:DEBUG:abstractprotocol:decode@308: result NAK, key Command execution, resp_format ['ack', 'Command execution', {'NAK': 'Failed', 'ACK': 'Successful'}]
2024-04-13 15:30:36,031:INFO:device:run_command@116: Decoded response {'_command': 'MCHGV0584,0530', '_command_description': 'Set battery charge voltages', 'raw_response': ['^0\x1bã\r', ''], 'WARNING0': ['Command MCHGV0584,0530 was rejected', '']}
2024-04-13 15:30:36,031:DEBUG:__init__:main@427: results: {'_command': 'MCHGV0584,0530', '_command_description': 'Set battery charge voltages', 'raw_response': ['^0\x1bã\r', ''], 'WARNING0': ['Command MCHGV0584,0530 was rejected', '']}
2024-04-13 15:30:36,032:INFO:__init__:get_output@40: attempting to create output processor: screen
2024-04-13 15:30:36,037:DEBUG:screen:__init__@17: processor.screen __init__ args: (), kwargs: {}
2024-04-13 15:30:36,038:DEBUG:__init__:main@433: Using output filter: None
2024-04-13 15:30:36,038:INFO:screen:output@23: Using output processor: screen
[... cut ...]
2024-04-13 15:30:36,039:DEBUG:screen:output@68: displayData: {'warning0': ['Command MCHGV0584,0530 was rejected', '']}
Command: MCHGV0584,0530 - Set battery charge voltages
--------------------------------------------------------------------------------
Parameter Value                 Unit
warning0  Command MCHGV0584,0530 was rejected       
--------------------------------------------------------------------------------

2024-04-13 15:30:36,040:DEBUG:__init__:main@459: Not daemon, so not looping
```

danieltroger, 2024-04-13:
Sometimes it also gives me this random result, maybe because I have the mppsolar service running at the same time?

But even with it stopped I can't set the charging voltages.

```
ubuntu@ubuntu:~$ sudo  mpp-solar -P PI17  -p /dev/hidraw0  -c 'MCHGV0584,0528'
Command: MCHGV0584,0528 - Set battery charge voltages
--------------------------------------------------------------------------------
Parameter                    Value              Unit
unknown_value_in_response_1  00095                  
unknown_value_in_response_2  00119                  
unknown_value_in_response_3  00787                  
unknown_value_in_response_4  005                    
unknown_value_in_response_5  1                      
unknown_value_in_response_6  1                      
unknown_value_in_response_7  1                      
unknown_value_in_response_8  1                      
unknown_value_in_response_9  2                      
unknown_value_in_response_10 2                      
--------------------------------------------------------------------------------

```

jblance, 2024-04-16:
still looks like somethign else is running - as a set command should return that much info

jblance, 2024-04-16:
can you try another setting command - eg set date/time
set battery charge voltages might be blocked by eg battery type setting (i know this was true for earlier pi30 inverters)

jblance, 2024-04-17:
i see you are trying to set it the same as it is set, can you try a slightly different settings?
I only know what I can get from the documentation and people have tested (so it is possible there is a different format for your particular inverter)

danieltroger, 2024-04-18:
Yeah, it doesn't work with a different voltage either

<details><summary>Expand log</summary>

```
ubuntu@ubuntu:~$ systemctl --user stop mpp-solar
ubuntu@ubuntu:~$ systemctl --user status mpp-solar
○ mpp-solar.service - MPP Solar Service
     Loaded: loaded (/etc/xdg/systemd/user/mpp-solar.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Thu 2024-04-18 08:37:50 UTC; 1s ago
    Process: 42021 ExecStart=/usr/bin/python3 /usr/local/bin/mpp-solar -C /etc/mpp-solar/mpp-solar.conf --daemon (code=killed, signal=TERM)
   Main PID: 42021 (code=killed, signal=TERM)
        CPU: 22.547s

Apr 18 08:37:37 ubuntu python3[42021]: Getting results from device: mppsolar device - name: Inverter1_MPI15K, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0xffffb4fd29e0>, protocol: PI17 prot>
Apr 18 08:37:40 ubuntu python3[42021]: Getting results from device: mppsolar device - name: Inverter1_MPI15K, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0xffffb4fd29e0>, protocol: PI17 prot>
Apr 18 08:37:43 ubuntu python3[42021]: Sleeping for 0 sec
Apr 18 08:37:43 ubuntu python3[42021]: Getting results from device: mppsolar device - name: Inverter1_MPI15K, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0xffffb4fd29e0>, protocol: PI17 prot>
Apr 18 08:37:45 ubuntu python3[42021]: Getting results from device: mppsolar device - name: Inverter1_MPI15K, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0xffffb4fd29e0>, protocol: PI17 prot>
Apr 18 08:37:48 ubuntu python3[42021]: Sleeping for 0 sec
Apr 18 08:37:48 ubuntu python3[42021]: Getting results from device: mppsolar device - name: Inverter1_MPI15K, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0xffffb4fd29e0>, protocol: PI17 prot>
Apr 18 08:37:50 ubuntu systemd[761]: Stopping MPP Solar Service...
Apr 18 08:37:50 ubuntu systemd[761]: Stopped MPP Solar Service.
Apr 18 08:37:50 ubuntu systemd[761]: mpp-solar.service: Consumed 22.547s CPU time.
ubuntu@ubuntu:~$ ps axu|grep mpp
ubuntu     42144  0.0  0.4   6416  1844 pts/0    S+   08:38   0:00 grep --color=auto mpp
ubuntu@ubuntu:~$ mpp-solar -p /dev/hidraw0 -P PI17  -D  -c MCHGV0570,0510
2024-04-18 08:38:31,988:INFO:__init__:main@205: Solar Device Command Utility, version: 0.16.29, python version: 3.10.12
2024-04-18 08:38:31,988:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-04-18 08:38:31,989:DEBUG:__init__:main@240: MqttBroker name: localhost, port: 1883, user: None
2024-04-18 08:38:31,990:DEBUG:__init__:main@242: udp port 5555
2024-04-18 08:38:31,990:DEBUG:__init__:main@244: Using Postgres None
2024-04-18 08:38:31,991:DEBUG:__init__:main@247: Using Mongo None with mppsolar
2024-04-18 08:38:31,991:INFO:__init__:main@358: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2024-04-18 08:38:32,002:DEBUG:__init__:main@362: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2024-04-18 08:38:32,002:DEBUG:device:__init__@34: __init__ args ()
2024-04-18 08:38:32,003:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0xffff9e8a2830>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway'}
2024-04-18 08:38:32,004:DEBUG:__init__:get_port_type@39: port matches hidraw
2024-04-18 08:38:32,004:INFO:__init__:get_port@93: Using hidrawio for communications
2024-04-18 08:38:32,010:DEBUG:__init__:get_protocol@13: Protocol PI17
2024-04-18 08:38:32,992:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0xffff9e8a2890>, protocol PI17 protocol handler
2024-04-18 08:38:32,993:DEBUG:__init__:main@408: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xffff9e8a27d0>, 'MCHGV0570,0510', 'MCHGV0570,0510', 'screen', None, None)]
2024-04-18 08:38:32,993:INFO:__init__:main@413: Looping 1 commands
2024-04-18 08:38:32,994:INFO:__init__:main@423: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0xffff9e8a2890>, protocol: PI17 protocol handler for command: MCHGV0570,0510, tag: MCHGV0570,0510, outputs: screen
2024-04-18 08:38:32,995:INFO:device:run_command@51: Running command MCHGV0570,0510
2024-04-18 08:38:32,995:INFO:pi17:get_full_command@896: Using protocol b'PI17' with 48 commands
2024-04-18 08:38:32,996:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'MCHGV0570,0510'
2024-04-18 08:38:32,996:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EY
2024-04-18 08:38:32,998:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EM
2024-04-18 08:38:33,000:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: ED
2024-04-18 08:38:33,001:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: EH
2024-04-18 08:38:33,003:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: LON
2024-04-18 08:38:33,004:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PA
2024-04-18 08:38:33,006:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PB
2024-04-18 08:38:33,007:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PC
2024-04-18 08:38:33,008:DEBUG:abstractprotocol:get_command_defn@77: Regex commands _command: PD
[... cut ...]
Ok, interesting to know that you've gotten here without reverse engineering and only with docs. Do you have any docs for this specific inverter? In your experience, will MPPSolar reply with docs when asked for this? I assume they're quite unresponsive because the [premature float bug](http://forums.aeva.asn.au/viewtopic.php?title=pip4048ms-inverter&p=59896&t=4332#p59896) is still not fixed, years later. Where did you get the docs that you have?

Worst case I assume one would have to sniff the communication between the windows app and the inverter, but I have no experience with doing that :(

jblance, 2024-04-18:
The docs are here https://github.com/jblance/mpp-solar/tree/master/docs/protocols
The docs have come from a variety of places and people (randomly) I have never managed to get any from mppsolar directly
If you can sniff any app the can set the voltages on the serial port we can figure out what is being sent

PaulEPop, 2024-06-03:
Changing voltages via /dev/hidraw wouldn't possible.
I have testing 'hidraw' = rejected
testing with 'ttyUSB0'  = 'ACK'
;)

danieltroger, 2024-06-03:
Ahh, I see @PaulEPop, thanks for the information. I wonder how the WatchPower/SolarPower PC application does it then, because that one can set the voltages via USB…

ttyUSB0 is serial, right? Would love to hear how you connect the inverter to the serial port on a raspberry pi

PaulEPop, 2024-06-03:
ttyUSB0 is serial, correct.
To your question.. I have used a USB2RS232 adapter, it was still laying around.

jblance, 2024-06-03:
can anyone sniffer the watchpower app setting the voltages via a 'hidraw' connection (ie direct USB without USB to RS232 adapter)

as it works via serial, the command must be correct - i expect it is to do with the breaking up of the send into multiple chunks to send via hidraw, but I thought the current approach had resolved the issue (obviously not)

unfortunately I dont have an inverter of that uses this protocol to test against

riogrande75, 2024-06-06:
@jblance MPI 15K needs solarpower app, watchpower app is for offgrid devices only (axpert,etc.).

@danieltroger Can you set the float charging voltage to 58,3V via SolarPower (windows)? Does it get stored correctly in memory and does it remain for some minutes? If you have BMS connected (via BMS-Box or Card) then this settings get overwritten in inverter's config.

BTW: Latest protocol description for hybrid inverters I got from voltronic can be found here: 
[Infini-Solar_10KW%252615KW_protocol_20230129.xlsx](https://www.mediafire.com/file/41fri58lu77bio0/Infini-Solar_10KW%252615KW_protocol_20230129.xlsx)

JakobTewes, 2025-02-12:
To me [this](https://github.com/syssi/esphome-pipsolar/issues/156#issuecomment-2561986502) looks like an option for "lookup tables"

danieltroger, 2025-10-18:
> @danieltroger Can you set the float charging voltage to 58,3V via SolarPower (windows)? Does it get stored correctly in memory and does it remain for some minutes? If you have BMS connected (via BMS-Box or Card) then this settings get overwritten in inverter's config.

Sorry for the late reply. I don't have BMS connected, I have my own solution for that.
I don't use SolarPower (windows) unless I absolutely have to. But when setting float charging voltage to 58,4v via the SolarPower iOS app, it persists and works fine.

Or at least used to - setting and reading values via the SolarPower app stopped working at `2025-10-16T01:35:44.269Z`, with the error `{ err: 2, desc: 'ERR_TIMEOUT' }`. However, just switching the API URL to https://www.dessmonitor.com/ made it work again, see https://github.com/danieltroger/mpi-15k-controller/commit/6f3ccd8726c5cf6b4e5e1abaefeb205a8c8b166e

Super cool with all the new commands people have reverse-engineered for PI17! I was ready to start configuring values via USB instead of the Taiwanese servers but setting charging voltage via USB still doesn't work :( I bought this USB to serial cable https://www.kjell.com/se/produkter/dator/kablar-adaptrar/seriell/luxorparts-konsolkabel-usb-a-p65935 but at 2400 baud rate I cannot get any communication at all with the inverter :( Don't know why.

However, with dessmonitor.com I have control again for now.

danieltroger, 2025-10-18:
> Can you set the float charging voltage to 58,3V via SolarPower (windows)?

Okay so I actually tried this now and can confirm that mpp-solar sends the command "correctly" over USB - SolarPower on windows CANNOT SET THE CHARGE VOLTAGE either lol. They have the same bug. (And SolarPower on windows also can't communicate with my inverter via serial). So right now the only way to do it programmatically for me is dessmonitor.com