jblance/mpp-solar #488: Non-programmer here.  I think something is amiss with paho-mqtt, mqttbrokerc.py and / or callbackapiversion. I'm flailing in the dark...

bobcov, 2024-04-20:
sudo mpp-solar --getDeviceId
Exception ignored in: <function Client.del at 0x7fd951822ee0>
Traceback (most recent call last):
File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 874, in del
self._reset_sockets()
File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
self._sock_close()
File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 1119, in _sock_close
if not self._sock:
AttributeError: 'Client' object has no attribute '_sock'
Traceback (most recent call last):
File "/usr/local/bin/mpp-solar", line 8, in
sys.exit(main())
File "/usr/local/lib/python3.8/dist-packages/mppsolar/init.py", line 230, in main
mqtt_broker = MqttBroker(
File "/usr/local/lib/python3.8/dist-packages/mppsolar/libs/mqttbrokerc.py", line 44, in init
self.mqttc = mqtt_client.Client()
TypeError: init() missing 1 required positional argument: 'callback_api_version

jblance, 2024-04-21:
the minimum python version is 3.10
the bold is to do with the markdown syntax (which is what is used for git issues check [the docs](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax)
`--getDeviceId` is really only useful for a real inverter
for testing try
`mppsolar -p test -c QPIGS` etc
see [the wiki](https://github.com/jblance/mpp-solar/wiki/Detailed-Usage) for more info on usage

bobcov, 2024-04-22:
Thank you very much for your guidance. I just made a system snapshot in preparation to upgrade from 3.8.10 to 3.11.  Should I return line 44 to it's original or leave the modification which got the mpp-solar commands to work?
Original line 44: self.mqttc = mqtt_client.Client()
Changed line 44: self.mqttc = mqtt_client.Client(mqtt_client.CallbackAPIVersion.VERSION2)
I got the errors shown in my first post before I changed the line (I have no idea if that was the right change, but it seems to work).

Will I need to install everything again after upgrading?
It seems that the inverter, Steca  Solarix MPPT 5020 understands the default protocol, but some values are in the wrong place or not defined.  There is a 12 volt 100amh battery attached for testing communications, nothing else is attached.
sudo` mpp-solar -p /dev/ttyUSB0 -c QPIGS
Command: QPIGS - General Status Parameters inquiry
--------------------------------------------------------------------------------
Parameter                       Value          	Unit
ac_input_voltage               12.0           	V   	{'icon': 'mdi:lightning'}
ac_input_frequency            12.49          	Hz  
ac_output_voltage               0.0            	V   
ac_output_frequency            0.0            	Hz  
ac_output_apparent_power   0              	VA  
ac_output_active_power        0              	W   
ac_output_load                     19             	%   
bus_voltage                           0              	V   
battery_voltage                     0.0            	V   
battery_charging_current       0              	A   
battery_capacity                    0              	%   
inverter_heat_sink_temperature 63             	°C  
pv_input_current_for_battery   0.0            	A   
pv_input_voltage               0.0            	V   
battery_voltage_from_scc       0              	V 

Here is the command you said to try:
mpp-solar -p test -c QPIGS
Command: QPIGS - General Status Parameters inquiry
--------------------------------------------------------------------------------
Parameter                                   Value          	Unit
ac_input_voltage                            0.0            	V   	{'icon': 'mdi:lightning'}
ac_input_frequency                          0.0            	Hz  
ac_output_voltage                           230.0          	V   
ac_output_frequency                         49.9           	Hz  
ac_output_apparent_power                    161            	VA  
ac_output_active_power                      119            	W   
ac_output_load                              3              	%   
bus_voltage                                 460            	V   
battery_voltage                             57.5           	V   
battery_charging_current                    12             	A   
battery_capacity                            100            	%   
inverter_heat_sink_temperature              69             	°C  
pv_input_current_for_battery                14.0           	A   
pv_input_voltage                            103.8          	V   
battery_voltage_from_scc                    57.45          	V   
battery_discharge_current                   0              	A   
is_sbu_priority_version_added               0              	bool
is_configuration_changed                    0              	bool
is_scc_firmware_updated                     1              	bool
is_load_on                                  1              	bool
is_battery_voltage_to_steady_while_charging 0              	bool
is_charging_on                              1              	bool
is_scc_charging_on                          1              	bool
is_ac_charging_on                           0              	bool
rsv1                                        0              	A   
rsv2                                        0              	A   
pv_input_power                              856            	W   
is_charging_to_float                        0              	bool
is_switched_on                              1              	bool
is_reserved                                 0              	bool

Where can I find the instructions on how to use the test files? I would like to get all of the graphing functions working, but I don
t have any real data because the inverter is not attached to anything.

Thank you again for your help. I don't have PayPal but if you are in Europe, I can donate by bank transfer or credit card (but not PayPal processing) in appreciation of your work and time.

bobcov, 2024-04-24:
User error

closed 2024-04-24 by bobcov