jblance/mpp-solar #489: Voltronic Infinisolar V II WP 5Kw  no data read

legolas3107, 2024-04-26:
Hello not being able to read any data from my Infinisolar V II WP 5Kw inverter I asked the store for the protocol used and I am attaching the pdf. I don't think there is in your mop-solar you could implement? Or maybe I'm the one who can't and can you help me? Thank you
[Infini-Solar.V.VII.VIII.VIV.V.II.6K.LV.Split-Phase.TWIN.protocol.202205.xlsx](https://github.com/jblance/mpp-solar/files/15131613/Infini-Solar.V.VII.VIII.VIV.V.II.6K.LV.Split-Phase.TWIN.protocol.202205.xlsx)

jblance, 2024-04-28:
what commands have you tried?

jblance, 2024-04-28:
at a quick glance this looks like PI18 protocol

legolas3107, 2024-04-29:
[image]
I wanted to find the protocol but it’s 0 the result. can you tell me how I can try?

Saentist, 2024-04-29:
@legolas3107 do you have firmware to see with commands are supported?

legolas3107, 2024-04-29:
I don’t have a firmware, I don’t know what you mean but I use the program Solarpower for pc

jblance, 2024-04-29:
try
`mpp-solar -p /dev/ttyUSB2 -P PI18 -c PI`
and/or
`mpp-solar -p /dev/ttyUSB2 -P PI17 -c PI`

legolas3107, 2024-05-02:
[image]
this is the result

legolas3107, 2024-05-02:
the inverter is connected via rs232 to usb adapter to the pc reads and changes the parameters from the program in windows, but no results as you see in the screen , I have to connect it directly to the usb?

jblance, 2024-05-02:
if the windows program is working at the same time it wont work (only 1 thing can access the serial port at once)
add the debug flag `-D` to the commands to see more info
try protocol `-P PI30` again (with the debug)

legolas3107, 2024-05-03:
root@raspberrypi4:/data/etc/dbus-mppsolar/mpp-solar# mpp-solar -p /dev/ttyUSB2 -P PI30 -c PI -D
2024-05-03 13:43:19,088:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2024-05-03 13:43:19,089:INFO:__init__:main@339: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB2 (porttype=None)" using protocol "PI30"
2024-05-03 13:43:19,096:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2024-05-03 13:43:19,097:DEBUG:device:__init__@31: __init__ args ()
2024-05-03 13:43:19,097:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB2', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2024-05-03 13:43:19,098:DEBUG:device:get_port_type@86: port matches ttyusb
2024-05-03 13:43:19,098:INFO:device:set_port@160: Using serialio for communications
2024-05-03 13:43:19,126:DEBUG:device:set_protocol@96: Protocol PI30
2024-05-03 13:43:19,184:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0xb6852da8>, protocol <mppsolar.protocols.pi30.pi30 object at 0xb6852d48>
2024-05-03 13:43:19,185:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb6852d00>, 'PI', 'PI', 'screen', None, None)]
2024-05-03 13:43:19,185:INFO:__init__:main@395: Looping 1 commands
2024-05-03 13:43:19,186:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0xb6852da8>, protocol: <mppsolar.protocols.pi30.pi30 object at 0xb6852d48> for command: PI, tag: PI, outputs: screen
2024-05-03 13:43:19,186:INFO:device:run_command@270: Running command PI
2024-05-03 13:43:19,187:INFO:abstractprotocol:get_full_command@31: Using protocol b'PI30' with 36 commands
2024-05-03 13:43:19,187:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'PI'
2024-05-03 13:43:19,188:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0xd7 0x52 0xd752
2024-05-03 13:43:19,188:DEBUG:abstractprotocol:get_full_command@38: full command: b'PI\xd7R\r'
2024-05-03 13:43:19,189:INFO:device:run_command@296: full command b'PI\xd7R\r' for command PI
2024-05-03 13:43:19,189:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'PI'
2024-05-03 13:43:19,189:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: F
2024-05-03 13:43:19,191:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: MCHGC
2024-05-03 13:43:19,192:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: MNCHGC
2024-05-03 13:43:19,194:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: MUCHGC
2024-05-03 13:43:19,195:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PBCV
2024-05-03 13:43:19,197:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PBDV
2024-05-03 13:43:19,199:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PBFT
2024-05-03 13:43:19,200:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PBT
2024-05-03 13:43:19,202:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PCP
2024-05-03 13:43:19,204:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PCVV
2024-05-03 13:43:19,205:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PE
2024-05-03 13:43:19,207:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PD
2024-05-03 13:43:19,208:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PGR
2024-05-03 13:43:19,209:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: POP
2024-05-03 13:43:19,211:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: POPM
2024-05-03 13:43:19,212:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PPCP
2024-05-03 13:43:19,214:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PPVOKC
2024-05-03 13:43:19,215:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PSDV
2024-05-03 13:43:19,217:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PSPB
2024-05-03 13:43:19,219:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: QPGS
2024-05-03 13:43:19,220:INFO:abstractprotocol:get_command_defn@57: No command_defn found for PI
2024-05-03 13:43:19,221:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB2, baudrate 2400
2024-05-03 13:43:19,227:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2024-05-03 13:43:19,336:WARNING:serialio:send_and_receive@33: Serial read error: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2024-05-03 13:43:19,337:INFO:serialio:send_and_receive@34: Command execution failed
2024-05-03 13:43:19,337:DEBUG:device:run_command@315: Send and Receive Response {'ERROR': ['Serial command execution failed', '']}
2024-05-03 13:43:19,338:DEBUG:__init__:main@409: results: {'ERROR': ['Serial command execution failed', '']}
2024-05-03 13:43:19,338:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2024-05-03 13:43:19,343:DEBUG:screen:__init__@16: processor.screen __init__ kwargs {}
2024-05-03 13:43:19,344:DEBUG:__init__:main@415: Using output filter: None
2024-05-03 13:43:19,344:INFO:screen:output@19: Using output processor: screen
2024-05-03 13:43:19,345:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['Serial command execution failed', '']}, 'tag': 'PI', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
error                           Serial command execution failed
2024-05-03 13:43:19,345:DEBUG:__init__:main@435: Not daemon, so not looping

jblance, 2024-05-06:
> 2024-05-03 13:43:19,336:WARNING:serialio:send_and_receive@33: Serial read error: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

This is the key point
Is anything else reading from the serial port?
Is this the correct port?

Saentist, 2024-05-07:
post result of this commands
```
lsusb
dmesg | egrep tty
```

legolas3107, 2024-05-08:
root@raspberrypi4:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 004: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 003: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@raspberrypi4:~# dmesg | egrep tty
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=D8:3A:DD:94:F4:9C vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 root=/dev/mmcblk0p3 rootwait
[    1.491868] fe201000.serial: ttyAMA0 at MMIO 0xfe201000 (irq = 36, base_baud = 0) is a PL011 rev2
[    1.501035] printk: console [ttyS0] disabled
[    1.501170] fe215040.serial: ttyS0 at MMIO 0xfe215040 (irq = 38, base_baud = 62500000) is a 16550
[    2.822679] printk: console [ttyS0] enabled
[    5.340633] usb 1-1.1: FTDI USB Serial Device converter now attached to ttyUSB0
[    5.397419] usb 1-1.2: FTDI USB Serial Device converter now attached to ttyUSB1
[    5.437535] usb 1-1.3: FTDI USB Serial Device converter now attached to ttyUSB2
root@raspberrypi4:~#

legolas3107, 2024-05-08:
there are other 2 usb connected of victron system with proprietary cables in raspberry that work fine

legolas3107, 2024-05-08:
data are not read in the pi4 with venusOS system and in the linux test pc

legolas3107, 2024-05-08:
I hope with your help to succeed in solving TNKS

Saentist, 2024-05-08:
No need to post lot of messages just edit old one 
and also use <> button for citrated terminal content

why not try all protocols until see some result
or maby your usb dongle is not supported

jblance, 2024-05-08:
> there are other 2 usb connected of victron system with proprietary cables in raspberry that work fine

what is reading these devices?
its not a protocol issue yet - something else is also connecting to the serial port
try `sudo fuser -v /dev/ttyUSB2`

legolas3107, 2024-05-09:
[image]
does not give me option -V

jblance, 2024-05-09:
There you go, the victron software is accessing the serial port, so you
won't get valid responses if you try to access it with mpp-solar at the
same time

legolas3107, 2024-05-09:
> <img alt="Screenshot 2024-05-02 155657" width="401" src="https://private-user-images.githubusercontent.com/78864739/327456480-462c9287-5c31-411b-b76a-17c4fdc734ce.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUyNDE1NjEsIm5iZiI6MTcxNTI0MTI2MSwicGF0aCI6Ii83ODg2NDczOS8zMjc0NTY0ODAtNDYyYzkyODctNWMzMS00MTFiLWI3NmEtMTdjNGZkYzczNGNlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTA5VDA3NTQyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE1NmQxNjg0YjNhYTBkYzA2ZTJmZGU3OWI4NDM1MTE0NjQxMjljYzYyY2Y0MzdjZTQwMTNhYjA4ZWYwNTBhMGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.CHUCpVaYzArVnnfgO5NP5HyYuUY7COzIKNXlfLa7Ii8"> this is the result
This test was done by the Raspberry in production but also in the Linux test PC where there is no Victron and I only connect the USB there is no response with any protocol that you see in the screenshot

jblance, 2024-05-09:
I can't see the screenshot

legolas3107, 2024-05-09:
[image]

jblance, 2024-05-09:
> 327456480-462c9287-5c31-411b-b76a-17c4fdc734ce.png (view on web)

Does the fuser command show anything for this device?
Also try protocol pi30

jblance, 2024-05-09:
Also is the port the same on this device?

legolas3107, 2024-05-09:
if we can read the inverter to the linux test pc then we can integrate it into the victron system " DarkZeros/dbus-mppsolar project "... on the pc test there is only usb0 connected in the day I send you the screen of the result of fuser

legolas3107, 2024-05-10:
[image]

legolas3107, 2024-05-10:
the command sometimes does not work but finally now from the test pc we have the result. thank you very much

jblance, 2024-05-12:
great news

closed 2024-05-12 by jblance