jblance/mpp-solar #496: wrong output if jk04 and jkv11 used in config (different proto per section)

mfu-mcosys, 2024-06-15:
if the config looks like this:
```
[JKBMS_UNTEN]
type=jkbms
protocol=jk04
port=C8:47:8C:E5:CC:9F
command=getCellData
outputs=screen

[JKBMS_BOX1_UNTEN] 
type=jkbms 
protocol=jkv11 
port=C8:47:80:13:CC:AE 
command=getCellData 
outputs=screen

```
the output for the first proto (jk04) is wrong.
If there is just one proto used in the config, it works.

I just upgraded from Python3.9 to Python3.11 to get latest mpp-solar (version: 0.16.34, python version: 3.11.2) ... but the problem is the same.

Any suggestions?

jblance, 2024-06-16:
I'll have a look. You might need to run a separate instance for each device
though

mfu-mcosys, 2024-06-16:
Thats what i do as workaround.
Just dont run as a daemon and use a one line per device in a bash script which is triggered per cron.

jblance, 2024-06-16:
You can run 2 daemons just need to call them something different and have 2
service files etc

mfu-mcosys, 2024-06-16:
Want work (tested it) - because the PI3 BLE-device seems to be blocked if in use by any of the 2 daemons.

jblance, 2024-06-16:
Oh, bummer.

The ble stack and code are pretty rough. I'll see what I can do with it -
though it might be in powermon code instead

firemen027, 2024-07-09:
I don't want to hijack the post, But Mppsolar run here since 2021 with one Bms (daly wired) one Balancer (JK04 bluetooth) nd one inverter mppsolar wired.
Run smooth since that. 
I justed add a second battery bank with JK BMS bluetooth (jkv11), since I added it, the first JK bms show erroned data in the same range of [mfu-mcosys] result. 

So we are in the same boat

jblance, 2024-07-09:
> I don't want to hijack the post, But Mppsolar run here since 2021 with one Bms (daly wired) one Balancer (JK04 bluetooth) nd one inverter mppsolar wired. Run smooth since that. I justed add a second [...]

can you post your config file (as this shouldnt happen)

firemen027, 2024-07-10:
My config

This `post` is from my tablet, hard time with post formating

 Required section
[SETUP]
# Number of seconds to pause at the end of each loop (0 is no pause)
pause=7
# Hostname / IP address of the MQTT broker
mqtt_broker=localhost
#mqtt_broker=screen
 

[PIP3048]
model=standard
protocol=PI30
port=/dev/hidraw0
baud=2400
#command=QPGS0
command=QPIGS
tag=PIP3048
outputs=mqtt

[DALY]
protocol=DALY
port=/dev/ttyUSB0
#port=/dev/ttyUSB0
porttype=DALY
baud=9600
#command=mosStatus
#command=SOC,status,cellVoltages
command=SOC#mosStatus#status#cellMinMaxVoltages#cellVoltages#cellMinMaxTemps
tag=DALY
outputs=mqtt

[JKBMS]
type=jkbms
protocol=JK04
port=3C:A5:51:86:53:FF
command=getCellData
tag=Balancer
#outputs=influx_mqtt
outputs=mqtt

[JKBANK2]
type=jkbms
protocol=JKV11
port=C8:47:80:0B:F1:95
command=getCellData
tag=Bank2
outputs=mqtt

jblance, 2024-07-10:
oh, nvm I missed that both JKBMSs were BLE
this will need to wait on a code level fix and BLE re-write - sorry