jblance/mpp-solar #499: Docker: Run mpp-solar as a daemon?

nullinger, 2024-07-02:
Hi,

Is it possible to use the docker image to run mpp-solar as a daemon?

When used as decribed in the [Wiki](https://github.com/jblance/mpp-solar/wiki/Installation#using-docker), it just seems to execute once and then terminates:
```
docker run --rm --name=mpp-solar jblance/mppsolar mpp-solar
```

According to the [docker-compose.yaml](https://github.com/jblance/mpp-solar/blob/master/docker/prod/docker-compose.yaml), there seems to be a `--daemon` parameter but then it won’t start:

```
# docker run --rm --name=mpp-solar jblance/mppsolar mpp-solar --daemon

You are missing dependencies in order to be able to use the --daemon flag.
To install them, use that command:
    python -m pip install 'mppsolar[systemd]'
```

Am I missing something?
Thank you in advance.

jblance, 2024-07-03:
No that's a problem with the docker build. I'll try to fix it

jblance, 2024-07-04:
Ive add the required library to the latest docker build, but havent tested
it yet
give it a go and let me know how you get on

nullinger, 2024-07-05:
Hi, I've pulled the latest image but in my tests the error is still the same.

jblance, 2024-07-09:
ok, my mistake - version 0.16.39 should fix this

nullinger, 2024-07-10:
Yes, it’s working, thank you!

Btw, the image size increased significantly from 77 MB to 403 MB. Any chance to shrink it?

jblance, 2024-07-10:
yeah I had to move from a python-slim image to a 'full' image to have the systemd components
do you know of a smaller image that supports systemd? (or other service construct??)

closed 2024-07-10 by jblance

nullinger, 2024-07-10:
Unfortunately not.
Would a cron job be possible?

jblance, 2024-07-10:
you can run it as a cronjob, dont need it to be a daemon for that
you can build a smaller image using a Dockerfile like detailed here https://github.com/jblance/mpp-solar/wiki/Docker#build-new-docker-image