jblance/mpp-solar #503: QPIWS command problem

a-mkh, 2024-07-09:
```
# docker run --rm --device=/dev/hidraw1 jblance/mppsolar:latest mpp-solar -p /dev/hidraw1 -c QPIWS
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.12/site-packages/mppsolar/__init__.py", line 438, in main
    results = _device.run_command(command=_command)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/mppsolar/devices/device.py", line 115, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/mppsolar/protocols/abstractprotocol.py", line 355, in decode
    key = resp_format[2][j]
          ~~~~~~~~~~~~~~^^^
IndexError: list index out of range
root@homeassistant:/etc/udev/rules.d# docker run --rm --device=/dev/hidraw1 jblance/mppsolar:latest mpp-solar -p /dev/hidraw1 -c QPIWS -D
2024-07-09 12:17:20,466:INFO:__init__:main@213: Solar Device Command Utility, version: 0.16.37, python version: 3.12.4
2024-07-09 12:17:20,466:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-07-09 12:17:20,466:DEBUG:__init__:main@248: MqttBroker name: localhost, port: 1883, user: None
2024-07-09 12:17:20,466:DEBUG:__init__:main@250: udp port 5555
2024-07-09 12:17:20,466:DEBUG:__init__:main@252: Using Postgres None
2024-07-09 12:17:20,467:DEBUG:__init__:main@255: Using Mongo None with mppsolar
2024-07-09 12:17:20,467:INFO:__init__:main@369: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw1 (porttype=None)" using protocol "PI30"
2024-07-09 12:17:20,469:DEBUG:__init__:main@373: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2024-07-09 12:17:20,469:DEBUG:device:__init__@34: __init__ args ()
2024-07-09 12:17:20,469:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw1', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7f7bac74ba70>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway', 'prom_output_dir': '/var/lib/node_exporter'}
2024-07-09 12:17:20,469:DEBUG:__init__:get_port_type@39: port matches hidraw
2024-07-09 12:17:20,469:INFO:__init__:get_port@93: Using hidrawio for communications
2024-07-09 12:17:20,471:DEBUG:__init__:get_protocol@13: Protocol PI30
2024-07-09 12:17:20,670:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0x7f7bac894920>, protocol PI30 protocol handler
2024-07-09 12:17:20,671:DEBUG:__init__:main@420: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f7bac74aa20>, 'QPIWS', 'QPIWS', 'screen', None, None)]
2024-07-09 12:17:20,671:INFO:__init__:main@425: Looping 1 commands
2024-07-09 12:17:20,671:INFO:__init__:main@435: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x7f7bac894920>, protocol: PI30 protocol handler for command: QPIWS, tag: QPIWS, outputs: screen
2024-07-09 12:17:20,671:INFO:device:run_command@51: Running command QPIWS
2024-07-09 12:17:20,671:INFO:abstractprotocol:get_full_command@61: Using protocol b'PI30' with 45 commands
2024-07-09 12:17:20,671:DEBUG:protocol_helpers:crcPI@308: Calculating CRC for b'QPIWS'
2024-07-09 12:17:20,671:DEBUG:protocol_helpers:crcPI@358: Generated CRC 0xb4 0xda 0xb4da
2024-07-09 12:17:20,671:DEBUG:abstractprotocol:get_full_command@67: full command: b'QPIWS\xb4\xda\r'
2024-07-09 12:17:20,671:INFO:device:run_command@81: full command b'QPIWS\xb4\xda\r' for command QPIWS
2024-07-09 12:17:20,671:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'QPIWS'
2024-07-09 12:17:20,671:DEBUG:abstractprotocol:get_command_defn@73: Found command QPIWS in protocol b'PI30'
2024-07-09 12:17:20,731:DEBUG:hidrawio:send_and_receive@28: length of to_send: 8
2024-07-09 12:17:20,731:DEBUG:hidrawio:send_and_receive@33: sending full_command in on shot
2024-07-09 12:17:21,782:DEBUG:hidrawio:send_and_receive@63: usb response was: b'(100000000000000000000000000000000000\x15\xcf\r'
2024-07-09 12:17:21,786:DEBUG:device:run_command@98: Send and Receive Response b'(100000000000000000000000000000000000\x15\xcf\r'
2024-07-09 12:17:21,786:DEBUG:abstractprotocol:decode@238: response passed to decode: b'(100000000000000000000000000000000000\x15\xcf\r'
2024-07-09 12:17:21,786:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'QPIWS'
2024-07-09 12:17:21,786:DEBUG:abstractprotocol:get_command_defn@73: Found command QPIWS in protocol b'PI30'
2024-07-09 12:17:21,786:DEBUG:protocol_helpers:crcPI@308: Calculating CRC for b'(100000000000000000000000000000000000'
2024-07-09 12:17:21,787:DEBUG:protocol_helpers:crcPI@358: Generated CRC 0x15 0xcf 0x15cf
2024-07-09 12:17:21,787:DEBUG:pi30:check_response_valid@1028: CRCs match
2024-07-09 12:17:21,787:DEBUG:abstractprotocol:decode@283: Processing response of type DEFAULT
2024-07-09 12:17:21,787:DEBUG:abstractprotocol:decode@287: trimmed and split responses: [b'100000000000000000000000000000000000']
2024-07-09 12:17:21,787:INFO:abstractprotocol:decode@294: Processing DEFAULT type responses
2024-07-09 12:17:21,787:DEBUG:abstractprotocol:decode@311: result 100000000000000000000000000000000000, key Warning, resp_format ['stat_flags', 'Warning', ['', 'Inverter fault', 'Bus over fault', 'Bus under fault', 'Bus soft fail fault', 'Line fail warning', 'OPV short warning', 'Inverter voltage too low fault', 'Inverter voltage too high fault', 'Over temperature fault', 'Fan locked fault', 'Battery voltage too high fault', 'Battery low alarm warning', 'Reserved', 'Battery under shutdown warning', 'Reserved', 'Overload fault', 'EEPROM fault', 'Inverter over current fault', 'Inverter soft fail fault', 'Self test fail fault', 'OP DC voltage over fault', 'Battery open fault', 'Current sensor fail fault', 'Battery short fault', 'Power limit warning', 'PV voltage high warning', 'MPPT overload fault', 'MPPT overload warning', 'Battery too low to charge warning', '', '']]
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.12/site-packages/mppsolar/__init__.py", line 438, in main
    results = _device.run_command(command=_command)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/mppsolar/devices/device.py", line 115, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/mppsolar/protocols/abstractprotocol.py", line 355, in decode
    key = resp_format[2][j]
          ~~~~~~~~~~~~~~^^^
IndexError: list index out of range
```

jblance, 2024-07-09:
version 0.16.38 includes a fix for this

a-mkh, 2024-07-10:
Many thanks, works as expected!

closed 2024-07-10 by jblance