jblance/mpp-solar #504: JKBMS V4.23.0 cannot read data

asenyshyn, 2024-07-17:
Hi, I have JKBMS JK_PB1A16S15P 
Hardware version V15A
Software V15.27

I cannot any data from it.

```
$ mpp-solar -p C8:47:80:11:C6:19 -P JK04 -D
2024-07-17 20:05:09,605:INFO:__init__:main@216: Solar Device Command Utility, version: 0.16.39, python version: 3.11.2
2024-07-17 20:05:09,605:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-07-17 20:05:09,606:DEBUG:__init__:main@251: MqttBroker name: localhost, port: 1883, user: None
2024-07-17 20:05:09,606:DEBUG:__init__:main@253: udp port 5555
2024-07-17 20:05:09,607:DEBUG:__init__:main@255: Using Postgres None
2024-07-17 20:05:09,607:DEBUG:__init__:main@258: Using Mongo None with mppsolar
2024-07-17 20:05:09,611:INFO:__init__:main@275: Daemon DISABLED
2024-07-17 20:05:09,611:INFO:__init__:main@379: Creating device "unnamed" (type: "mppsolar") on port "C8:47:80:11:C6:19 (porttype=None)" using protocol "JK04"
2024-07-17 20:05:09,617:DEBUG:__init__:main@383: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2024-07-17 20:05:09,618:DEBUG:device:__init__@34: __init__ args ()
2024-07-17 20:05:09,618:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': 'C8:47:80:11:C6:19', 'protocol': 'JK04', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7f987e9d90>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway', 'prom_output_dir': '/var/lib/node_exporter'}
2024-07-17 20:05:09,619:DEBUG:__init__:get_port_type@51: port matches jkble ':'
2024-07-17 20:05:09,619:INFO:__init__:get_port@105: Using jkbleio for communications
2024-07-17 20:05:09,681:DEBUG:__init__:get_protocol@13: Protocol JK04
2024-07-17 20:05:10,339:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.jkbleio.JkBleIO object at 0x7f98820e50>, protocol JK04 - JKBMS BLE 4 byte data communication protocol handler
2024-07-17 20:05:10,340:DEBUG:__init__:main@430: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f9877b350>, '', '', 'screen', None, None)]
2024-07-17 20:05:10,340:INFO:__init__:main@435: Looping 1 commands
2024-07-17 20:05:10,341:INFO:__init__:main@441: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.jkbleio.JkBleIO object at 0x7f98820e50>, protocol: JK04 - JKBMS BLE 4 byte data communication protocol handler for command: , tag: , outputs: screen
2024-07-17 20:05:10,341:INFO:device:run_command@51: Running command
2024-07-17 20:05:10,341:INFO:jkabstractprotocol:get_full_command@74: Using protocol b'JK04' with 2 commands
2024-07-17 20:05:10,341:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2024-07-17 20:05:10,342:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getCellData'
2024-07-17 20:05:10,342:DEBUG:abstractprotocol:get_command_defn@73: Found command getCellData in protocol b'JK04'
2024-07-17 20:05:10,342:DEBUG:jkabstractprotocol:get_full_command@79: self._command = getCellData
2024-07-17 20:05:10,342:DEBUG:jkabstractprotocol:get_full_command@90: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-07-17 20:05:10,343:DEBUG:jkabstractprotocol:get_full_command@98: cmd with command code: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-07-17 20:05:10,343:DEBUG:jkabstractprotocol:get_full_command@100: cmd with crc: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2024-07-17 20:05:10,343:INFO:device:run_command@81: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2024-07-17 20:05:10,343:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2024-07-17 20:05:10,344:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getCellData'
2024-07-17 20:05:10,344:DEBUG:abstractprotocol:get_command_defn@73: Found command getCellData in protocol b'JK04'
2024-07-17 20:05:10,344:INFO:jkabstractprotocol:get_full_command@74: Using protocol b'JK04' with 2 commands
2024-07-17 20:05:10,345:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2024-07-17 20:05:10,345:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getCellData'
2024-07-17 20:05:10,345:DEBUG:abstractprotocol:get_command_defn@73: Found command getCellData in protocol b'JK04'
2024-07-17 20:05:10,345:DEBUG:jkabstractprotocol:get_full_command@79: self._command = getCellData
2024-07-17 20:05:10,346:DEBUG:jkabstractprotocol:get_full_command@90: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-07-17 20:05:10,346:DEBUG:jkabstractprotocol:get_full_command@98: cmd with command code: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-07-17 20:05:10,346:DEBUG:jkabstractprotocol:get_full_command@100: cmd with crc: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2024-07-17 20:05:10,346:INFO:jkbleio:send_and_receive@35: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2024-07-17 20:05:10,347:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2024-07-17 20:05:10,347:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getCellData'
2024-07-17 20:05:10,347:DEBUG:abstractprotocol:get_command_defn@73: Found command getCellData in protocol b'JK04'
2024-07-17 20:05:10,348:DEBUG:jkbleio:send_and_receive@39: expected record type 2 for command getCellData
2024-07-17 20:05:10,348:INFO:jkbleio:ble_connect@63: Attempting to connect to C8:47:80:11:C6:19
2024-07-17 20:05:10,661:DEBUG:jkbleio:ble_get_data@88: Command: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2024-07-17 20:05:10,698:DEBUG:jkbledelegate:handleNotification@27: From handle: 0x12 Got 4 bytes of data
2024-07-17 20:05:10,698:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'AT\r\n')
2024-07-17 20:05:10,699:DEBUG:jkabstractprotocol:wipe_to_start@150: SOR not found in record
2024-07-17 20:05:10,699:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'')
2024-07-17 20:05:10,699:DEBUG:jkbledelegate:handleNotification@38: Not expected type of record - wiping data bytearray(b'')
2024-07-17 20:05:10,699:DEBUG:jkabstractprotocol:is_record_complete@166: No SOR found in record looking for completeness
2024-07-17 20:05:10,810:INFO:jkbleio:ble_get_data@97: Connected to b''
2024-07-17 20:05:10,961:INFO:jkbleio:ble_get_data@113: Read characteristic: Characteristic <ffe1>, handle 12
2024-07-17 20:05:10,961:INFO:jkbleio:ble_get_data@121: Enable 0x0b handle {'rsp': ['wr']}
2024-07-17 20:05:10,962:INFO:jkbleio:ble_get_data@124: Enable read handle {'rsp': ['wr']}
2024-07-17 20:05:10,963:INFO:jkbleio:ble_get_data@128: Write getInfo to read handle {'rsp': ['wr']}
2024-07-17 20:05:11,092:DEBUG:jkbledelegate:handleNotification@27: From handle: 0x12 Got 128 bytes of data
2024-07-17 20:05:11,092:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'AT\r\nU\xaa\xeb\x90\x03\x1aJK_PB1A16S15P\x00\x00\x0015A\x00\x00\x00\x00\x0015.27\x00\x00\x00\xfc\x86\x04\x00\x06\x00\x00\x0048v105a\x00\x00\x00\x00\x00\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00240708\x00\x0040119460493\x00000\x00JK-BMS\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00140703')
2024-07-17 20:05:11,093:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'U\xaa\xeb\x90\x03\x1aJK_PB1A16S15P\x00\x00\x0015A\x00\x00\x00\x00\x0015.27\x00\x00\x00\xfc\x86\x04\x00\x06\x00\x00\x0048v105a\x00\x00\x00\x00\x00\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00240708\x00\x0040119460493\x00000\x00JK-BMS\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00140703')
2024-07-17 20:05:11,093:DEBUG:jkbledelegate:handleNotification@38: Not expected type of record - wiping data bytearray(b'U\xaa\xeb\x90\x03\x1aJK_PB1A16S15P\x00\x00\x0015A\x00\x00\x00\x00\x0015.27\x00\x00\x00\xfc\x86\x04\x00\x06\x00\x00\x0048v105a\x00\x00\x00\x00\x00\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00240708\x00\x0040119460493\x00000\x00JK-BMS\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00140703')
[... cut ...]
2024-07-17 20:05:11,280:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'\xaaU\x90\xeb\xc8\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00DAT\r\n')
2024-07-17 20:05:11,281:DEBUG:jkabstractprotocol:wipe_to_start@150: SOR not found in record
2024-07-17 20:05:11,281:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'')
2024-07-17 20:05:11,281:DEBUG:jkbledelegate:handleNotification@38: Not expected type of record - wiping data bytearray(b'')
2024-07-17 20:05:11,281:DEBUG:jkabstractprotocol:is_record_complete@166: No SOR found in record looking for completeness
```

Could anyone suggest how to get it working?

git-greatguy, 2024-08-31:
Same here on JK_PB2A16S15P.
i however once had data ouput  on getinfo. nothing since.

It loops on these 
```
2024-08-31 12:47:21,448:DEBUG:jkbledelegate:handleNotification@27: From handle: 0x12 Got 4 bytes of data
2024-08-31 12:47:21,448:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'AT\r\n')
2024-08-31 12:47:21,448:DEBUG:jkabstractprotocol:wipe_to_start@150: SOR not found in record
2024-08-31 12:47:21,448:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'')
2024-08-31 12:47:21,448:DEBUG:jkbledelegate:handleNotification@38: Not expected type of record - wiping data bytearray(b'')
2024-08-31 12:47:21,449:DEBUG:jkabstractprotocol:is_record_complete@166: No SOR found in record looking for completeness
2024-08-31 12:47:21,654:DEBUG:jkbledelegate:handleNotification@27: From handle: 0x12 Got 4 bytes of data
2024-08-31 12:47:21,654:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'AT\r\n')
2024-08-31 12:47:21,655:DEBUG:jkabstractprotocol:wipe_to_start@150: SOR not found in record
2024-08-31 12:47:21,655:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'')
2024-08-31 12:47:21,655:DEBUG:jkbledelegate:handleNotification@38: Not expected type of record - wiping data bytearray(b'')
2024-08-31 12:47:21,656:DEBUG:jkabstractprotocol:is_record_complete@166: No SOR found in record looking for completeness
```

Looks like a data parse issue ?

StraightSh00ter, 2024-11-10:
Same problem here ... cannot read data with JK_PB1A16S10P
Any help here?

TomSawyer8006, 2025-01-21:
Same here with JK_PB2A16S20P, hw v15A, sw v15.32