jblance/mpp-solar #505: QUESTION

devyaz, 2024-07-26:
has anyone tried this on ESP32?

Saentist, 2024-07-26:
There is few ESP devices with can run linux and they are slow.
Guess all need to be rewritten on MicroPython

see this alternative project's
https://github.com/scottwday/InverterOfThings
https://github.com/syssi/esphome-pipsolar
https://github.com/chipsi/esphome-pipsolar-voltronic
https://github.com/syssi/esphome-smg-ii

But not all are updated and compatible with devices as Mpp-solar

devyaz, 2024-07-29:
am hoping to put the ESP32 in the inverter, connected to the inverter via UART(TTL-RS232) have its own display (16x2) send events via SMS(A9G board)

jblance, 2024-07-29:
unfortunately not - I started but never made much progress.
the ESPHome work is good from what I hear and can be largely standalone
I think it has the components to do what you need as long as the inverter
is supported

devyaz, 2024-07-29:
Ok kool, youre very helpful! Ill try when i get time to transpile from C to micropython

Saentist, 2024-08-02:
> am hoping to put the ESP32 in the inverter, connected to the inverter via UART(TTL-RS232) have its own display (16x2) send events via SMS(A9G board) and controll via SMS(Telegram later on)

https://www.cnx-software.com/2024/08/01/nbiot-lte-m-air-monitor-solar-powered-esp32-s3-multi-sensor-board-with-sim7080g-nb-iot-lte-m-modem/

;) be more solar, be more modern :)

closed 2024-10-10 by jblance