jblance/mpp-solar #506: JKBMS and RS485 strange output?

axyaxyfr, 2024-07-27:
I have two JK_PB2A16S15P connected to eachother and an RS485 connection to a rpi running this. Hardware ver 15.XA and sw is 15.10.
It connects to the BMS but the values look strange. I use the supplied cable from eelbattery to connect...
Any suggestions?

ps@raspberrypi:~ $ jkbms -p /dev/ttyUSB0 -b 115200 -P JK485 -D
2024-07-27 22:23:52,104:INFO:__init__:main@214: Solar Device Command Utility, version: 0.15.62
2024-07-27 22:23:52,105:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-07-27 22:23:52,106:DEBUG:__init__:main@239: MqttBroker name: localhost, port: 1883, user: None
2024-07-27 22:23:52,106:DEBUG:__init__:main@241: udp port 5555
2024-07-27 22:23:52,106:DEBUG:__init__:main@243: Using Postgres None
2024-07-27 22:23:52,106:DEBUG:__init__:main@246: Using Mongo None with mppsolar
2024-07-27 22:23:52,107:INFO:__init__:main@347: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB0 (porttype=None)" using protocol "JK485"
2024-07-27 22:23:52,112:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2024-07-27 22:23:52,113:DEBUG:device:__init__@33: __init__ args ()
2024-07-27 22:23:52,113:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'JK485', 'baud': 115200, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7607bca0>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2024-07-27 22:23:52,113:DEBUG:__init__:get_port_type@69: port matches ttyusb
2024-07-27 22:23:52,114:INFO:__init__:get_port@111: Using serialio for communications
2024-07-27 22:23:52,128:DEBUG:__init__:get_protocol@13: Protocol JK485
2024-07-27 22:23:52,363:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0x7607bda8>, protocol JKBMS RS485 serial communication protocol handler
2024-07-27 22:23:52,363:DEBUG:__init__:main@402: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x7607bcb8>, '', '', 'screen', None, None)]
2024-07-27 22:23:52,363:INFO:__init__:main@407: Looping 1 commands
2024-07-27 22:23:52,364:INFO:__init__:main@417: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x7607bda8>, protocol: JKBMS RS485 serial communication protocol handler for command: , tag: , outputs: screen
2024-07-27 22:23:52,364:INFO:device:run_command@50: Running command 
2024-07-27 22:23:52,364:INFO:jk485:get_full_command@100: Using protocol b'JK485' with 1 commands
2024-07-27 22:23:52,364:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getBalancerData
2024-07-27 22:23:52,365:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'getBalancerData'
2024-07-27 22:23:52,365:DEBUG:abstractprotocol:get_command_defn@72: Found command getBalancerData in protocol b'JK485'
2024-07-27 22:23:52,365:DEBUG:jk485:get_full_command@114: cmd with header: bytearray(b'U\xaa\x00\x00\x00\x00\x00')
2024-07-27 22:23:52,365:DEBUG:jk485:get_full_command@117: cmd with header + slave address: bytearray(b'U\xaa\x01\x00\x00\x00\x00')
2024-07-27 22:23:52,366:DEBUG:jk485:get_full_command@122: cmd with command code and frame data: bytearray(b'U\xaa\x01\xff\x00\x00\x00')
2024-07-27 22:23:52,366:DEBUG:jk485:get_full_command@125: cmd with crc: bytearray(b'U\xaa\x01\xff\x00\x00\xff')
2024-07-27 22:23:52,366:INFO:device:run_command@78: full command bytearray(b'U\xaa\x01\xff\x00\x00\xff') for command getBalancerData
2024-07-27 22:23:52,366:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getBalancerData
2024-07-27 22:23:52,367:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'getBalancerData'
2024-07-27 22:23:52,367:DEBUG:abstractprotocol:get_command_defn@72: Found command getBalancerData in protocol b'JK485'
2024-07-27 22:23:52,367:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 115200
2024-07-27 22:23:52,370:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2024-07-27 22:23:53,727:DEBUG:serialio:send_and_receive@30: serial response was: b'\x02\x10\x16 \x00\x01\x02\x00\x00\xc2\x01\x03\x10\x16 \x00\x01\x02\x00\x00\xcf\x91\x04\x10\x16 \x00\x01\x02\x00\x00\xe9\xa1\x05'
2024-07-27 22:23:53,729:DEBUG:device:run_command@95: Send and Receive Response b'\x02\x10\x16 \x00\x01\x02\x00\x00\xc2\x01\x03\x10\x16 \x00\x01\x02\x00\x00\xcf\x91\x04\x10\x16 \x00\x01\x02\x00\x00\xe9\xa1\x05'
2024-07-27 22:23:53,730:INFO:abstractprotocol:decode@245: response passed to decode: b'\x02\x10\x16 \x00\x01\x02\x00\x00\xc2\x01\x03\x10\x16 \x00\x01\x02\x00\x00\xcf\x91\x04\x10\x16 \x00\x01\x02\x00\x00\xe9\xa1\x05'
2024-07-27 22:23:53,730:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getBalancerData
2024-07-27 22:23:53,731:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'getBalancerData'
2024-07-27 22:23:53,731:DEBUG:abstractprotocol:get_command_defn@72: Found command getBalancerData in protocol b'JK485'
2024-07-27 22:23:53,732:INFO:abstractprotocol:decode@289: Processing response of type POSITIONAL
2024-07-27 22:23:53,732:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 2, 'Header', '']
2024-07-27 22:23:53,733:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 1, 'Slave Address', '']
2024-07-27 22:23:53,733:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 1, 'Command Code', '']
2024-07-27 22:23:53,734:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/100', 2, 'Total Battery Voltage', 'V']
2024-07-27 22:23:53,734:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Average Cell Voltage', 'V']
2024-07-27 22:23:53,735:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Int', 1, 'Number of Cells', '']
2024-07-27 22:23:53,735:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Int', 1, 'Highest Cell', '']
2024-07-27 22:23:53,736:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Int', 1, 'Lowest Cell', '']
2024-07-27 22:23:53,736:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 1, 'Charging / Discharging', '']
2024-07-27 22:23:53,737:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 1, 'Alarm - todo', '']
2024-07-27 22:23:53,737:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Voltage Difference', 'V']
2024-07-27 22:23:53,738:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Balance Current', 'A']
2024-07-27 22:23:53,738:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Balance Trigger Voltage', 'V']
2024-07-27 22:23:53,739:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Max Balance Current', 'A']
2024-07-27 22:23:53,739:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 1, 'Balance On / Off', '']
2024-07-27 22:23:53,739:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Int', 1, 'Set Number of Cells', '']
2024-07-27 22:23:53,740:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Voltage Cell01', 'V']
2024-07-27 22:23:53,740:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Voltage Cell02', 'V']
2024-07-27 22:23:53,741:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Voltage Cell03', 'V']
2024-07-27 22:23:53,741:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Voltage Cell04', 'V']
2024-07-27 22:23:53,742:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Voltage Cell05', 'V']
2024-07-27 22:23:53,742:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Voltage Cell06', 'V']
2024-07-27 22:23:53,743:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Voltage Cell07', 'V']
2024-07-27 22:23:53,743:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Voltage Cell08', 'V']
2024-07-27 22:23:53,744:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Voltage Cell09', 'V']
2024-07-27 22:23:53,744:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Voltage Cell10', 'V']
[... 288 lines cut ...]
voltage_cell05          -5.727         	V   
voltage_cell06          0.0            	V   
voltage_cell07          0.0            	V   
[... cut ...]
voltage_cell19          0.0            	V   
voltage_cell20          0.0            	V   
voltage_cell21          0.0            	V   
voltage_cell22          0.0            	V   
voltage_cell23          0.0            	V   
voltage_cell24          0.0            	V   
temperature             0              	°C  
checksum                               	    
2024-07-27 22:23:53,852:DEBUG:__init__:main@452: Not daemon, so not looping
ps@raspberrypi:~ $ lsusb
Bus 001 Device 004: ID 1a86:7523 QinHeng Electronics CH340 serial converter

axyaxyfr, 2024-07-28:
Also tried BT with JK04, but it runs until I break and gives this:

2024-07-28 08:22:41,254:DEBUG:jkabstractprotocol:wipe_to_start@150: SOR not found in record
2024-07-28 08:22:41,255:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'')
2024-07-28 08:22:41,255:DEBUG:jkbledelegate:handleNotification@38: Not expected type of record - wiping data bytearray(b'')
2024-07-28 08:22:41,255:DEBUG:jkabstractprotocol:is_record_complete@166: No SOR found in record looking for completeness
2024-07-28 08:22:41,459:DEBUG:jkbledelegate:handleNotification@27: From handle: 0x12 Got 4 bytes of data
2024-07-28 08:22:41,460:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'AT\r\n')
2024-07-28 08:22:41,460:DEBUG:jkabstractprotocol:wipe_to_start@150: SOR not found in record
2024-07-28 08:22:41,461:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'')
2024-07-28 08:22:41,461:DEBUG:jkbledelegate:handleNotification@38: Not expected type of record - wiping data bytearray(b'')
2024-07-28 08:22:41,462:DEBUG:jkabstractprotocol:is_record_complete@166: No SOR found in record looking for completeness
2024-07-28 08:22:41,759:DEBUG:jkbledelegate:handleNotification@27: From handle: 0x12 Got 4 bytes of data
2024-07-28 08:22:41,760:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'AT\r\n')
2024-07-28 08:22:41,760:DEBUG:jkabstractprotocol:wipe_to_start@150: SOR not found in record
2024-07-28 08:22:41,761:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'')
2024-07-28 08:22:41,761:DEBUG:jkbledelegate:handleNotification@38: Not expected type of record - wiping data bytearray(b'')
2024-07-28 08:22:41,762:DEBUG:jkabstractprotocol:is_record_complete@166: No SOR found in record looking for completeness
^CTraceback (most recent call last):
  File "/home/ps/.local/bin/jkbms", line 8, in <module>
    sys.exit(main())
  File "/home/ps/.local/lib/python3.9/site-packages/mppsolar/__init__.py", line 420, in main
    results = _device.run_command(command=_command)
  File "/home/ps/.local/lib/python3.9/site-packages/mppsolar/devices/device.py", line 89, in run_command
    raw_response = self._port.send_and_receive(
  File "/home/ps/.local/lib/python3.9/site-packages/mppsolar/inout/jkbleio.py", line 37, in send_and_receive
    response = self.ble_get_data(full_command)
  File "/home/ps/.local/lib/python3.9/site-packages/mppsolar/inout/jkbleio.py", line 125, in ble_get_data
    if self._device.waitForNotifications(1.0):
  File "/usr/local/lib/python3.9/dist-packages/bluepy/btle.py", line 560, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/usr/local/lib/python3.9/dist-packages/bluepy/btle.py", line 407, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/usr/local/lib/python3.9/dist-packages/bluepy/btle.py", line 342, in _waitResp
    fds = self._poller.poll(timeout*1000)
KeyboardInterrupt

jblance, 2024-07-28:
> 2024-07-27 ***@***.***: Solar Device Command
> 2024-07-27 ***@***.***: udp port 5555
> 2024-07-27 ***@***.***: Using Postgres None
> 2024-07-27 ***@***.***: Using Mongo None with
> 2024-07-27 ***@***.***: Creating device "unnamed"
> 2024-07-27 ***@***.***: device_class <class
> 2024-07-27 ***@***.***: port matches ttyusb
> 2024-07-27 ***@***.***: Using serialio for
> 2024-07-27 ***@***.***: Protocol JK485
> 2024-07-27 ***@***.***: Commands
> 2024-07-27 ***@***.***: Looping 1 commands
> 2024-07-27 ***@***.***: Getting results from
> 2024-07-27 ***@***.***: Running command
> 2024-07-27 ***@***.***: Using protocol
> 2024-07-27 ***@***.***: Found
> 2024-07-27 ***@***.***: cmd with
> 2024-07-27 ***@***.***: cmd with header
> 2024-07-27 ***@***.***: full command
> 2024-07-27 ***@***.***: port
> 2024-07-27 ***@***.***: Executing
> 2024-07-27 ***@***.***: serial
> 2024-07-27 ***@***.***: Send and Receive
> 2024-07-27 ***@***.***: response passed
> 2024-07-27 ***@***.***: Processing
> 2024-07-27 ***@***.***: Got
> 2024-07-27 ***@***.***: trimmed and
> 2024-07-27 ***@***.***: Got defn
> 2024-07-27 ***@***.***: Hex
> 2024-07-27 ***@***.***: Hex b'\x16'
> 2024-07-27 ***@***.***: Hex b' '
> 2024-07-27 ***@***.***: Hex b'\x00'
> 2024-07-27 ***@***.***: Hex b'\xc2'
> 2024-07-27 ***@***.***: Hex b'\x01'
> 2024-07-27 ***@***.***: Hex b'\x03'
> 2024-07-27 ***@***.***: Hex b'\x10'
> 2024-07-27 ***@***.***: Hex b'\x91'
> 2024-07-27 ***@***.***: Hex b'\x04'
> 2024-07-27 ***@***.***: Hex b'
> 2024-07-27 ***@***.***: Hex b''
> 2024-07-27 ***@***.***: Decoded response
> 2024-07-27 ***@***.***: attempting to create
> 2024-07-27 ***@***.***: Using output filter: None
> 2024-07-27 ***@***.***: Not daemon, so not looping

The response doesnt match any of the protocols I have documentation for -
perhaps the baud rate is wrong
can you try at 9600?

axyaxyfr, 2024-07-29:
Sure thing. With 9600:

ps@raspberrypi:~ $ jkbms -p /dev/ttyUSB0 -b 9600 -P JK485 -D
2024-07-29 06:40:49,507:INFO:__init__:main@214: Solar Device Command Utility, version: 0.15.62
2024-07-29 06:40:49,507:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-07-29 06:40:49,508:DEBUG:__init__:main@239: MqttBroker name: localhost, port: 1883, user: None
2024-07-29 06:40:49,508:DEBUG:__init__:main@241: udp port 5555
2024-07-29 06:40:49,509:DEBUG:__init__:main@243: Using Postgres None
2024-07-29 06:40:49,509:DEBUG:__init__:main@246: Using Mongo None with mppsolar
2024-07-29 06:40:49,509:INFO:__init__:main@347: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB0 (porttype=None)" using protocol "JK485"
2024-07-29 06:40:49,524:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2024-07-29 06:40:49,525:DEBUG:device:__init__@33: __init__ args ()
2024-07-29 06:40:49,525:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'JK485', 'baud': 9600, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x760bbca0>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2024-07-29 06:40:49,526:DEBUG:__init__:get_port_type@69: port matches ttyusb
2024-07-29 06:40:49,526:INFO:__init__:get_port@111: Using serialio for communications
2024-07-29 06:40:49,557:DEBUG:__init__:get_protocol@13: Protocol JK485
2024-07-29 06:40:49,877:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0x760bbda8>, protocol JKBMS RS485 serial communication protocol handler
2024-07-29 06:40:49,877:DEBUG:__init__:main@402: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x760bbcb8>, '', '', 'screen', None, None)]
2024-07-29 06:40:49,878:INFO:__init__:main@407: Looping 1 commands
2024-07-29 06:40:49,878:INFO:__init__:main@417: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x760bbda8>, protocol: JKBMS RS485 serial communication protocol handler for command: , tag: , outputs: screen
2024-07-29 06:40:49,878:INFO:device:run_command@50: Running command 
2024-07-29 06:40:49,879:INFO:jk485:get_full_command@100: Using protocol b'JK485' with 1 commands
2024-07-29 06:40:49,879:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getBalancerData
2024-07-29 06:40:49,879:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'getBalancerData'
2024-07-29 06:40:49,879:DEBUG:abstractprotocol:get_command_defn@72: Found command getBalancerData in protocol b'JK485'
2024-07-29 06:40:49,879:DEBUG:jk485:get_full_command@114: cmd with header: bytearray(b'U\xaa\x00\x00\x00\x00\x00')
2024-07-29 06:40:49,880:DEBUG:jk485:get_full_command@117: cmd with header + slave address: bytearray(b'U\xaa\x01\x00\x00\x00\x00')
2024-07-29 06:40:49,880:DEBUG:jk485:get_full_command@122: cmd with command code and frame data: bytearray(b'U\xaa\x01\xff\x00\x00\x00')
2024-07-29 06:40:49,880:DEBUG:jk485:get_full_command@125: cmd with crc: bytearray(b'U\xaa\x01\xff\x00\x00\xff')
2024-07-29 06:40:49,880:INFO:device:run_command@78: full command bytearray(b'U\xaa\x01\xff\x00\x00\xff') for command getBalancerData
2024-07-29 06:40:49,881:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getBalancerData
2024-07-29 06:40:49,881:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'getBalancerData'
2024-07-29 06:40:49,881:DEBUG:abstractprotocol:get_command_defn@72: Found command getBalancerData in protocol b'JK485'
2024-07-29 06:40:49,882:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 9600
2024-07-29 06:40:49,885:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2024-07-29 06:40:51,005:DEBUG:serialio:send_and_receive@30: serial response was: b'I\xef\x8e\x0c\x00\x00\xd8\x08H@\x00\x00\x00\x00\x00\x84\x19\x00@\x80\x80 \x03\x00\x84\xfe\x00'
2024-07-29 06:40:51,006:DEBUG:device:run_command@95: Send and Receive Response b'I\xef\x8e\x0c\x00\x00\xd8\x08H@\x00\x00\x00\x00\x00\x84\x19\x00@\x80\x80 \x03\x00\x84\xfe\x00'
2024-07-29 06:40:51,007:INFO:abstractprotocol:decode@245: response passed to decode: b'I\xef\x8e\x0c\x00\x00\xd8\x08H@\x00\x00\x00\x00\x00\x84\x19\x00@\x80\x80 \x03\x00\x84\xfe\x00'
2024-07-29 06:40:51,008:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getBalancerData
2024-07-29 06:40:51,008:DEBUG:abstractprotocol:get_command_defn@70: Processing command 'getBalancerData'
2024-07-29 06:40:51,008:DEBUG:abstractprotocol:get_command_defn@72: Found command getBalancerData in protocol b'JK485'
2024-07-29 06:40:51,010:INFO:abstractprotocol:decode@289: Processing response of type POSITIONAL
2024-07-29 06:40:51,010:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 2, 'Header', '']
2024-07-29 06:40:51,011:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 1, 'Slave Address', '']
2024-07-29 06:40:51,011:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Str', 1, 'Command Code', '']
2024-07-29 06:40:51,012:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/100', 2, 'Total Battery Voltage', 'V']
2024-07-29 06:40:51,013:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['BigHex2Short:r/1000', 2, 'Average Cell Voltage', 'V']
2024-07-29 06:40:51,013:DEBUG:jkabstractprotocol:get_responses@127: Got defn: ['Hex2Int', 1, 'Number of Cells', '']
[... cut ...]
voltage_cell17          0.0            	V   
voltage_cell18          0.0            	V   
voltage_cell19          0.0            	V   
voltage_cell20          0.0            	V   
voltage_cell21          0.0            	V   
voltage_cell22          0.0            	V   
voltage_cell23          0.0            	V   
voltage_cell24          0.0            	V   
temperature             0              	°C  
checksum                               	    
2024-07-29 06:40:51,176:DEBUG:__init__:main@452: Not daemon, so not looping

jblance, 2024-07-29:
i dont think the code work with 'real' rs485 - pretty sure someone else found this out an made a script that worked - looks like they used the esphome solution https://github.com/txubelaxu/esphome-jk-bms
i have bought a jkbms with rs485 interface that i need to install to test against

axyaxyfr, 2024-08-06:
Is it also an unknown protocol with BT? I get the same is issue #504 
Thanks

lukolszewski, 2024-08-18:
Just like the poster above I too am getting nothing back from the BMS using mppsolar, but sending to it '4E 57 00 13 00 00 00 00 06 03 00 00 00 00 00 00 68 00 00 01 29' responds will all data.

I tried all sorts of commands, but I believe the closes to success was: 
jkbms -P jkserial -b 115200 -I --porttype serial -p /dev/ttyUSB5

the result is:
> jkbms -P jkserial -b 115200 -I --porttype serial -p /dev/ttyUSB5 -D
2024-08-18 09:28:42,082:INFO:__init__:main@216: Solar Device Command Utility, version: 0.16.40-dev, python version: 3.11.2
2024-08-18 09:28:42,082:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-08-18 09:28:42,082:DEBUG:__init__:main@251: MqttBroker name: localhost, port: 1883, user: None
2024-08-18 09:28:42,082:DEBUG:__init__:main@253: udp port 5555
2024-08-18 09:28:42,083:DEBUG:__init__:main@255: Using Postgres None
2024-08-18 09:28:42,083:DEBUG:__init__:main@258: Using Mongo None with mppsolar
2024-08-18 09:28:42,083:INFO:__init__:main@275: Daemon DISABLED
2024-08-18 09:28:42,083:INFO:__init__:main@379: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB5 (porttype=serial)" using protocol "jkserial"
2024-08-18 09:28:42,084:DEBUG:__init__:main@383: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2024-08-18 09:28:42,084:DEBUG:device:__init__@34: __init__ args ()
2024-08-18 09:28:42,085:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB5', 'protocol': 'jkserial', 'baud': 115200, 'porttype': 'serial', 'mqtt_brok
er': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7fff38d51190>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/push
gateway', 'prom_output_dir': '/var/lib/node_exporter'}
2024-08-18 09:28:42,085:INFO:__init__:get_port@81: Port overide - using port 'serial'
2024-08-18 09:28:42,085:DEBUG:__init__:get_port_type@66: port matches serial
2024-08-18 09:28:42,085:INFO:__init__:get_port@111: Using serialio for communications
2024-08-18 09:28:42,087:DEBUG:__init__:get_protocol@13: Protocol jkserial
2024-08-18 09:28:42,211:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0x7fff39993410>, protocol JKBMS TTL serial communication pro
tocol handler
2024-08-18 09:28:42,212:DEBUG:__init__:main@430: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x7fff39931050>, '', '', 'screen', None, None)]
2024-08-18 09:28:42,212:INFO:__init__:main@435: Looping 1 commands
2024-08-18 09:28:42,212:INFO:__init__:main@441: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x7fff39993410>, protocol
: JKBMS TTL serial communication protocol handler for command: , tag: , outputs: screen
2024-08-18 09:28:42,212:INFO:device:run_command@51: Running command
2024-08-18 09:28:42,212:DEBUG:jkserial:get_full_command@117: Using protocol b'JKSERIAL' with 1 commands
2024-08-18 09:28:42,212:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getBalancerData'
2024-08-18 09:28:42,212:DEBUG:abstractprotocol:get_command_defn@73: Found command getBalancerData in protocol b'JKSERIAL'
2024-08-18 09:28:42,212:DEBUG:jkserial:get_full_command@159: cmd with crc: bytearray(b'NW\x00\x13\x00\x00\x00\x00\x03\x03\x00\x00\x00\x00\x00\x00h\x00\x00\x01&')
2024-08-18 09:28:42,212:INFO:device:run_command@81: full command bytearray(b'NW\x00\x13\x00\x00\x00\x00\x03\x03\x00\x00\x00\x00\x00\x00h\x00\x00\x01&') for command getBalancerData
2024-08-18 09:28:42,212:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getBalancerData'
2024-08-18 09:28:42,212:DEBUG:abstractprotocol:get_command_defn@73: Found command getBalancerData in protocol b'JKSERIAL'
2024-08-18 09:28:42,212:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB5, baudrate 115200
2024-08-18 09:28:42,213:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2024-08-18 09:28:42,313:DEBUG:serialio:send_and_receive@30: serial response was: b'NW\x01!\x00\x00\x00\x00\x03\x00\x01y0\x01\r'
2024-08-18 09:28:42,314:DEBUG:device:run_command@98: Send and Receive Response b'NW\x01!\x00\x00\x00\x00\x03\x00\x01y0\x01\r'
2024-08-18 09:28:42,314:DEBUG:abstractprotocol:decode@238: response passed to decode: b'NW\x01!\x00\x00\x00\x00\x03\x00\x01y0\x01\r'
2024-08-18 09:28:42,314:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getBalancerData'
2024-08-18 09:28:42,314:DEBUG:abstractprotocol:get_command_defn@73: Found command getBalancerData in protocol b'JKSERIAL'
2024-08-18 09:28:42,314:DEBUG:abstractprotocol:decode@283: Processing response of type POSITIONAL
2024-08-18 09:28:42,314:DEBUG:jkserial:get_responses@187: get_responses: responses [b'NW\x01!\x00\x00\x00\x00\x03\x00\x01', b'y', b'0', b'\x01', b'\r', b'', b'', b'', b'', b'', b'',
b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'
', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'']
2024-08-18 09:28:42,314:DEBUG:abstractprotocol:decode@287: trimmed and split responses: [b'NW\x01!\x00\x00\x00\x00\x03\x00\x01', b'y', b'0', b'\x01', b'\r', b'', b'', b'', b'', b'',
[... cut ...]
I looked at the translated Chinese document for something else we can interpret as the end of the frame. I found this. At the end there is always 0x68 followed by 4 byte checksum, of which 2 bytes are not used and are zero. So it seems every frame ends with (hex) 68 00 00 XX XX and this is one simple method to detect the end (a better one would be to process the header and get the length, but this is the faster method I chose today).

So after making the following changes: https://github.com/lukolszewski/mpp-solar/tree/feature/jk_serialio

It appears to be working fine!

jblance, 2024-08-18:
> jkbms -P jkserial -b 115200 -I --porttype serial -p /dev/ttyUSB5

`--porttype daly` should work too

lukolszewski, 2024-08-18:
> > jkbms -P jkserial -b 115200 -I --porttype serial -p /dev/ttyUSB5
> 
> `--porttype daly` should work too

Indeed it does :sweat_smile:

gianfrdp, 2024-09-17:
Hello, which protocol is configured on BMS UART?
Are you using UART2?
Did someone try using RS232 port?