jblance/mpp-solar #509: Axpert MKS IV  Hybrid  stops responding every few weeks

ThisIsQasim, 2024-08-18:
Thanks for the great project. I have a rebranded Axpert MKS IV Twin Hybrid 6kW inverter that I use with mpp_solar and it works great with PI18 over USB (`/dev/hidraw0`). However, the inverter just stops responding every few weeks and I have to restart it. I remember someone else reported the same issue but I can’t find it. Any ideas for a fix?

jblance, 2024-08-18:
I havent investigated this (that I can recall)
Is the problem with the inverter or does mppsolar fail?
Any debug / logs from the time of failure?
Have you tried other resolution approaches?

ThisIsQasim, 2024-08-18:
It’s an issue with the inverter. It just stops responding. Actually the `hidraw0` device also disappears. Not sure what to try. Was wondering if you had any ideas.

jblance, 2024-08-18:
try https://www.linux.org/threads/resetting-the-usb-subsystem.10404/

closed 2024-10-10 by jblance