jblance/mpp-solar #510: Sending JK BMS PB-x firmware update command via BLE

tarek-zy, 2024-08-26:
Hi, thank you for your great project and support!
I'm a Software Engineer, and I have a JK BMS (PB-2A16S15P) with a faulty RS485-1 (right besides CAN port), which makes me unable to update my BMS firmware or even connect it to my inverter, as there is no communication possible via that port at all, so I had to do a lot of tests and sniffed all communication possible.
When I try to update the firmware via RS485-2 ports that are designated for parallel connection with other JK BMS units, the BMS shuts down / crashes immediately. We are a growing number of owners having this fatal bug and despite several attempts to contact JK support and other popular YouTube influencers, we are out of luck so far.
Since you have full knowledge of the communication protocols of JK BMS (PB series), is it possible to edit the command to make it possible to be executed via RS485-2 or BLE?

RS485-1 Firmware update command (HEX):
01 10 16 26 00 01 02 00 00 D6 97

jblance, 2024-09-22:
Did this command work?
What process did you follow (be good to add it to mppsolar if possible)

tarek-zy, 2024-09-22:
The command I provided here works on RS485 only, and I didn't get any help from anyone yet to convert it to a BLE command. So, no updates on this issue whatsoever, I'm still stuck with the same fatal bug with no help.

jblance, 2024-09-22:
Oh I see - I dont see how the command can update the firmware tho - surely you'd need to send the new firmware as well

tarek-zy, 2024-09-23:
I totally understand what you're saying.
Actually, as I sniffed all RS485 communication right when the official JK BMS Windows app tries to do a normal or force update for the BMS. In both methods, it starts the update process by sending the aforementioned command then it starts sending the new firmware file in packets until it finishes uploading it.
This fatal bug relies in that the BMS never responds to the command over RS485, so that's why I'm trying to send it via BLE. For the update process to be successful, the BMS should reply with HEX (15) after receiving the update command, than after each firmware file packet it should reply with HEX (06).

So, the whole idea is to do the firmware update process over BLE instead of RS485, as it's not working at all in a faulty BMS.
I, with other professionals, have had a lot of extensive discussions and analysis in a German forum (akkudoktor.com) about this fatal bug. We figured out all these information and put all the pieces together, but have got no help at all in doing the update process successfully on our faulty BMSes over BLE, as it's still impossible for us to get it done over RS485.
And as I mentioned earlier, JK never replied to us to assist with their products, and I tried a few times to contact Andy from Off-Grid Garage Australia on YouTube, but never got a reply from him.

jblance, 2024-09-23:
The BLE commands look like [20 bytes long)  (where the 0x97 is the command designator)
`b'\xaa\x55\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11'`
compared to: `01 10 16 26 00 01 02 00 00 D6 97` I'm not sure how to translate
There seem to be many protocols for JK BMSs
jkserial has a 21 byte command (that looks closer)
`b'NW\x00\x13\x00\x00\x00\x00\x06\x03\x00\x00\x00\x00\x00\x00h\x00\x00\x01)'`
With these bytes looking important:
```
# command byte: 0x01 (activation), 0x02 (write), 0x03 (read), 0x05 (password), 0x06 (read all)
byte[8] = command_byte
byte[9] = 0x03                         # frame source: 0x00 (bms), 0x01 (bluetooth), 0x02 (gps), 0x03 (computer)
byte[10] = 0x00                        # frame type: 0x00 (read data), 0x01 (reply frame), 0x02 (BMS active upload)
byte[11] = command_code                # register: 0x00 (read all registers), 0x8E...0xBF (holding registers)
```        
So not really sure how to map the command you supplied can map - do you have any other commands that could be used as comparsions

tarek-zy, 2024-09-23:
Yeah, sure!
I've put together a fine extract of all commands and responses that goes through the JK BMS Windows app. 
Please see the attached file for other commands that you can recognize for sure, as they match what you already have in your library.

[Commands.txt](https://github.com/user-attachments/files/17092391/Commands.txt)

jblance, 2024-09-24:
Wow - thats kinda weird

the response looks like the jk02 / jk04 protocol, but the commands are completely new (and I assume that the posted response is just the start of the data?)

this is via the RS485 port?

tarek-zy, 2024-09-29:
I'm so sorry for my late response.
Yes, my previously posted responses are only the start of the data, and yes, this is R485 on the new PB series. I just sniffed some fresh data for you for every command I provided:

```
COMMAND:
0110161C0001020000D3CD

RESPONSE:
55AAEB9003054A4B5F5042324131365331355000000031352E584100000031352E3130000000450BCE00C401000041545220424D5300000000000000000031323334000000000000000000000000323430353130000033313231333438303038350030303000000000000000000000000000000000003430393600000000000000000000000000000000000000000000000000000000FEFFFFFF2FE9010200000000901F00000000C0D8E7FE1F00000100000000000000000100CF030000000000000000000000000000DF07000000000000000000000000000001CF03000000000000000000000000000A00000101000000000000000000000000000000000000000000000008070000323201000000000000000000000000000000000000FE9FE9FF07000000000000005F0110161C0001C447

COMMAND:
0110161E0001020000D22F

RESPONSE:
55AAEB900105E40C00005A0A0000960A0000420E0000AA0D000005000000AB0D00008C0A0000AC0D0000480D0000C4090000F04902001E0000003C000000F04902001E0000003C00000005000000D0070000580200002602000058020000260200009CFFFFFFCEFFFFFFE80300002003000008000000010000000100000001000000605B0300DC050000160D00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010000000500000060E3160050033C3218FEFFFFFF3FE9010200000000500110161E00016587

COMMAND:
011016200001020000D6F1

RESPONSE:
55AAEB9002051D0D1D0D1D0D1C0D1C0D1D0D1C0D1C0D000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000FF0000001D0D01000400420042004E00520057005700640066000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000002C0100000000E4680000604F0700B1450000280126010000000000000058B9F30200605B030065000000E0135601640000007A0BCE0001010000000000000000000000000000FF0001000000E8032F00000066F43F40000000007D0A000000010101000600001E020000000000002C011E011E01B503ED19ED08AE1C00008051010000000000000000000000000000FEFF7FDD0F0100B007000000B5011016200001044B
```

tarek-zy, 2024-09-29:
Please also note that project "esphome-jk-bms" started supporting series PB of the JK BMS a while ago, if you would like to take a look at the differences in the data structure than previous series of their products

jblance, 2024-10-01:
I couldnt find anything in the esphome-jk-bms code that matches your
commands (not that I really understand the esphome stuff)

Ive added a jkpb protocol to the github source tree - be interested to see
what results it gives

tarek-zy, 2024-10-09:
Sorry for being late!
I referred you to the esphome-jk-bms repo since they already implemented a variant for the newest model series (JK-PB) for ESP8266 in the following file:
https://github.com/syssi/esphome-jk-bms/blob/main/esp8266-jk-pb-modbus-example.yaml

Please also note that I don't own an ESP microcontroller yet, so I can't help with any testing right now. I thought that it would be easy to convert commands between protocols. I tried to contact @syssi a while ago, but I never got help on this matter.