jblance/mpp-solar #511: Heltec balancer support

B4rtOl0, 2024-09-04:
Hello,
Is there any chance to add Heltec/Neey  over BLE balancers ? As far as i know ESPhome already support this devices:
https://github.com/syssi/esphome-jk-bms/blob/main/components/heltec_balancer_ble/heltec_balancer_ble.cpp

jblance, 2024-09-08:
im adding this support in the powermon code at the moment - not sure if I'll backport to mppsolar (as the BLE code is a mess and quite specific to the JK BMS)

B4rtOl0, 2024-09-09:
Hey, thanks for your reply. If you need help in testing i can offer you access to my RaspberryPI where you can check everything.

B4rtOl0, 2024-09-09:
Can i ask to keep compatibility with python3.9 ? Most likely we all trying to use old hardware for monitoring devices and powermon uses match and its not compatible with python < 3.11

Saentist, 2024-09-09:
@B4rtOl0 try Alpine linux native

B4rtOl0, 2024-09-09:
@jblance  OK thanks got it working on chroot but seems like neey protocol is not finished yet. I'll wait and if you need me to test anything feel free to ask :-)

B4rtOl0, 2024-09-09:
@jblance I've made some tests with ble_neey and it is working quite good:
----------------------------------------------------------------------
Command: cell_info - information about the cells
----------------------------------------------------------------------
Parameter                  Value                                  Unit 
cell_01_voltage            3.2806                                 V    
cell_02_voltage            3.2818                                 V    
cell_03_voltage            3.2811                                 V    
cell_04_voltage            3.2814                                 V    
cell_05_voltage            3.2817                                 V    
cell_06_voltage            3.2827                                 V    
cell_07_voltage            3.2822                                 V    
cell_08_voltage            3.2816                                 V    
cell_09_voltage            3.2854                                 V    
cell_10_voltage            3.2807                                 V    
cell_11_voltage            3.2804                                 V    
cell_12_voltage            3.2833                                 V    
cell_13_voltage            3.2816                                 V    
cell_14_voltage            3.2824                                 V    
cell_15_voltage            3.2825                                 V    
cell_16_voltage            3.2804                                 V    
cell_01_resistance         0.19512                                Ω    
cell_01_resistance         0.19829                                Ω    
cell_01_resistance         0.20327                                Ω    
cell_01_resistance         0.20662                                Ω    
cell_01_resistance         0.21115                                Ω    
cell_01_resistance         0.21643                                Ω    
cell_01_resistance         0.22186                                Ω    
cell_01_resistance         0.22466                                Ω    
cell_01_resistance         0.23357                                Ω    
cell_01_resistance         0.23647                                Ω    
cell_01_resistance         0.23175                                Ω    
cell_01_resistance         0.22627                                Ω    
cell_01_resistance         0.22366                                Ω    
cell_01_resistance         0.21768                                Ω    
cell_01_resistance         0.21257                                Ω    
cell_01_resistance         0.20499                                Ω    
total_voltage              52.5097                                V    
average_cell_voltage       3.2819                                 V    
delta_cell_voltage         0.0051                                 V    
max_voltage_cell           9                                           
min_voltage_cell           11                                          
operation_status           Low_battery_voltage_balancing_stopped       
balancing_current          0.0                                    A    
temperature_1              22.739999771118164                     °C   
temperature_2              22.739999771118164                     °C   
cell_detection_failed      0x00 0x00 0x00                              
cell_overvoltage_failed    0x00 0x00 0x00                              
cell_undervoltage_failed   0x00 0x00 0x00                              
cell_polarity_error        0x00 0x00 0x00                              
excessive_line_resistance  0x00 0x00 0x00                              
overheating                0x0                                         
charging_fault             0x0                                         
discharge_fault            0x0                                         
read_write_error           0x0                                         

One thing - its not disconnecting properly from the device so before next request i have to power off/power on bluetooth device.

jblance, 2024-09-09:
Can you post a info run `-I`
Powermon is very wip atm

B4rtOl0, 2024-09-09:
@jblance  Sure. 
root@OrangePI:/# powermon -I -C /ble_neey.yaml 
2024-09-09 19:35:28,743:INFO:__init__:main@108: Power Device Monitoring Utility, version: 1.0.8-dev, python version: 3.11.10
2024-09-09 19:35:28,744:INFO:__init__:main@120: Using config file: /ble_neey.yaml
2024-09-09 19:35:28,755:INFO:__init__:main@131: Config validation successful
2024-09-09 19:35:28,755:INFO:__init__:main@147: config: {'device': {'name': 'BLE_Advertising', 'id': 'GW-24S4EB', 'model': 'SmartShunt 500A/50mV', 'manufacturer': 'Victron', 'port': {'type': 'ble', 'mac': '00:E4:4C:21:DB:9C', 'protocol': 'NEEY'}}, 'commands': [{'command': 'cell_info', 'outputs': [{'type': 'screen', 'format': {'type': 'table', 'draw_lines': False}}]}, {'command': 'get_settings', 'outputs': [{'type': 'screen', 'format': {'type': 'table', 'draw_lines': False}}]}, {'command': 'device_info', 'outputs': [{'type': 'screen', 'format': {'type': 'table', 'draw_lines': False}}]}], 'loop': 'once', 'debuglevel': 20}
2024-09-09 19:35:28,755:INFO:__init__:main@151: MqttBroker DISABLED
2024-09-09 19:35:28,780:INFO:abstractprotocol:check_definitions_count@121: Using protocol:b'NEEY' with 4 commands (device_info, cell_info, factory_defaults, get_settings)
2024-09-09 19:35:28,780:INFO:__init__:main@159: Adding command, config: {'command': 'cell_info', 'outputs': [{'type': 'screen', 'format': {'type': 'table', 'draw_lines': False}}]}
2024-09-09 19:35:28,783:INFO:__init__:main@159: Adding command, config: {'command': 'get_settings', 'outputs': [{'type': 'screen', 'format': {'type': 'table', 'draw_lines': False}}]}
2024-09-09 19:35:28,783:INFO:__init__:main@159: Adding command, config: {'command': 'device_info', 'outputs': [{'type': 'screen', 'format': {'type': 'table', 'draw_lines': False}}]}
2024-09-09 19:35:28,783:INFO:__init__:main@161: Device: BLE_Advertising, self.device_info.device_id='GW-24S4EB', self.device_info.model='SmartShunt 500A/50mV', self.device_info.manufacturer='Victron', port: BlePort: self.mac='00:E4:4C:21:DB:9C', protocol:NEEY protocol handler for NEEY balanceer, self.client=None, self.error_message=None, mqtt_broker: MqttBroker DISABLED, commands:[<powermon.commands.command.Command object at 0xffff8ae45c90>, <powermon.commands.command.Command object at 0xffff8ae46310>, <powermon.commands.command.Command object at 0xffff8ae46750>]
2024-09-09 19:35:28,783:INFO:__init__:main@165: Daemon DISABLED
2024-09-09 19:35:28,784:INFO:apicoordinator:from_config@49: No api definition in config
2024-09-09 19:35:28,784:INFO:__init__:main@171: ApiCoordinator DISABLED
2024-09-09 19:35:28,785:INFO:device:initialize@149: initializing device
2024-09-09 19:35:28,785:INFO:device:run@192: Running command: Command: self.code='cell_info' self.full_command=None, self.command_type='basic',             [_outs='outputs.Screen: outputs the results to the screen as per the formatter supplied'], last_run='Not yet run', next_run='09 Sep 2024 19:35:28', trigger: every 60 loops togo: 0, CommandDefinition: self.code='cell_info', self.description='information about the cells', self.result_type=<ResultType.CONSTRUCT: 8>, reading_definition count: 66, self.command_code='02', self.command_type=<CommandType.SERIAL_READ_UNTIL_DONE: 'serial_read_util_done'> self.override=None self.template=None
2024-09-09 19:35:28,785:INFO:bleport:connect@113: bleport connecting. mac:00:E4:4C:21:DB:9C
2024-09-09 19:35:29,969:INFO:bleport:connect@119: got bledevice: 00:E4:4C:21:DB:9C: GW-24S4EB
2024-09-09 19:35:29,973:INFO:bleport:connect@122: got bleclient: BleakClient, 00:E4:4C:21:DB:9C
2024-09-09 19:35:29,981:WARNING:version:check_bluez_version@57: Could not determine BlueZ version, bluetoothctl not available, assuming 5.51+
2024-09-09 19:35:31,218:INFO:bleport:connect@125: bleclient connected
2024-09-09 19:35:31,252:INFO:neey:get_full_command@377: Using protocol b'NEEY' with 4 commands
2024-09-09 19:35:31,785:INFO:result:decode_responses@105: result.response passed to decode: [('start_flag', b'U\xaa'), ('module_address', b'\x11'), ('function', b'\x01'), ('command', 2), ('length', 300), ('frame_counter', 125), ('cell_01_voltage', 3.2408323287963867), ('cell_02_voltage', 3.2423293590545654), ('cell_03_voltage', 3.2415809631347656), ('cell_04_voltage', 3.2421047687530518), ('cell_05_voltage', 3.2421796321868896), ('cell_06_voltage', 3.243377447128296), ('cell_07_voltage', 3.2425537109375), ('cell_08_voltage', 3.2423293590545654), ('cell_09_voltage', 3.245248556137085), ('cell_10_voltage', 3.2414309978485107), ('cell_11_voltage', 3.2410566806793213), ('cell_12_voltage', 3.2443504333496094), ('cell_13_voltage', 3.2422544956207275), ('cell_14_voltage', 3.2430031299591064), ('cell_15_voltage', 3.2435269355773926), ('cell_16_voltage', 3.239859104156494), ('cell_01_resistance', 0.19512179493904114), ('cell_02_resistance', 0.1982932686805725), ('cell_03_resistance', 0.20327050983905792), ('cell_04_resistance', 0.2066194862127304), ('cell_05_resistance', 0.21115070581436157), ('cell_06_resistance', 0.21643008291721344), ('cell_07_resistance', 0.2218562215566635), ('cell_08_resistance', 0.22466187179088593), ('cell_09_resistance', 0.2335664927959442), ('cell_10_resistance', 0.23647283017635345), ('cell_11_resistance', 0.231746643781662), ('cell_12_resistance', 0.22627060115337372), ('cell_13_resistance', 0.22366324067115784), ('cell_14_resistance', 0.21768471598625183), ('cell_15_resistance', 0.21257445216178894), ('cell_16_resistance', 0.2049923837184906), ('total_voltage', 51.878021240234375), ('average_c [... cut ...]
hw_version        HW-3.2.0        
sw_version        ZH-1.2.4        
protocol_version  V1.0.0          
production_date   20221201        
power_on_count    9               
total_runtime     13655785   sec  
2024-09-09 19:35:32,182:INFO:device:finalize@153: finalizing device
2024-09-09 19:35:32,182:INFO:bleport:disconnect@143: ble port disconnecting, BleakClient, 00:E4:4C:21:DB:9C True

Attached log as well. Let me know if you need debug or anything more.
[neey.txt](https://github.com/user-attachments/files/16935638/neey.txt)

B4rtOl0, 2024-09-09:
@jblance 
My config:

# BLE advertising config
device:
  name: BLE_Advertising
  id: GW-24S4EB
  model: SmartShunt 500A/50mV
  manufacturer: Victron
  port:
    type: ble
    mac: 00:E4:4C:21:DB:9C
    protocol: NEEY

commands:
- command: cell_info
  outputs:
  - type: screen
    format: 
      type: table
      draw_lines: False
- command: get_settings
  outputs:
  - type: screen
    format: 
      type: table
      draw_lines: False
- command: device_info
  outputs:
  - type: screen
    format: 
      type: table
      draw_lines: False
loop: once

jblance, 2024-09-09:
> 2024-09-09 19:35:29,981:WARNING:version:check_bluez_version@57: Could not determine BlueZ version, bluetoothctl not available, assuming 5.51+

Can you try `bluetoothctl` on the command line as port the result
Add can you check that `bluez` and `bluez-utils` package are installed and what version they are

B4rtOl0, 2024-09-09:
@jblance 
Yeah as i said it is running in chroot environment but bluetooth is on the host:
root@OrangePI:~# bluetoothctl --version
bluetoothctl: 5.60

bluez                                 5.60-0ubuntu2.2

jblance, 2024-09-09:
I updated the logging on the bleport, can you get the latest source from github and post a `-I` run of both a success run and a 'bluetooth restart needed' run
cheers

B4rtOl0, 2024-09-10:
Done, attached two files.
[neey-ok.txt](https://github.com/user-attachments/files/16942131/neey-ok.txt)
[neey-notok.txt](https://github.com/user-attachments/files/16942132/neey-notok.txt)

B4rtOl0, 2024-09-10:
It is in chroot environment so couldnt use bluetoothctl. Ill try to fix it.

B4rtOl0, 2024-09-10:
2024-09-10 08:54:00,195:INFO:bleport:disconnect@152: bluetoothctlcommunicate - outs: b'Waiting to connect to bluetoothd...\r\x1b[22P\x1b[0;94m[bluetooth]\x1b[0m# \r\x1b[0;94m[GW-24S4EB]\x1b[0m# \r\x1b[KAgent registered\n\x1b[0;94m[GW-24S4EB]\x1b[0m# \r\x1b[K[\x01\x1b[0;93m\x02CHG\x01\x1b[0m\x02] Controller 04:42:1A:57:2A:50 Pairable: yes\n\x1b[0;94m[GW-24S4EB]\x1b[0m# disconnect 00:E4:4C:21:DB:9C\nAttempting to disconnect from 00:E4:4C:21:DB:9C\n\x1b[0;94m[GW-24S4EB]\x1b[0m# \r\x1b[K', errs: None

It is working fine now after installing bluez in chroot environment. I didnt know and havent noticed you are using bluetoothctl to manage connection.

B4rtOl0, 2024-09-10:
It's all good, thank you very much !
One suggestion - cells resistance descriptions are broken in your code (each cell is cell_01). Not a problem for me but you may want to change it.
All the rest works awesome.

B4rtOl0, 2024-09-10:
@jblance  If you need me to test anything just tag me anytime and forgive me please for making this thread in wrong project :((((

jblance, 2024-10-10:
resistance descriptions should be fixed in git code now

closed 2024-10-10 by jblance