jblance/mpp-solar #512: Axpert Inverex Aerox 5.2KW not able to get all values

abdulrehman1937, 2024-09-05:
Hi, I have an Inverex (a local brand that uses Voltronic boards). I can fetch almost all data using the PI30 protocol. But I am having issues fetching energy-related data like pv_generation_today, and pv_generation_month. 
`mpp-solar -p /dev/hidraw0 -P PI30MAX --getDeviceId` `PI30:022:MKS2-5200`.
This data is shown on the inverter screen. Do you have any idea how can I get these values? I have tried different commands that I found in other issues but none worked.

jblance, 2024-09-05:
what commands have you tried? 
can you post the command run and output (add `-I` for any commands that give a result other than NAK)

abdulrehman1937, 2024-09-06:
[image]
[image]
I have attached screenshot of commands with `-I` flag. 
I have tried many commands found in other different issues in repo. None of them worked.

jblance, 2024-09-08:
> <https://github.com/user-attachments/assets/9a1dc924-ff33-4b9f-96f3-a58e3000826f> Screenshot.2024-09-06.at.7.01.17.PM.png

Have you tried the QEx commands - ie QED20240901 etc?

abdulrehman1937, 2024-09-09:
Yes tried not working.
[image]

jblance, 2024-09-09:
> Yes tried not working.
> Screenshot.2024-09-09.at.7.57.15.PM.png (view on web)

Can you try `-c QLT`

If that also doesn't work it looks like it is another variant of pi30 and I
don't know what the commands are

jblance, 2024-09-09:
Having a quick google this looks like a custom device - might be worth
seeing if you can get the protocol documentation from the supplier?

jblance, 2024-10-10:
Can you use the products own software to see what commands are sent?