jblance/mpp-solar #515: mppsolar do not display correctly on JK-B1A8S20P

dearteno, 2024-10-02:
**JK-B1A8S20P on rasberypie5**
```mppsolar -p C8:47:80:18:F3:33 -n "JK-B1A8S20P" -P JK02 -c getCellData 
jkbms -p C8:47:80:18:F3:33 -n "JK-B1A8S20P" -P JK02 -c getCellData
##Result
battery_power        0              W
balance_current      0.0            A
battery_t1           0.0            °C
battery_t2           0.0            °C
mos_temp             0.0            °C
percent_remain       0              %
capacity_remain      15728.64       Ah
nominal_capacity     0.0            Ah
cycle_count          28357          
cycle_capacity       0.0            Ah```

jblance, 2024-10-02:
Add the `-I` flag and post the output

dearteno, 2024-10-02:
```
jkbms -I -p C8:47:80:18:F3:33 -n "JK-B1A8S20P" -P JK02 -c getCellData 
2024-10-02 14:55:49,863:INFO:__init__:main@216: Solar Device Command Utility, version: 0.16.39, python version: 3.11.2
2024-10-02 14:55:49,864:INFO:__init__:main@275: Daemon DISABLED
2024-10-02 14:55:49,864:INFO:__init__:main@379: Creating device "JK-B1A8S20P" (type: "jkbms") on port "C8:47:80:18:F3:33 (porttype=None)" using protocol "JK02"
2024-10-02 14:55:49,866:INFO:__init__:get_port@105: Using jkbleio for communications
2024-10-02 14:55:50,027:INFO:__init__:main@435: Looping 1 commands
2024-10-02 14:55:50,027:INFO:__init__:main@441: Getting results from device: jkbms device - name: JK-B1A8S20P, port: <mppsolar.inout.jkbleio.JkBleIO object at 0xf6b53990>, protocol: JK02 - JKBMS BLE 2 byte data communication protocol handler for command: getCellData, tag: getCellData, outputs: screen
2024-10-02 14:55:50,027:INFO:device:run_command@51: Running command getCellData
2024-10-02 14:55:50,027:INFO:jkabstractprotocol:get_full_command@74: Using protocol b'JK02' with 3 commands
2024-10-02 14:55:50,028:INFO:device:run_command@81: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2024-10-02 14:55:50,028:INFO:jkabstractprotocol:get_full_command@74: Using protocol b'JK02' with 3 commands
2024-10-02 14:55:50,028:INFO:jkbleio:send_and_receive@35: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2024-10-02 14:55:50,028:INFO:jkbleio:ble_connect@63: Attempting to connect to C8:47:80:18:F3:33
2024-10-02 14:55:50,893:INFO:jkbleio:ble_get_data@97: Connected to b''
2024-10-02 14:55:51,404:INFO:jkbleio:ble_get_data@113: Read characteristic: Characteristic <ffe1>, handle 12
2024-10-02 14:55:51,405:INFO:jkbleio:ble_get_data@121: Enable 0x0b handle {'rsp': ['wr']}
2024-10-02 14:55:51,405:INFO:jkbleio:ble_get_data@124: Enable read handle {'rsp': ['wr']}
2024-10-02 14:55:51,405:INFO:jkbleio:ble_get_data@128: Write getInfo to read handle {'rsp': ['wr']}
2024-10-02 14:55:57,655:INFO:jkbleio:ble_get_data@143: Write command to read handle
2024-10-02 14:55:57,655:INFO:jkbleio:ble_get_data@149: Grabbing 1 records (after inital response)
2024-10-02 14:56:01,230:INFO:jkbleio:ble_get_data@154: jkbleio: ble_get_dataa: Got 1 records
2024-10-02 14:56:01,230:INFO:jkbleio:ble_disconnect@80: Disconnecting BLE Device...
2024-10-02 14:56:01,234:INFO:protocol_helpers:uptime@82: Uptime result 188D2H46M9S
2024-10-02 14:56:01,234:INFO:device:run_command@116: Decoded response {'_command': 'getCellData', '_command_description': 'BLE Cell Data inquiry', 'raw_response': ['Uªë\x90\x02IÑ\x0fÒ\x0fÒ\x0fÒ\x0fÕ\x0fÒ\x0fÕ\x0f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x7f\x00\x00\x00Ó\x0f\x04\x00\x04\x036\x006\x006\x007\x006\x006\x006\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x01\x00\x00\x00\x00Ån\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00ñ\x00ø\x00\x00\x00\x00\x00\x00\x00\x00Qìm\x04\x00@~\x05\x00\x00\x00\x00\x00\x02º\x01\x00d\x00\x00\x00<"\x0e\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00ÿ\x00\x01\x00\x00\x00Ó\x03\x00\x00\x00\x00,\x0eC@\x00\x00\x00\x00\x13\x0b\x00\x00\x00\x01\x00\x01\x00\x06\x00\x00\\V\x8d\x00\x00\x00\x00\x00\x06\x01\x00\x00\x00\x00Ò\x03<"\x0e\x00\x03\x00\x00\x00\x80Q\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00þÿ\x7fÜ\x0f\x01\x00\x80\x03\x00\x00\x00û', ''], 'Header': ['55aaeb90', ''], 'Record_Type': ['02', ''], 'Record_Counter': [73, ''], 'Voltage_Cell01': [4.049, 'V'], 'Voltage_Cell02': [4.05, 'V'], 'Voltage_Cell03': [4.05, 'V'], 'Voltage_Cell04': [4.05, 'V'], 'Voltage_Cell05': [4.053, 'V'], 'Voltage_Cell06': [4.05, 'V'], 'Voltage_Cell07': [4.053, 'V'], 'Voltage_Cell08': [0.0, 'V'], 'Voltage_Cell09': [0.0, 'V'], 'Voltage_Cell10': [0.0, 'V'], 'Voltage_Cell11': [0.0, 'V'], 'Voltage_Cell12': [0.0, 'V'], 'Voltage_Cell13': [0.0, 'V'], 'Voltage_Cell14': [0.0, 'V'], 'Voltage_Cell15': [0.0, 'V'], 'Voltage_Cell16': [0.0, 'V'], 'Voltage_Cell17': [0.0, 'V'], 'Voltage_Cell18': [0.0, 'V'], 'Voltage_Cell19': [0.0, 'V'], 'Voltage_Cell20': [0.0, 'V'], 'Voltage_Cell21': [0.0, 'V'], 'Voltage_Cell22': [0.0, 'V'], 'Voltage_Cell23': [0.0, 'V'], 'Voltage_Cell24': [0.0, 'V'], 'Average_Cell_Voltage': [0.0, 'V'], 'Delta_Cell_Voltage': [0.0, 'V'], 'Current_Balancer': [0.0, ''], 'Resistance_Cell01': [0.0, 'Ohm'], 'Resistance_Cell02': [0.0, 'Ohm'], 'Resistance_Cell03': [0.0, 'Ohm'], 'Resistance_Cell04': [0.127, 'Ohm'], 'Resistance_Cell05': [0.0, 'Ohm'], 'Resistance_Cell06': [4.051, 'Ohm'], 'Resistance_Cell07': [0.004, 'Ohm'], 'Resistance_Cell08': [0.772, 'Ohm'], 'Resistance_Cell09': [0.054, 'Ohm'], 'Resistance_Cell10': [0.054, 'Ohm'], 'Resistance_Cell11': [0.054, 'Ohm'], 'Resistance_Cell12': [0.055, 'Ohm'], 'Resistance_Cell13': [0.054, 'Ohm'], 'Resistance_Cell14': [0.054, 'Ohm'], 'Resistance_Cell15': [0.054, 'Ohm'], 'Resistance_Cell16': [0.0, 'Ohm'], 'Resistance_Cell17': [0.0, 'Ohm'], 'Resistance_Cell18': [0.0, 'Ohm'], 'Resistance_Cell19': [0.0, 'Ohm'], 'Resistance_Cell20': [0.0, 'Ohm'], 'Resistance_Cell21': [0.0, 'Ohm'], 'Resistance_Cell22': [0.0, 'Ohm'], 'Resistance_Cell23': [0.0,  [... cut ...]
battery_t1           0.0            °C
battery_t2           0.0            °C
mos_temp             0.0            °C
percent_remain       0              %
capacity_remain      17170.432      Ah
nominal_capacity     0.0            Ah
cycle_count          28357          
cycle_capacity       0.0            Ah
time                 188D2H46M9S    
current_charge       25.6           
current_discharge    0.0            
--------------------------------------------------------------------------------
```

jblance, 2024-10-02:
> 2024-10-02 ***@***.***: Solar Device Command Utility, version: 0.16.39, python version: 3.11.2
> 2024-10-02 ***@***.***: Daemon DISABLED
> 2024-10-02 ***@***.***: Creating device "JK-B1A8S20P" (type: "jkbms") on port "C8:47:80:18:F3:33 (porttype=None)" using protocol "JK02"
> 2024-10-02 ***@***.***: Using jkbleio for communications
> 2024-10-02 ***@***.***: Looping 1 commands
> 2024-10-02 ***@***.***: Getting results from device: jkbms device - name: JK-B1A8S20P, port: <mppsolar.inout.jkbleio.JkBleIO object at 0xf6b53990>, protocol: JK02 - JKBMS BLE 2 byte data communication protocol handler for command: getCellData, tag: getCellData, outputs: screen
> 2024-10-02 ***@***.***: Running command getCellData
> 2024-10-02 ***@***.***: Using protocol b'JK02' with 3 commands
> 2024-10-02 ***@***.***: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
> 2024-10-02 ***@***.***: Attempting to connect to C8:47:80:18:F3:33
> 2024-10-02 ***@***.***: Connected to b''
> 2024-10-02 ***@***.***: Read characteristic: Characteristic <ffe1>, handle 12
> 2024-10-02 ***@***.***: Enable 0x0b handle {'rsp': ['wr']}
> 2024-10-02 ***@***.***: Enable read handle {'rsp': ['wr']}
> 2024-10-02 ***@***.***: Write getInfo to read handle {'rsp': ['wr']}
> 2024-10-02 ***@***.***: Write command to read handle
> 2024-10-02 ***@***.***: Grabbing 1 records (after inital response)
> 2024-10-02 ***@***.***: jkbleio: ble_get_dataa: Got 1 records
> 2024-10-02 ***@***.***: Disconnecting BLE Device...
> 2024-10-02 ***@***.***: Uptime result 188D2H46M9S
> [... cut ...] l_Capacity': [0.0, 'Ah'], 'Cycle_Count': [28357, ''], 'Cycle_Capacity': [0.0, 'Ah'], 'Time': ['188D2H46M9S', ''], 'Current_Charge': [25.6, ''], 'Current_Discharge': [0.0, '']}
> 2024-10-02 ***@***.***: attempting to create output processor: screen
> 2024-10-02 ***@***.***: Using output processor: screen

Try some of the other protocols, eg jk02_32

dearteno, 2024-10-02:
i think it's done
```
 mppsolar -p C8:47:80:18:F3:33 -n "JK-B1A8S20P" -P JK02_32 -c getCellData 
Command: getCellData - BLE Cell Data inquiry
--------------------------------------------------------------------------------
Parameter            Value          Unit
header               55aaeb90       
record_type          02             
record_counter       113            
voltage_cell01       4.049          V
voltage_cell02       4.051          V
voltage_cell03       4.05           V
voltage_cell04       4.05           V
voltage_cell05       4.051          V
voltage_cell06       4.053          V
voltage_cell07       4.053          V
voltage_cell08       0.0            V
voltage_cell09       0.0            V
voltage_cell10       0.0            V
voltage_cell11       0.0            V
voltage_cell12       0.0            V
voltage_cell13       0.0            V
voltage_cell14       0.0            V
voltage_cell15       0.0            V
voltage_cell16       0.0            V
voltage_cell17       0.0            V
voltage_cell18       0.0            V
voltage_cell19       0.0            V
voltage_cell20       0.0            V
voltage_cell21       0.0            V
voltage_cell22       0.0            V
voltage_cell23       0.0            V
voltage_cell24       0.0            V
voltage_cell25       0.0            V
voltage_cell26       0.0            V
voltage_cell27       0.0            V
voltage_cell28       0.0            V
voltage_cell29       0.0            V
voltage_cell30       0.0            V
voltage_cell31       0.0            V
voltage_cell32       0.0            V
average_cell_voltage 4.051          V
delta_cell_voltage   0.004          V
resistance_cell01    0.054          Ohm
resistance_cell02    0.054          Ohm
resistance_cell03    0.054          Ohm
resistance_cell04    0.055          Ohm
resistance_cell05    0.054          Ohm
resistance_cell06    0.054          Ohm
resistance_cell07    0.054          Ohm
resistance_cell08    0.0            Ohm
resistance_cell09    0.0            Ohm
resistance_cell10    0.0            Ohm
resistance_cell11    0.0            Ohm
resistance_cell12    0.0            Ohm
resistance_cell13    0.0            Ohm
resistance_cell14    0.0            Ohm
resistance_cell15    0.0            Ohm
resistance_cell16    0.0            Ohm
resistance_cell17    0.0            Ohm
resistance_cell18    0.0            Ohm
resistance_cell19    0.0            Ohm
resistance_cell20    0.0            Ohm
resistance_cell21    0.0            Ohm
resistance_cell22    0.0            Ohm
resistance_cell23    0.0            Ohm
resistance_cell24    0.0            Ohm
resistance_cell25    0.0            Ohm
resistance_cell26    0.0            Ohm
resistance_cell27    0.0            Ohm
resistance_cell28    0.0            Ohm
resistance_cell29    0.0            Ohm
resistance_cell30    0.0            Ohm
resistance_cell31    0.0            Ohm
resistance_cell32    0.0            Ohm
mos_temp             26.3           °C
battery_voltage      28.357         V
battery_power        0.0            W
battery_current      0.0            A
battery_t1           24.2           °C
battery_t2           24.8           °C
balance_current      0.0            A
percent_remain       81             %
capacity_remain      290.274        Ah
nominal_capacity     360.0          Ah
cycle_count          0              
cycle_capacity       113.164        Ah
time                 10D18H2M58S   
```

closed 2024-10-07 by jblance