jblance/mpp-solar #516: Fragmented cellVoltage serial response with Daly BMS

klvbdmh, 2024-10-07:
Occasionally, reported cellVoltage values are outside of the norm. The app and the UART monitor show correct values all of the time.

# Setup
I connected my Daly BMS (16S 48V 250A) to Raspberry Pi 4B through the UART to USB cable.

```
mpp-solar --version
Solar Device Command Utility, version: 0.16.39, python version: 3.11.2
```

# Input
`mpp-solar -n DalyBMS -p /dev/ttyUSB0 -b 9600 -P DALY --porttype=dalyserial -c cellVoltages -D`

# Output (debug)

```
2024-10-07 07:50:31,025:INFO:__init__:main@216: Solar Device Command Utility, version: 0.16.39, python version: 3.11.2
2024-10-07 07:50:31,025:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-10-07 07:50:31,025:DEBUG:__init__:main@251: MqttBroker name: localhost, port: 1883, user: None
2024-10-07 07:50:31,025:DEBUG:__init__:main@253: udp port 5555
2024-10-07 07:50:31,025:DEBUG:__init__:main@255: Using Postgres None
2024-10-07 07:50:31,026:DEBUG:__init__:main@258: Using Mongo None with mppsolar
2024-10-07 07:50:31,027:INFO:__init__:main@275: Daemon DISABLED
2024-10-07 07:50:31,027:INFO:__init__:main@379: Creating device "DalyBMS" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=dalyserial)" using protocol "DALY"
2024-10-07 07:50:31,030:DEBUG:__init__:main@383: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2024-10-07 07:50:31,030:DEBUG:device:__init__@34: __init__ args ()
2024-10-07 07:50:31,030:DEBUG:device:__init__@35: __init__ kwargs {'name': 'DalyBMS', 'port': '/dev/ttyUSB0', 'protocol': 'DALY', 'baud': 9600, 'porttype': 'dalyserial', 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7f8d379b10>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway', 'prom_output_dir': '/var/lib/node_exporter'}
2024-10-07 07:50:31,031:INFO:__init__:get_port@81: Port overide - using port 'dalyserial'
2024-10-07 07:50:31,031:DEBUG:__init__:get_port_type@57: port matches daly
2024-10-07 07:50:31,031:INFO:__init__:get_port@117: Using dalyserialio for communications
2024-10-07 07:50:31,038:DEBUG:__init__:get_protocol@13: Protocol DALY
2024-10-07 07:50:31,327:DEBUG:device:__init__@39: __init__ name DalyBMS, port <mppsolar.inout.dalyserialio.DalySerialIO object at 0x7f8d37b0d0>, protocol Daly BMS protocol handler
2024-10-07 07:50:31,327:DEBUG:__init__:main@430: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f8d379fd0>, 'cellVoltages', 'cellVoltages', 'screen', None, None)]
2024-10-07 07:50:31,327:INFO:__init__:main@435: Looping 1 commands
2024-10-07 07:50:31,328:INFO:__init__:main@441: Getting results from device: mppsolar device - name: DalyBMS, port: <mppsolar.inout.dalyserialio.DalySerialIO object at 0x7f8d37b0d0>, protocol: Daly BMS protocol handler for command: cellVoltages, tag: cellVoltages, outputs: screen
2024-10-07 07:50:31,328:INFO:device:run_command@51: Running command cellVoltages
2024-10-07 07:50:31,328:INFO:daly:get_full_command@212: Using protocol b'DALY' with 7 commands
2024-10-07 07:50:31,328:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'cellVoltages'
2024-10-07 07:50:31,328:DEBUG:abstractprotocol:get_command_defn@73: Found command cellVoltages in protocol b'DALY'
2024-10-07 07:50:31,328:DEBUG:daly:get_full_command@231: full command: b'\xa5\x80\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc2\n'
2024-10-07 07:50:31,328:INFO:device:run_command@81: full command b'\xa5\x80\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc2\n' for command cellVoltages
2024-10-07 07:50:31,328:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'cellVoltages'
2024-10-07 07:50:31,328:DEBUG:abstractprotocol:get_command_defn@73: Found command cellVoltages in protocol b'DALY'
2024-10-07 07:50:31,328:DEBUG:dalyserialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 9600
2024-10-07 07:50:31,335:DEBUG:dalyserialio:send_and_receive@22: Executing command via dalyserialio...
2024-10-07 07:50:31,836:DEBUG:dalyserialio:send_and_receive@32: bytes waiting 92
2024-10-07 07:50:32,336:DEBUG:dalyserialio:send_and_receive@32: bytes waiting 0
2024-10-07 07:50:32,337:DEBUG:dalyserialio:send_and_receive@38: serial response was: b'{\xa5\x01\x95\x08\x06\x0c\xce\x0c\xd0\x0c\xd0\xa0{\xa5\x01\x95\x08\x01\x0c\xd1\x0c\xd0\x0c\xce\xa0w\xa5\x01\x95\x08\x02\x0c\xd0\x0c\xcf\x0c\xd0\xa0x\xa5\x01\x95\x08\x03\x0c\xcf\x0c\xd0\x0c\xcf\xa0x\xa5\x01\x95\x08\x04\x0c\xd0\x0c\xcf\x0c\xd0\xa0z\xa5\x01\x95\x08\x05\x0c\xcf\x0c\xd0\x0c\xcf\xa0z\xa5\x01\x95\x08\x06\x0c\xcf\x0c\xd0\x0c\xcf\xa0{'
2024-10-07 07:50:32,340:DEBUG:device:run_command@98: Send and Receive Response b'{\xa5\x01\x95\x08\x06\x0c\xce\x0c\xd0\x0c\xd0\xa0{\xa5\x01\x95\x08\x01\x0c\xd1\x0c\xd0\x0c\xce\xa0w\xa5\x01\x95\x08\x02\x0c\xd0\x0c\xcf\x0c\xd0\xa0x\xa5\x01\x95\x08\x03\x0c\xcf\x0c\xd0\x0c\xcf\xa0x\xa5\x01\x95\x08\x04\x0c\xd0\x0c\xcf\x0c\xd0\xa0z\xa5\x01\x95\x08\x05\x0c\xcf\x0c\xd0\x0c\xcf\xa0z\xa5\x01\x95\x08\x06\x0c\xcf\x0c\xd0\x0c\xcf\xa0{'
2024-10-07 07:50:32,341:DEBUG:abstractprotocol:decode@238: response passed to decode: b'{\xa5\x01\x95\x08\x06\x0c\xce\x0c\xd0\x0c\xd0\xa0{\xa5\x01\x95\x08\x01\x0c\xd1\x0c\xd0\x0c\xce\xa0w\xa5\x01\x95\x08\x02\x0c\xd0\x0c\xcf\x0c\xd0\xa0x\xa5\x01\x95\x08\x03\x0c\xcf\x0c\xd0\x0c\xcf\xa0x\xa5\x01\x95\x08\x04\x0c\xd0\x0c\xcf\x0c\xd0\xa0z\xa5\x01\x95\x08\x05\x0c\xcf\x0c\xd0\x0c\xcf\xa0z\xa5\x01\x95\x08\x06\x0c\xcf\x0c\xd0\x0c\xcf\xa0{'
2024-10-07 07:50:32,341:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'cellVoltages'
2024-10-07 07:50:32,341:DEBUG:abstractprotocol:get_command_defn@73: Found command cellVoltages in protocol b'DALY'
2024-10-07 07:50:32,341:DEBUG:daly:check_response_valid@245: checking validity of b'{\xa5\x01\x95\x08\x06\x0c\xce\x0c\xd0\x0c\xd0\xa0{\xa5\x01\x95\x08\x01\x0c\xd1\x0c\xd0\x0c\xce\xa0w\xa5\x01\x95\x08\x02\x0c\xd0\x0c\xcf\x0c\xd0\xa0x\xa5\x01\x95\x08\x03\x0c\xcf\x0c\xd0\x0c\xcf\xa0x\xa5\x01\x95\x08\x04\x0c\xd0\x0c\xcf\x0c\xd0\xa0z\xa5\x01\x95\x08\x05\x0c\xcf\x0c\xd0\x0c\xcf\xa0z\xa5\x01\x95\x08\x06\x0c\xcf\x0c\xd0\x0c\xcf\xa0{'
2024-10-07 07:50:32,342:INFO:daly:check_response_valid@251: is multiframe response - assuming ok for now
2024-10-07 07:50:32,342:DEBUG:abstractprotocol:decode@283: Processing response of type MULTIFRAME-POSITIONAL
2024-10-07 07:50:32,342:INFO:daly:get_responses@292: Multi frame response with 8 frames
2024-10-07 07:50:32,343:DEBUG:abstractprotocol:decode@287: trimmed and split responses: [[b'{', b'\xa5', b'\x01', b'\x95', b'\x08', b'\x06\x0c', b'\xce\x0c', b'\xd0\x0c', b'\xd0', b'\xa0'], [b'{', b'\xa5', b'\x01', b'\x95', b'\x08', b'\x01\x0c', b'\xd1\x0c', b'\xd0\x0c', b'\xce', b'\xa0'], [b'w', b'\xa5', b'\x01', b'\x95', b'\x08', b'\x02\x0c', b'\xd0\x0c', b'\xcf\x0c', b'\xd0', b'\xa0'], [b'x', b'\xa5', b'\x01', b'\x95', b'\x08', b'\x03\x0c', b'\xcf\x0c', b'\xd0\x0c', b'\xcf', b'\xa0'], [b'x', b'\xa5', b'\x01', b'\x95', b'\x08', b'\x04\x0c', b'\xd0\x0c', b'\xcf\x0c', b'\xd0', b'\xa0'], [b'z', b'\xa5', b'\x01', b'\x95', b'\x08', b'\x05\x0c', b'\xcf\x0c', b'\xd0\x0c', b'\xcf', b'\xa0'], [b'z', b'\xa5', b'\x01', b'\x95', b'\x08', b'\x06\x0c', b'\xcf\x0c', b'\xd0\x0c', b'\xcf', b'\xa0'], [b'{', b'', b'', b'', b'', b'', b'', b'', b'', b'']]
2024-10-07 07:50:32,343:DEBUG:abstractprotocol:decode@411: Processing MULTIFRAME-POSITIONAL type responses
2024-10-07 07:50:32,344:DEBUG:abstractprotocol:decode@415: got 8 frames
2024-10-07 07:50:32,344:DEBUG:abstractprotocol:decode@496: Processing POSITIONAL type responses
2024-10-07 07:50:32,344:DEBUG:abstractprotocol:decode@523: Got defn ['discard', 1, 'start flag', '']
2024-10-07 07:50:32,345:DEBUG:abstractprotocol:process_response@118: Processing data_type: discard for data_name: start flag, raw_value b'{'
2024-10-07 07:50:32,345:DEBUG:abstractprotocol:process_response@124: Discarding start flag:b'{'
2024-10-07 07:50:32,345:DEBUG:abstractprotocol:decode@496: Processing POSITIONAL type responses
2024-10-07 07:50:32,345:DEBUG:abstractprotocol:decode@523: Got defn ['discard', 1, 'module address', '']
2024-10-07 07:50:32,346:DEBUG:abstractprotocol:process_response@118: Processing data_type: discard for data_name: module address, raw_value b'\xa5'
2024-10-07 07:50:32,346:DEBUG:abstractprotocol:process_response@124: Discarding module address:b'\xa5'
2024-10-07 07:50:32,346:DEBUG:abstractprotocol:decode@496: Processing POSITIONAL type responses
2024-10-07 07:50:32,347:DEBUG:abstractprotocol:decode@523: Got defn ['discard', 1, 'command id', '']
[... cut ...]
\xa5\x01\x96\x08\x017\x00\x00\x00\x00\x000\xac
\xa5\x01\x94\x08\x10\x01\x00\x00\x00\x00\x0fT\xb6
```

Here we have more fragmented outputs. There are also chunks from different `command_ids` (`\x93`, `\x96`, `\x94`).

I observed that the correct readings have exactly 78 bytes (6 \* 13). Anything else results in incorrect readings.

# Things I've tried

- running the command with `--porttype=daly` makes many responses incomplete
- `baud` rate other than `9600` doesn't return anything at all
- `P=DALY40` doesn't seem to have any effect

jblance, 2024-10-15:
I've got this working in powermon code, I'll see if I can backport to mppsolar (but i might not be able to easily)

```python
 ➜ /workspaces/powermon (master) $ powermon -C tests/config/ble_daly.yaml -I
2024-10-15 21:23:14,305:INFO:runner:main@108: Power Device Monitoring Utility, version: 1.0.10-dev, python version: 3.11.7
2024-10-15 21:23:14,306:INFO:runner:main@120: Using config file: tests/config/ble_daly.yaml
2024-10-15 21:23:14,308:INFO:runner:main@131: Config validation successful
2024-10-15 21:23:14,309:INFO:runner:main@147: config: {'device': {'name': 'BLE_Advertising', 'serial_number': 'SmartShunt Bat2', 'model': 'SmartShunt 500A/50mV', 'manufacturer': 'Victron', 'port': {'type': 'test', 'protocol': 'DALY', 'response_number': 3}}, 'commands': [{'command': 'cell_voltages', 'outputs': [{'type': 'screen', 'format': {'type': 'table', 'draw_lines': True}}]}], 'loop': 'once', 'debuglevel': 20}
2024-10-15 21:23:14,309:INFO:runner:main@151: MqttBroker DISABLED
2024-10-15 21:23:14,311:INFO:abstractprotocol:check_definitions_count@123: Using protocol:b'DALY' with 6 commands (SOC, cell_voltage_range, temperatures, mosfet, status, cell_voltages)
2024-10-15 21:23:14,311:INFO:runner:main@174: Adding command, config: {'command': 'cell_voltages', 'outputs': [{'type': 'screen', 'format': {'type': 'table', 'draw_lines': True}}]}
2024-10-15 21:23:14,312:INFO:runner:main@176: Device: BLE_Advertising, self.device_info.serial_number='SmartShunt Bat2', self.device_info.model='SmartShunt 500A/50mV', self.device_info.manufacturer='Victron', port: Test port, mqtt_broker: MqttBroker DISABLED, commands:[<powermon.commands.command.Command object at 0x7403c413b250>]
2024-10-15 21:23:14,312:INFO:runner:main@180: Daemon DISABLED
2024-10-15 21:23:14,312:INFO:apicoordinator:from_config@49: No api definition in config
2024-10-15 21:23:14,312:INFO:runner:main@186: ApiCoordinator DISABLED
2024-10-15 21:23:14,312:INFO:device:initialize@153: initializing device
2024-10-15 21:23:14,312:INFO:device:run@196: Running command: Command: self.code='cell_voltages' self.full_command=None, self.command_type='basic',             [_outs='outputs.Screen: outputs the results to the screen as per the formatter supplied'], last_run='Not yet run', next_run='15 Oct 2024 21:23:14', trigger: every 60 loops togo: 0, CommandDefinition: self.code='cell_voltages', self.description='cell_voltages', self.result_type=<ResultType.CONSTRUCT: 8>, reading_definition count: 30, self.command_code='95', self.command_type=<CommandType.SERIAL_READ_UNTIL_DONE: 'serial_read_util_done'>, self.command_data=None self.override=None self.template=None
2024-10-15 21:23:14,313:INFO:daly:get_full_command@297: Using protocol b'DALY' with 6 commands
2024-10-15 21:23:14,313:INFO:testport:send_and_receive@67: Selected response number 3
2024-10-15 21:23:14,313:INFO:result:decode_responses@105: result.response passed to decode: [('cell_16_voltage', 3278), ('cell_17_voltage', 3280), ('cell_18_voltage', 3280), ('cell_01_voltage', 3281), ('cell_02_voltage', 3280), ('cell_03_voltage', 3278), ('cell_04_voltage', 3280), ('cell_05_voltage', 3279), ('cell_06_voltage', 3280), ('cell_07_voltage', 3279), ('cell_08_voltage', 3280), ('cell_09_voltage', 3279), ('cell_10_voltage', 3280), ('cell_11_voltage', 3279), ('cell_12_voltage', 3280), ('cell_13_voltage', 3279), ('cell_14_voltage', 3280), ('cell_15_voltage', 3279), ('cell_16_voltage', 3279), ('cell_17_voltage', 3280), ('cell_18_voltage', 3279)], result_type ResultType.CONSTRUCT
2024-10-15 21:23:14,314:INFO:device:run@200: Got result: Result: self.command.command_definition.description='cell_voltages': self.is_valid=True, self.error=False - self.error_messages=[], self.raw_response=b'{\xa5\x01\x95\x08\x06\x0c\xce\x0c\xd0\x0c\xd0\xa0{\xa5\x01\x95\x08\x01\x0c\xd1\x0c\xd0\x0c\xce\xa0w\xa5\x01\x95\x08\x02\x0c\xd0\x0c\xcf\x0c\xd0\xa0x\xa5\x01\x95\x08\x03\x0c\xcf\x0c\xd0\x0c\xcf\xa0x\xa5\x01\x95\x08\x04\x0c\xd0\x0c\xcf\x0c\xd0\xa0z\xa5\x01\x95\x08\x05\x0c\xcf\x0c\xd0\x0c\xcf\xa0z\xa5\x01\x95\x08\x06\x0c\xcf\x0c\xd0\x0c\xcf\xa0{', Reading: self.data_name='cell_16_voltage', self.data_value=3.278, self.data_unit='V', self.is_valid=True,Reading: self.data_name='cell_17_voltage', self.data_value=3.28, self.data_unit='V', self.is_valid=True,Reading: self.data_name='cell_18_voltage', self.data_value=3.28, self.data_unit='V', self.is_valid=True,Reading: self.data_name='cell_01_voltage', self.data_value=3.281, self.data_unit='V', self.is_valid=True,Reading: self.data_name='cell_02_voltage', self.data_value=3.28, self.data_unit='V', self.is_valid=True,Reading: self.data_name='cell_03_voltage', self.data_value=3.278, self.data_unit='V', self.is_valid=True,Reading: self.data_name='cell_04_voltage', self.data_value=3.28, self.data_unit='V', self.is_valid=True,Reading: self.data_name='cell_05_voltage', self.data_value=3.279, self.data_unit='V', self.is_valid=True,Reading: self.data_name='cell_06_voltage', self.data_value=3.28, self.data_unit='V', self.is_valid=True,Reading: self.data_name='cell_07_voltage', self.data_value=3.279, self.data_unit='V', self.is_valid=True,Reading: self.data_name='cell_08_voltage', self.data_value=3.28, self.data_unit='V', self.is_valid=True,Reading: self.data_name='cell_09_voltage', self.data_value=3.279, self.data_unit='V', self.is_valid=True,Reading: self.data_name='cell_10_voltage', self.data_value=3.28, self.data_unit='V', self.is_valid=True,Reading: self.data_name='cell_11_voltage', self.data_value=3.279, self.data_unit='V [... cut ...]
║ cell_10_voltage │ 3.28  │ V            ║
║ cell_11_voltage │ 3.279 │ V            ║
║ cell_12_voltage │ 3.28  │ V            ║
║ cell_13_voltage │ 3.279 │ V            ║
║ cell_14_voltage │ 3.28  │ V            ║
║ cell_15_voltage │ 3.279 │ V            ║
║ cell_16_voltage │ 3.279 │ V            ║
║ cell_17_voltage │ 3.28  │ V            ║
║ cell_18_voltage │ 3.279 │ V            ║
╚═════════════════╧═══════╧══════════════╝
2024-10-15 21:23:14,314:INFO:device:finalize@157: finalizing device
```

klvbdmh, 2024-10-16:
When I have some free time over the weekend, I'll check out powermon. I like the table formatter.

Offtopic, but I see you're using construct for parsing. That's awesome, I used it too when I was troubleshooting my issue (and on some other projects).

klvbdmh, 2024-10-20:
I ran the powermon overnight and reported voltage values were within the norm. It works well so far. Thank you.

I had one hiccup with the abnormal values elsewhere (soc, current), but I don't have any debug log. I'll see if I can reproduce it - anyway, that's for another repo.

klvbdmh, 2024-10-20:
I'm not sure exactly what caused it. I noticed the powermon was sending mqtt data (confirmed with MQTT Explorer), but Home Assistant wasn't updating the sensors for a few minutes. I restarted the services and there was a single update with abnormal values - but not all of them. Interestingly, individual cell voltages, capacity_ah, and cycles were fine.

It hasn't happened again so far, so it might be tricky to catch it again with debug. Maybe I can edit my install files so it writes a detailed log if SOC is over 100%.