jblance/mpp-solar #517: Installation error

ogry, 2024-10-09:
After installing mppsolar in Windows10 with `pip install mppsolar` I would try to install the ble version to test funcionality

`python -m pip install "mppsolar[ble]"
Requirement already satisfied: mppsolar[ble] in c:\users\ogryd\appdata\local\programs\python\python313\lib\site-packages (0.16.39)
Collecting bluepy<2.0.0,>=1.3.0 (from mppsolar[ble])
  Using cached bluepy-1.3.0.tar.gz (217 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: paho-mqtt<2.0.0,>=1.6 in c:\users\ogryd\appdata\local\programs\python\python313\lib\site-packages (from mppsolar[ble]) (1.6.1)
Requirement already satisfied: pydantic>=1.8.0 in c:\users\ogryd\appdata\local\programs\python\python313\lib\site-packages (from mppsolar[ble]) (2.9.2)
Requirement already satisfied: pyserial>3 in c:\users\ogryd\appdata\local\programs\python\python313\lib\site-packages (from mppsolar[ble]) (3.5)
Requirement already satisfied: annotated-types>=0.6.0 in c:\users\ogryd\appdata\local\programs\python\python313\lib\site-packages (from pydantic>=1.8.0->mppsolar[ble]) (0.7.0)
Requirement already satisfied: pydantic-core==2.23.4 in c:\users\ogryd\appdata\local\programs\python\python313\lib\site-packages (from pydantic>=1.8.0->mppsolar[ble]) (2.23.4)
Requirement already satisfied: typing-extensions>=4.12.2 in c:\users\ogryd\appdata\local\programs\python\python313\lib\site-packages (from pydantic>=1.8.0->mppsolar[ble]) (4.12.2)
Building wheels for collected packages: bluepy
  Building wheel for bluepy (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for bluepy (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [21 lines of output]
      C:\Users\ogryd\AppData\Local\Temp\pip-build-env-71b2sf1b\overlay\Lib\site-packages\setuptools\dist.py:452: SetuptoolsDeprecationWarning: Invalid dash-separated options
      !!

              ********************************************************************************
              Usage of dash-separated 'description-file' will not be supported in future
              versions. Please use the underscore name 'description_file' instead.

              This deprecation is overdue, please update your project and remove deprecated
              calls to avoid build errors in the future.

              See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
              ********************************************************************************

      !!
        opt = self.warn_dash_deprecation(opt, section)
      running bdist_wheel
      running build
      running build_py
      Working dir is C:\Users\ogryd\AppData\Local\Temp\pip-install-hfxfhat5\bluepy_ff45354024384f68b3555fd9ee8c0103
      execute make -C ./bluepy clean
      error: [WinError 2] El sistema no puede encontrar el archivo especificado
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for bluepy
Failed to build bluepy
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (bluepy)
`

then, I uninstalled mppsolar[ble] and I tryied to install again the basic mppsolar, but now I cannot run the basic mppsolar:

`>mpp-solar -p COM5: -c QPI
You are missing dependencies in order to be able to use that output.
To install them, use that command:
    python -m pip install 'mppsolar[ble]'
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\ogryd\AppData\Local\Programs\Python\Python313\Scripts\mpp-solar.exe\__main__.py", line 7, in <module>
    sys.exit(main())
             ~~~~^^
  File "C:\Users\ogryd\AppData\Local\Programs\Python\Python313\Lib\site-packages\mppsolar\__init__.py", line 385, in main
    device = device_class(
        name=args.name,
    ...<9 lines>...
        prom_output_dir=prom_output_dir,
    )
  File "C:\Users\ogryd\AppData\Local\Programs\Python\Python313\Lib\site-packages\mppsolar\devices\mppsolar.py", line 7, in __init__
    super().__init__(*args, **kwargs)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "C:\Users\ogryd\AppData\Local\Programs\Python\Python313\Lib\site-packages\mppsolar\devices\device.py", line 37, in __init__
    self._port = get_port(**kwargs)
                 ~~~~~~~~^^^^^^^^^^
  File "C:\Users\ogryd\AppData\Local\Programs\Python\Python313\Lib\site-packages\mppsolar\inout\__init__.py", line 106, in get_port
    from mppsolar.inout.jkbleio import JkBleIO
  File "C:\Users\ogryd\AppData\Local\Programs\Python\Python313\Lib\site-packages\mppsolar\inout\jkbleio.py", line 13, in <module>
    from .jkbledelegate import jkBleDelegate
  File "C:\Users\ogryd\AppData\Local\Programs\Python\Python313\Lib\site-packages\mppsolar\inout\jkbledelegate.py", line 1, in <module>
    from bluepy import btle
ModuleNotFoundError: No module named 'bluepy'
`

jblance, 2024-10-09:
your port `COM5:` is identifying as a BLE port (due to the `:`)
either remove the semi-colon and/or add `--porttype=serial` (or whatever type is connected to COM5)

PS I have never tested on Windows, so have no idea if it will work or not - its really intended for linux systems

klvbdmh, 2024-10-10:
> My main computer is a Debain 12, but unfortunately the USB to serial adapter doesn't want to work on it.

Can you share more details? Is it the permissions error or something else?

ogry, 2024-10-11:
Really don't know, as I changed the adaptor and now I'm back over my debian.

BTW, the ANERN AN SCI MAX 10200 (a cheap clone of Voltronic)  doesn't play well with the PIP protocol, or even with the PIP30, which is what it sometimes responds to. Many requests are slow to respond, long response times, some remain unanswered for minutes... however, the RS232 port seems to actually behave like a MODBUS RTU, and reading the registers through it is turning out to be much quicker and simpler than it seemed.