jblance/mpp-solar #518: ademon not working with alpine and openrc

Tuxist, 2024-10-17:
any solution for this problem ?

jblance, 2024-10-20:
not really in mppsolar - the daemon is too hardcoded
I've abstracted the daemon code in powermon (but havent written an openrc module as yet)
what protocol are you using?

jblance, 2024-10-20:
allthough it looks like if you write a service file mppsolar would probably work (see https://github.com/OpenRC/openrc/blob/master/service-script-guide.md and 'not well behaved' daemons)

cordelster, 2025-06-07:
@Tuxist I'm working on this, though it's still in a slight flux as I also added the ability to run as a pyinstaller elf. The code in the repo is there now, though I will need direct feed back to me of anything that pops up that in particular to the openRC implementation.  I have "turned off the pyinstaller specific stuff that allows the program to separate from the pyinstaller bootstrap which pretty sure you will not use anyway until i have ironed out everything in the normal openRC implementation. There is one more PR for john that adds watchdog handling in openRC and it should be good to use. I run it in Alpiine myself and gentoo, and John opened to the door to be able to focus on this back in march.