jblance/mpp-solar #519: How to use mpp-solar on windows?

mabs239, 2024-11-09:
I want to run this utility on windows 
mpp-solar -p /dev/ttyUSB0 -P VED -c vedtext --porttype vserial -b 19200 
How do I change “/dev/ttyUSB0” to “COM26” in CMD? I have tried following other options without success.
USB-SERIAL CH340 (COM26)

USB\VID_1A86&PID_7523\5&18598057&0&1

mabs239, 2024-11-09:
I had some mistake executing the command it is successful with following command:
mpp-solar -p "COM26"  -c QPIRI --porttype vserial -b 2400

C:\Users\GhulamNabi>mpp-solar -p "COM26"  -c QPIRI --porttype vserial -b 2400
Command: QPIRI - Current Settings inquiry
--------------------------------------------------------------------------------
Parameter                    Value          Unit
ac_input_voltage             220.0          V
ac_input_current             5.4            A
ac_output_voltage            220.0          V
ac_output_frequency          50.0           Hz
ac_output_current            5.4            A
ac_output_apparent_power     1200           VA
ac_output_active_power       1200           W
battery_voltage              12.0           V
battery_recharge_voltage     12.0           V
battery_under_voltage        10.5           V
battery_bulk_charge_voltage  14.6           V
battery_float_charge_voltage 13.5           V
battery_type                 Flooded
max_ac_charging_current      10             A
max_charging_current         80             A
input_voltage_range          Appliance
output_source_priority       SBU first
charger_source_priority      Only solar charging permitted
max_parallel_units           1              units
machine_type                 Off Grid
topology                     transformerless
output_mode                  single machine output
battery_redischarge_voltage  13.8           V
pv_ok_condition              As long as one unit of inverters has connect PV, parallel system will consider PV OK
pv_power_balance             PV input max power will be the sum of the max charged power and loads power
--------------------------------------------------------------------------------

jblance, 2024-11-11:
ok - looks like it is working

closed 2024-11-11 by jblance