jblance/mpp-solar #520: Request to decode Q2 command

mabs239, 2024-11-09:
While the result for Q1 command is decoded properly, Q2 command output is not decoded. I know these are not documented but since Q1 has been done, so should Q2.

Here is the output of Q2 on my inverter:

C:\Users\GhulamNabi>mpp-solar -p "COM26"  -c Q2 --porttype vserial -b 2400
Command: Q2 - No description found
--------------------------------------------------------------------------------
Parameter Value          Unit
warning   No definition for command Q2 in protocol b'PI30'
response  (00 00 01 0072 0064 0069 0076 232.5 00.33 00.3 00 00 00 00 01 1000 04432 00 004432 12.2 00 000 0000 0000 0069 0072 0018 0073 0000n×
--------------------------------------------------------------------------------

jblance, 2024-11-09:
Do you know what the response decodes to?

mabs239, 2024-11-11:
I haven't found any documentation but seems like there is no new information that is not being shown by other query commands.

jblance, 2024-11-11:
if you can determine what each field is, i can add a decode - it helps for
completeness