jblance/mpp-solar #521: Problems connecting to JiKong BMS PB1A16S10P via bluetooth

iko3, 2024-12-14:
Hello, 
I'm trying to connect to my JiKong BMS PB1A16S10P via bluetooth from raspberry pi, but it doesn't seem to work.
I'm able to connect from the mobile app and read all the data there, but I would also like to have access to my BMS when I'm away.

Here is my output

`(jkbms-monitoring) pi@raspberrypi3:~ $ jkbms -p C8:XX:XX:XX:XX:XX -P JK02 -c getInfo
2024-12-14 20:28:51,632:WARNING:jkbleio:ble_connect@67: Cannot connect to mac C8:XX:XX:XX:XX:XX - exceeded 3 attempts
2024-12-14 20:28:51,632:ERROR:jkbleio:send_and_receive@46: Failed to connect to C8:XX:XX:XX:XX:XX
Command: getInfo - BLE Device Information inquiry
--------------------------------------------------------------------------------
Parameter      Value          Unit
validity_check Error: Response was empty
--------------------------------------------------------------------------------

`

Any idea what might be wrong?

jblance, 2024-12-15:
could be that the bluetooth stack has errored (often happens with the library Im using)
try a `sudo systemctl restart bluetooth`
also add debug flag to command `-D` and post the output

iko3, 2024-12-15:
Restarted bluetooth, same result...
Here is the output:

`(jkbms-monitoring) pi@raspberrypi3:~ $ jkbms -p C8:XXXXXXX -P JK02 -c getInfo -D
2024-12-15 23:50:32,933:INFO:__init__:main@216: Solar Device Command Utility, version: 0.16.39, python version: 3.11.2
2024-12-15 23:50:32,933:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-12-15 23:50:32,934:DEBUG:__init__:main@251: MqttBroker name: localhost, port: 1883, user: None
2024-12-15 23:50:32,934:DEBUG:__init__:main@253: udp port 5555
2024-12-15 23:50:32,934:DEBUG:__init__:main@255: Using Postgres None
2024-12-15 23:50:32,934:DEBUG:__init__:main@258: Using Mongo None with mppsolar
2024-12-15 23:50:32,937:INFO:__init__:main@275: Daemon DISABLED
2024-12-15 23:50:32,938:INFO:__init__:main@379: Creating device "unnamed" (type: "jkbms") on port "C8:XXXXXXXX (porttype=None)" using protocol "JK02"
2024-12-15 23:50:32,943:DEBUG:__init__:main@383: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2024-12-15 23:50:32,944:DEBUG:device:__init__@34: __init__ args ()
2024-12-15 23:50:32,944:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': 'C8:XXXXXXX', 'protocol': 'JK02', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7fabfbc410>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway', 'prom_output_dir': '/var/lib/node_exporter'}
2024-12-15 23:50:32,944:DEBUG:__init__:get_port_type@51: port matches jkble ':'
2024-12-15 23:50:32,944:INFO:__init__:get_port@105: Using jkbleio for communications
2024-12-15 23:50:32,998:DEBUG:__init__:get_protocol@13: Protocol JK02
2024-12-15 23:50:33,621:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.jkbleio.JkBleIO object at 0x7fabef3f90>, protocol JK02 - JKBMS BLE 2 byte data communication protocol handler
2024-12-15 23:50:33,622:DEBUG:__init__:main@430: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x7fabef3790>, 'getInfo', 'getInfo', 'screen', None, None)]
2024-12-15 23:50:33,622:INFO:__init__:main@435: Looping 1 commands
2024-12-15 23:50:33,623:INFO:__init__:main@441: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.jkbleio.JkBleIO object at 0x7fabef3f90>, protocol: JK02 - JKBMS BLE 2 byte data communication protocol handler for command: getInfo, tag: getInfo, outputs: screen
2024-12-15 23:50:33,623:INFO:device:run_command@51: Running command getInfo
2024-12-15 23:50:33,623:INFO:jkabstractprotocol:get_full_command@74: Using protocol b'JK02' with 3 commands
2024-12-15 23:50:33,623:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getInfo
2024-12-15 23:50:33,624:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getInfo'
2024-12-15 23:50:33,624:DEBUG:abstractprotocol:get_command_defn@73: Found command getInfo in protocol b'JK02'
2024-12-15 23:50:33,624:DEBUG:jkabstractprotocol:get_full_command@79: self._command = getInfo
2024-12-15 23:50:33,624:DEBUG:jkabstractprotocol:get_full_command@90: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-12-15 23:50:33,624:DEBUG:jkabstractprotocol:get_full_command@98: cmd with command code: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-12-15 23:50:33,625:DEBUG:jkabstractprotocol:get_full_command@100: cmd with crc: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11')
2024-12-15 23:50:33,625:INFO:device:run_command@81: full command bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11') for command getInfo
2024-12-15 23:50:33,625:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getInfo
2024-12-15 23:50:33,625:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getInfo'
2024-12-15 23:50:33,625:DEBUG:abstractprotocol:get_command_defn@73: Found command getInfo in protocol b'JK02'
2024-12-15 23:50:33,626:INFO:jkabstractprotocol:get_full_command@74: Using protocol b'JK02' with 3 commands
2024-12-15 23:50:33,626:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getInfo
2024-12-15 23:50:33,626:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getInfo'
2024-12-15 23:50:33,626:DEBUG:abstractprotocol:get_command_defn@73: Found command getInfo in protocol b'JK02'
2024-12-15 23:50:33,627:DEBUG:jkabstractprotocol:get_full_command@79: self._command = getInfo
2024-12-15 23:50:33,627:DEBUG:jkabstractprotocol:get_full_command@90: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-12-15 23:50:33,627:DEBUG:jkabstractprotocol:get_full_command@98: cmd with command code: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-12-15 23:50:33,627:DEBUG:jkabstractprotocol:get_full_command@100: cmd with crc: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11')
2024-12-15 23:50:33,627:INFO:jkbleio:send_and_receive@35: full command bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11') for command getInfo
2024-12-15 23:50:33,628:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getInfo
2024-12-15 23:50:33,628:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getInfo'
[... cut ...] keep_case': False}
2024-12-15 23:50:34,601:DEBUG:screen:output@68: displayData: {'validity_check': ['Error: Response was empty', '']}
Command: getInfo - BLE Device Information inquiry
--------------------------------------------------------------------------------
Parameter      Value          Unit
validity_check Error: Response was empty
--------------------------------------------------------------------------------

2024-12-15 23:50:34,601:DEBUG:__init__:main@476: Not daemon, so not looping
`

jblance, 2024-12-15:
still not connecting the device's mac.. is anything else connected at the same time?
try `sudo hcitool lescan`
or use `bluetoothctl` to see what the system is reporting for devices

iko3, 2024-12-16:
I can see my BMS in the list of available devices. Both `hcitool lescan` and `bluetoothctl scan` can see it

jblance, 2024-12-16:
can you try installing powermon and running
`powermon-cli --bleScan --details --advData --getChars --address <mac address>`
this uses a different BLE library (more modern) compared to mpp-solar and does a detailed query - so will show what the newer python lib can see

iko3, 2024-12-16:
could you help me finding it? I tried to google for `powermon` but can't find it

iko3, 2024-12-16:
getting this error when trying to run powermon-cli
```python
(jkbms-monitoring) pi@raspberrypi3:~ $ powermon-cli --bleScan --details --advData --getChars --address c8:47:80:1c:19:87
Traceback (most recent call last):
  File "/usr/local/bin/powermon-cli", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/powermon/cli.py", line 272, in main
    ble_scan(args)
  File "/home/pi/.local/lib/python3.11/site-packages/powermon/cli.py", line 71, in ble_scan
    asyncio.run(scan_function(args.get('advData')))
                              ^^^^^^^^
AttributeError: 'Namespace' object has no attribute 'get'
```

jblance, 2024-12-16:
hopefully the new version fixes that bug