jblance/mpp-solar #522: JK BMS PB1A16S20 no data from bluetooth

slickm0nty, 2024-12-30:
Not getting any results for a new JK BMS using jk04, jk02, or jkpb protocols. I've also tried the rs485 connection and neither is retrieving data.

mpp-soloar version:
`Solar Device Command Utility, version: 0.16.40-dev, python version: 3.12.7`

`mark@solar-pi:~$ jkbms -p C8:47:80:13:E7:69 -n 'JK_PB2A16S20P-01' -D
2024-12-29 20:59:10,920:INFO:__init__:main@216: Solar Device Command Utility, version: 0.16.40-dev, python version: 3.12.7
2024-12-29 20:59:10,921:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2024-12-29 20:59:10,921:DEBUG:__init__:main@251: MqttBroker name: localhost, port: 1883, user: None
2024-12-29 20:59:10,922:DEBUG:__init__:main@253: udp port 5555
2024-12-29 20:59:10,922:DEBUG:__init__:main@255: Using Postgres None
2024-12-29 20:59:10,922:DEBUG:__init__:main@258: Using Mongo None with mppsolar
2024-12-29 20:59:10,926:INFO:__init__:main@275: Daemon DISABLED
2024-12-29 20:59:10,927:INFO:__init__:main@379: Creating device "JK_PB2A16S20P-01" (type: "jkbms") on port "C8:47:80:13:E7:69 (porttype=None)" using protocol "JK04"
2024-12-29 20:59:10,934:DEBUG:__init__:main@383: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2024-12-29 20:59:10,934:DEBUG:device:__init__@34: __init__ args ()
2024-12-29 20:59:10,935:DEBUG:device:__init__@35: __init__ kwargs {'name': 'JK_PB2A16S20P-01', 'port': 'C8:47:80:13:E7:69', 'protocol': 'JK04', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0xffffb5ed3020>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway', 'prom_output_dir': '/var/lib/node_exporter'}
2024-12-29 20:59:10,935:DEBUG:__init__:get_port_type@52: port matches jkble ':'
2024-12-29 20:59:10,935:INFO:__init__:get_port@109: Using jkbleio for communications
2024-12-29 20:59:11,013:DEBUG:__init__:get_protocol@13: Protocol JK04
2024-12-29 20:59:11,860:DEBUG:device:__init__@39: __init__ name JK_PB2A16S20P-01, port <mppsolar.inout.jkbleio.JkBleIO object at 0xffffb5ed36b0>, protocol JK04 - JKBMS BLE 4 byte data communication protocol handler
2024-12-29 20:59:11,861:DEBUG:__init__:main@430: Commands [(<mppsolar.devices.jkbms.jkbms object at 0xffffb5ed3680>, '', '', 'screen', None, None)]
2024-12-29 20:59:11,861:INFO:__init__:main@435: Looping 1 commands
2024-12-29 20:59:11,861:INFO:__init__:main@441: Getting results from device: jkbms device - name: JK_PB2A16S20P-01, port: <mppsolar.inout.jkbleio.JkBleIO object at 0xffffb5ed36b0>, protocol: JK04 - JKBMS BLE 4 byte data communication protocol handler for command: , tag: , outputs: screen
2024-12-29 20:59:11,862:INFO:device:run_command@51: Running command
2024-12-29 20:59:11,862:INFO:jkabstractprotocol:get_full_command@74: Using protocol b'JK04' with 2 commands
2024-12-29 20:59:11,862:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2024-12-29 20:59:11,862:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getCellData'
2024-12-29 20:59:11,862:DEBUG:abstractprotocol:get_command_defn@73: Found command getCellData in protocol b'JK04'
2024-12-29 20:59:11,863:DEBUG:jkabstractprotocol:get_full_command@79: self._command = getCellData
2024-12-29 20:59:11,863:DEBUG:jkabstractprotocol:get_full_command@90: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-12-29 20:59:11,863:DEBUG:jkabstractprotocol:get_full_command@98: cmd with command code: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-12-29 20:59:11,863:DEBUG:jkabstractprotocol:get_full_command@100: cmd with crc: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2024-12-29 20:59:11,863:INFO:device:run_command@81: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2024-12-29 20:59:11,864:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2024-12-29 20:59:11,864:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getCellData'
2024-12-29 20:59:11,864:DEBUG:abstractprotocol:get_command_defn@73: Found command getCellData in protocol b'JK04'
2024-12-29 20:59:11,864:INFO:jkabstractprotocol:get_full_command@74: Using protocol b'JK04' with 2 commands
2024-12-29 20:59:11,864:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2024-12-29 20:59:11,864:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getCellData'
2024-12-29 20:59:11,865:DEBUG:abstractprotocol:get_command_defn@73: Found command getCellData in protocol b'JK04'
2024-12-29 20:59:11,865:DEBUG:jkabstractprotocol:get_full_command@79: self._command = getCellData
2024-12-29 20:59:11,865:DEBUG:jkabstractprotocol:get_full_command@90: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-12-29 20:59:11,865:DEBUG:jkabstractprotocol:get_full_command@98: cmd with command code: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-12-29 20:59:11,865:DEBUG:jkabstractprotocol:get_full_command@100: cmd with crc: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2024-12-29 20:59:11,866:INFO:jkbleio:send_and_receive@35: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2024-12-29 20:59:11,866:DEBUG:jkabstractprotocol:get_command_defn@105: get_command_defn for: getCellData
2024-12-29 20:59:11,866:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getCellData'
2024-12-29 20:59:11,866:DEBUG:abstractprotocol:get_command_defn@73: Found command getCellData in protocol b'JK04'
2024-12-29 20:59:11,866:DEBUG:jkbleio:send_and_receive@39: expected record type 2 for command getCellData
2024-12-29 20:59:11,866:INFO:jkbleio:ble_connect@63: Attempting to connect to C8:47:80:13:E7:69
2024-12-29 20:59:12,125:DEBUG:jkbledelegate:handleNotification@27: From handle: 0x12 Got 4 bytes of data
2024-12-29 20:59:12,125:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'AT\r\n')
2024-12-29 20:59:12,125:DEBUG:jkabstractprotocol:wipe_to_start@150: SOR not found in record
2024-12-29 20:59:12,126:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'')
2024-12-29 20:59:12,126:DEBUG:jkbledelegate:handleNotification@38: Not expected type of record - wiping data bytearray(b'')
2024-12-29 20:59:12,126:DEBUG:jkabstractprotocol:is_record_complete@166: No SOR found in record looking for completeness
2024-12-29 20:59:12,142:DEBUG:jkbleio:ble_get_data@88: Command: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2024-12-29 20:59:12,292:INFO:jkbleio:ble_get_data@97: Connected to b''
2024-12-29 20:59:12,330:DEBUG:jkbledelegate:handleNotification@27: From handle: 0x12 Got 4 bytes of data
2024-12-29 20:59:12,331:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'AT\r\n')
2024-12-29 20:59:12,331:DEBUG:jkabstractprotocol:wipe_to_start@150: SOR not found in record
2024-12-29 20:59:12,331:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'')
2024-12-29 20:59:12,331:DEBUG:jkbledelegate:handleNotification@38: Not expected type of record - wiping data bytearray(b'')
2024-12-29 20:59:12,331:DEBUG:jkabstractprotocol:is_record_complete@166: No SOR found in record looking for completeness
2024-12-29 20:59:12,442:INFO:jkbleio:ble_get_data@113: Read characteristic: Characteristic <ffe1>, handle 12
2024-12-29 20:59:12,443:INFO:jkbleio:ble_get_data@121: Enable 0x0b handle {'rsp': ['wr']}
2024-12-29 20:59:12,444:INFO:jkbleio:ble_get_data@124: Enable read handle {'rsp': ['wr']}
2024-12-29 20:59:12,445:INFO:jkbleio:ble_get_data@128: Write getInfo to read handle {'rsp': ['wr']}
2024-12-29 20:59:12,518:DEBUG:jkbledelegate:handleNotification@27: From handle: 0x12 Got 128 bytes of data
[... 107 lines cut ...]
2024-12-29 20:59:15,496:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'AT\r\n')
2024-12-29 20:59:15,496:DEBUG:jkabstractprotocol:wipe_to_start@150: SOR not found in record
2024-12-29 20:59:15,497:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'')
[... cut ...]
2024-12-29 20:59:16,097:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'')
2024-12-29 20:59:16,097:DEBUG:jkbledelegate:handleNotification@38: Not expected type of record - wiping data bytearray(b'')
2024-12-29 20:59:16,097:DEBUG:jkabstractprotocol:is_record_complete@166: No SOR found in record looking for completeness
2024-12-29 20:59:16,302:DEBUG:jkbledelegate:handleNotification@27: From handle: 0x12 Got 4 bytes of data
2024-12-29 20:59:16,303:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'AT\r\n')`

TomSawyer8006, 2025-01-21:
Same Problem...

JK_PB2A16S20P
HW v15a
SW v15.32

No data, via Bluetooth
 -> continuous loop.
"SOR not found in record
.....
Pre wipe to start bytearray(b'AT\r\n')"
<repeat>

I also tried via 485 (in rs-485-1 and in rs-485-2 connector; with different dip switch changes = master / member)
-> empty response:
get_responses: responses [b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'']
and then crash when "Processing format string...."

It works with my older JK (v11 - using "jk02") but the latests v15 I was unable to make it work.

Any settings or ideas are welcome.

Thank you!

orf-kitzler-walli, 2025-02-09:
I can confirm this issue on my batteries as well.

GrinchUA, 2025-04-08:
I still can't get the data.
https://github.com/jblance/mpp-solar/issues/529

ofcaah, 2025-04-28:
I'm having the same issue with what seems to be JK-BD6A17S8P

````
$ jkbms -p C8:47:80:20:9A:91 -c getInfo -D
2025-04-28 03:09:42,726:INFO:__init__:main@216: Solar Device Command Utility, version: 0.16.55, python version: 3.11.2
2025-04-28 03:09:42,727:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2025-04-28 03:09:42,727:DEBUG:__init__:main@251: MqttBroker name: localhost, port: 1883, user: None
2025-04-28 03:09:42,728:DEBUG:__init__:main@253: udp port 5555
2025-04-28 03:09:42,728:DEBUG:__init__:main@255: Using Postgres None
2025-04-28 03:09:42,729:DEBUG:__init__:main@258: Using Mongo None with mppsolar
2025-04-28 03:09:42,734:INFO:__init__:main@275: Daemon DISABLED
2025-04-28 03:09:42,735:INFO:__init__:main@379: Creating device "unnamed" (type: "jkbms") on port "C8:47:80:20:9A:91 (porttype=None)" using protocol "JK04"
2025-04-28 03:09:42,742:DEBUG:__init__:main@383: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2025-04-28 03:09:42,743:DEBUG:device:__init__@34: __init__ args ()
2025-04-28 03:09:42,743:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': 'C8:47:80:20:9A:91', 'protocol': 'JK04', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0xf77dbb30>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway', 'prom_output_dir': '/var/lib/node_exporter'}
2025-04-28 03:09:42,743:DEBUG:__init__:get_port_type@52: port matches jkble ':'
2025-04-28 03:09:42,744:INFO:__init__:get_port@109: Using jkbleio for communications
2025-04-28 03:09:42,815:DEBUG:__init__:get_protocol@13: Protocol JK04
2025-04-28 03:09:43,518:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.jkbleio.JkBleIO object at 0xf6fd3ad0>, protocol JK04 - JKBMS BLE 4 byte data communication protocol handler
2025-04-28 03:09:43,519:DEBUG:__init__:main@430: Commands [(<mppsolar.devices.jkbms.jkbms object at 0xf6fd3ab0>, 'getInfo', 'getInfo', 'screen', None, None)]
2025-04-28 03:09:43,519:INFO:__init__:main@435: Looping 1 commands
2025-04-28 03:09:43,520:INFO:__init__:main@441: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.jkbleio.JkBleIO object at 0xf6fd3ad0>, protocol: JK04 - JKBMS BLE 4 byte data communication protocol handler for command: getInfo, tag: getInfo, outputs: screen
2025-04-28 03:09:43,520:INFO:device:run_command@51: Running command getInfo
2025-04-28 03:09:43,521:INFO:jkabstractprotocol:get_full_command@76: Using protocol b'JK04' with 2 commands
2025-04-28 03:09:43,521:DEBUG:jkabstractprotocol:get_command_defn@119: get_command_defn for: getInfo
2025-04-28 03:09:43,521:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getInfo'
2025-04-28 03:09:43,522:DEBUG:abstractprotocol:get_command_defn@73: Found command getInfo in protocol b'JK04'
2025-04-28 03:09:43,522:DEBUG:jkabstractprotocol:get_full_command@81: self._command = getInfo
2025-04-28 03:09:43,523:DEBUG:jkabstractprotocol:get_full_command@92: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2025-04-28 03:09:43,523:DEBUG:jkabstractprotocol:get_full_command@112: cmd with command code: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2025-04-28 03:09:43,523:DEBUG:jkabstractprotocol:get_full_command@114: cmd with crc: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11')
2025-04-28 03:09:43,524:INFO:device:run_command@81: full command bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11') for command getInfo
2025-04-28 03:09:43,524:DEBUG:jkabstractprotocol:get_command_defn@119: get_command_defn for: getInfo
2025-04-28 03:09:43,524:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getInfo'
2025-04-28 03:09:43,525:DEBUG:abstractprotocol:get_command_defn@73: Found command getInfo in protocol b'JK04'
2025-04-28 03:09:43,525:INFO:jkabstractprotocol:get_full_command@76: Using protocol b'JK04' with 2 commands
2025-04-28 03:09:43,525:DEBUG:jkabstractprotocol:get_command_defn@119: get_command_defn for: getInfo
2025-04-28 03:09:43,526:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getInfo'
2025-04-28 03:09:43,526:DEBUG:abstractprotocol:get_command_defn@73: Found command getInfo in protocol b'JK04'
2025-04-28 03:09:43,527:DEBUG:jkabstractprotocol:get_full_command@81: self._command = getInfo
2025-04-28 03:09:43,527:DEBUG:jkabstractprotocol:get_full_command@92: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2025-04-28 03:09:43,527:DEBUG:jkabstractprotocol:get_full_command@112: cmd with command code: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2025-04-28 03:09:43,528:DEBUG:jkabstractprotocol:get_full_command@114: cmd with crc: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11')
2025-04-28 03:09:43,528:INFO:jkbleio:send_and_receive@35: full command bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11') for command getInfo
2025-04-28 03:09:43,528:DEBUG:jkabstractprotocol:get_command_defn@119: get_command_defn for: getInfo
2025-04-28 03:09:43,529:DEBUG:abstractprotocol:get_command_defn@71: Processing command 'getInfo'
[... cut ...]
2025-04-28 03:10:50,506:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'AT\r\n')
2025-04-28 03:10:50,507:DEBUG:jkabstractprotocol:wipe_to_start@171: SOR not found in record
2025-04-28 03:10:50,507:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'')
2025-04-28 03:10:50,507:DEBUG:jkbledelegate:handleNotification@38: Not expected type of record - wiping data bytearray(b'')
2025-04-28 03:10:50,507:DEBUG:jkabstractprotocol:is_record_complete@187: No SOR found in record looking for completeness
````

last 6 lines loop forever in both cases.