jblance/mpp-solar #524: Axpert King 24V 3KVA

JiMcNulty, 2025-01-08:
Hi
Thanks for all the hard work you put into this repo!

Wanted to ask before I dive into this rabbit hole
Do you if axpert king will work with this?
If not - do you know what protocol it's using?

jblance, 2025-01-12:
I dont know - but best bet is to try to connect and query the deviceId - this will give some info that can help

JiMcNulty, 2025-02-14:
Hi,
I tried it, and it almost works but this command is failing for some reason 

`root@raspberrypi2:~# mpp-solar -p /dev/hidraw0 -c QPIWS -I
2025-02-14 17:05:54,180:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2025-02-14 17:05:54,181:INFO:__init__:main@339: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI30"
2025-02-14 17:05:54,192:INFO:device:set_port@142: Using hidrawio for communications
2025-02-14 17:05:54,289:INFO:__init__:main@395: Looping 1 commands
2025-02-14 17:05:54,290:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x76793df0>, protocol: <mppsolar.protocols.pi30.pi30 object at 0x767b1388> for command: QPIWS, tag: QPIWS, outputs: screen
2025-02-14 17:05:54,291:INFO:device:run_command@270: Running command QPIWS
2025-02-14 17:05:54,291:INFO:abstractprotocol:get_full_command@31: Using protocol b'PI30' with 36 commands
2025-02-14 17:05:54,292:INFO:device:run_command@296: full command b'QPIWS\xb4\xda\r' for command QPIWS
2025-02-14 17:05:55,710:INFO:abstractprotocol:decode@175: response passed to decode: b'(000001000000000000000000000000000000\xb2\xa7\r'
2025-02-14 17:05:55,712:INFO:abstractprotocol:decode@215: Processing response of type DEFAULT
2025-02-14 17:05:55,713:INFO:abstractprotocol:decode@226: Processing DEFAULT type responses
Traceback (most recent call last):
  File "/usr/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3.8/site-packages/mppsolar/__init__.py", line 408, in main
    results = _device.run_command(command=_command)
  File "/usr/lib/python3.8/site-packages/mppsolar/devices/device.py", line 333, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/usr/lib/python3.8/site-packages/mppsolar/protocols/abstractprotocol.py", line 284, in decode
    key = resp_format[2][j]
IndexError: list index out of range
`

Any ideas?

jblance, 2025-02-18:
the version is quite old (0.16.39 is latest and you're on 0.7.76) - suggest you update and try again

jblance, 2025-08-14:
> 2025-08-14 ***@***.***: Solar Device Command Utility, version: 0.16.55, python version: 3.12.9
> 2025-08-14 ***@***.***: Daemon DISABLED
> 2025-08-14 ***@***.***: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI30"
> 2025-08-14 ***@***.***: Using hidrawio for communications
> 2025-08-14 ***@***.***: No module found for protocol pi30
> 2025-08-14 ***@***.***: Looping 1 commands
> 2025-08-14 ***@***.***: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x75b22708>, protocol: None for command: QPIWS, tag: QPIWS, outputs: screen
> 2025-08-14 ***@***.***: Running command QPIWS
> 2025-08-14 ***@***.***: Attempted to run command with no protocol defined
> 2025-08-14 ***@***.***: attempting to create output processor: screen
> 2025-08-14 ***@***.***: Using output processor: screen
> Command: Unknown command - No description found
> --------------------------------------------------------------------------------
> Parameter Value          Unit
> error     Attempted to run command with no protocol defined
> --------------------------------------------------------------------------------

Weird bug there somewhere - can you do a full debug (`-D`) run and post the
output

JiMcNulty, 2025-08-14:
```

root@raspberrypi2:~# mpp-solar -p /dev/hidraw0 -c QPIWS -I -D
2025-08-14 21:37:14,587:INFO:__init__:main@216: Solar Device Command Utility, version: 0.16.55, python version: 3.12.9
2025-08-14 21:37:14,588:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2025-08-14 21:37:14,589:DEBUG:__init__:main@251: MqttBroker name: localhost, port: 1883, user: None
2025-08-14 21:37:14,590:DEBUG:__init__:main@253: udp port 5555
2025-08-14 21:37:14,591:DEBUG:__init__:main@255: Using Postgres None
2025-08-14 21:37:14,592:DEBUG:__init__:main@258: Using Mongo None with mppsolar
2025-08-14 21:37:14,601:INFO:__init__:main@275: Daemon DISABLED
2025-08-14 21:37:14,602:INFO:__init__:main@379: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI30"
2025-08-14 21:37:14,617:DEBUG:__init__:main@383: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2025-08-14 21:37:14,618:DEBUG:device:__init__@34: __init__ args ()
2025-08-14 21:37:14,619:DEBUG:device:__init__@35: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x75bd1630>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway', 'prom_output_dir': '/var/lib/node_exporter'}
2025-08-14 21:37:14,620:DEBUG:__init__:get_port_type@40: port matches hidraw
2025-08-14 21:37:14,621:INFO:__init__:get_port@97: Using hidrawio for communications
2025-08-14 21:37:14,631:DEBUG:__init__:get_protocol@13: Protocol PI30
2025-08-14 21:37:15,671:ERROR:__init__:get_protocol@21: No module found for protocol pi30
2025-08-14 21:37:15,674:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0x75bd1720>, protocol None
2025-08-14 21:37:15,675:DEBUG:__init__:main@430: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x75bd1738>, 'QPIWS', 'QPIWS', 'screen', None, None)]
2025-08-14 21:37:15,676:INFO:__init__:main@435: Looping 1 commands
2025-08-14 21:37:15,677:INFO:__init__:main@441: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x75bd1720>, protocol: None for command: QPIWS, tag: QPIWS, outputs: screen
2025-08-14 21:37:15,679:INFO:device:run_command@51: Running command QPIWS
2025-08-14 21:37:15,680:ERROR:device:run_command@54: Attempted to run command with no protocol defined
2025-08-14 21:37:15,681:DEBUG:__init__:main@443: results: {'ERROR': ['Attempted to run command with no protocol defined', '']}
2025-08-14 21:37:15,682:INFO:__init__:get_output@40: attempting to create output processor: screen
2025-08-14 21:37:15,689:DEBUG:screen:__init__@17: processor.screen __init__ args: (), kwargs: {}
2025-08-14 21:37:15,690:DEBUG:__init__:main@449: Using output filter: None
2025-08-14 21:37:15,690:INFO:screen:output@23: Using output processor: screen
2025-08-14 21:37:15,691:DEBUG:screen:output@24: kwargs {'data': {'ERROR': ['Attempted to run command with no protocol defined', '']}, 'tag': 'QPIWS', 'name': 'unnamed', 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x75bd1630>, 'udp_port': 5555, 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway', 'prom_output_dir': '/var/lib/node_exporter', 'mqtt_topic': None, 'filter': None, 'excl_filter': None, 'keep_case': False}
2025-08-14 21:37:15,692:DEBUG:screen:output@68: displayData: {'error': ['Attempted to run command with no protocol defined', '']}
Command: Unknown command - No description found
--------------------------------------------------------------------------------
Parameter Value          Unit
error     Attempted to run command with no protocol defined
--------------------------------------------------------------------------------

2025-08-14 21:37:15,694:DEBUG:__init__:main@476: Not daemon, so not looping

```

jblance, 2025-08-14:
> 2025-08-14 ***@***.***: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
> 2025-08-14 ***@***.***: MqttBroker name: localhost, port: 1883, user: None
> 2025-08-14 ***@***.***: udp port 5555
> 2025-08-14 ***@***.***: Using Postgres None
> 2025-08-14 ***@***.***: Using Mongo None with mppsolar
> 2025-08-14 ***@***.***: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
> 2025-08-14 ***@***.***: __init__ args ()
> 2025-08-14 ***@***.***: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x75bd1630>, 'udp_port': 5555, 'mongo_url': None, 'mongo_db': 'mppsolar', 'push_url': 'http://localhost:9091/metrics/job/pushgateway', 'prom_output_dir': '/var/lib/node_exporter'}
> 2025-08-14 ***@***.***: port matches hidraw
> 2025-08-14 ***@***.***: Protocol PI30
> 2025-08-14 ***@***.***: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0x75bd1720>, protocol None
> 2025-08-14 ***@***.***: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x75bd1738>, 'QPIWS', 'QPIWS', 'screen', None, None)]
> 2025-08-14 ***@***.***: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x75bd1720>, protocol: None for command: QPIWS, tag: QPIWS, outputs: screen
> 2025-08-14 ***@***.***: results: {'ERROR': ['Attempted to run command with no protocol defined', '']}
> 2025-08-14 ***@***.***: processor.screen __init__ args: (), kwargs: {}
> 2025-08-14 ***@***.***: Using output filter: None
> [... cut ...] ics/job/pushgateway', 'prom_output_dir': '/var/lib/node_exporter', 'mqtt_topic': None, 'filter': None, 'excl_filter': None, 'keep_case': False}
> 2025-08-14 ***@***.***: displayData: {'error': ['Attempted to run command with no protocol defined', '']}
> 2025-08-14 ***@***.***: Not daemon, so not looping

seems to not be finding the protocol definitions can you run
`mpp-solar -p /dev/hidraw0 -P`

JiMcNulty, 2025-08-14:
```

root@raspberrypi2:~# mpp-solar -p /dev/hidraw0 -P -D
2025-08-14 22:05:11,683:INFO:__init__:main@216: Solar Device Command Utility, version: 0.16.55, python version: 3.12.9
2025-08-14 22:05:11,684:INFO:__init__:get_output@40: attempting to create output processor: screen
2025-08-14 22:05:11,690:DEBUG:screen:__init__@17: processor.screen __init__ args: (), kwargs: {}
2025-08-14 22:05:12,765:INFO:__init__:list_protocols@48: Error in module abstractprotocol: No module named 'zoneinfo'
2025-08-14 22:05:12,797:INFO:__init__:list_protocols@48: Error in module daly: No module named 'zoneinfo'
2025-08-14 22:05:12,831:INFO:__init__:list_protocols@48: Error in module daly40: No module named 'zoneinfo'
2025-08-14 22:05:12,868:INFO:__init__:list_protocols@48: Error in module jk02: No module named 'zoneinfo'
2025-08-14 22:05:12,903:INFO:__init__:list_protocols@48: Error in module jk02_32: No module named 'zoneinfo'
2025-08-14 22:05:12,934:INFO:__init__:list_protocols@48: Error in module jk04: No module named 'zoneinfo'
2025-08-14 22:05:12,964:INFO:__init__:list_protocols@48: Error in module jk232: No module named 'zoneinfo'
2025-08-14 22:05:12,998:INFO:__init__:list_protocols@48: Error in module jk485: No module named 'zoneinfo'
2025-08-14 22:05:13,029:INFO:__init__:list_protocols@48: Error in module jkabstractprotocol: No module named 'zoneinfo'
2025-08-14 22:05:13,064:INFO:__init__:list_protocols@48: Error in module jkpb: No module named 'zoneinfo'
2025-08-14 22:05:13,093:INFO:__init__:list_protocols@48: Error in module jkserial: No module named 'zoneinfo'
2025-08-14 22:05:13,127:INFO:__init__:list_protocols@48: Error in module jkv11: No module named 'zoneinfo'
2025-08-14 22:05:13,161:INFO:__init__:list_protocols@48: Error in module pi16: No module named 'zoneinfo'
2025-08-14 22:05:13,192:INFO:__init__:list_protocols@48: Error in module pi17: No module named 'zoneinfo'
2025-08-14 22:05:13,225:INFO:__init__:list_protocols@48: Error in module pi17infini: No module named 'zoneinfo'
2025-08-14 22:05:13,258:INFO:__init__:list_protocols@48: Error in module pi17m058: No module named 'zoneinfo'
2025-08-14 22:05:13,289:INFO:__init__:list_protocols@48: Error in module pi18: No module named 'zoneinfo'
2025-08-14 22:05:13,324:INFO:__init__:list_protocols@48: Error in module pi18lvx: No module named 'zoneinfo'
2025-08-14 22:05:13,357:INFO:__init__:list_protocols@48: Error in module pi18sv: No module named 'zoneinfo'
2025-08-14 22:05:13,391:INFO:__init__:list_protocols@48: Error in module pi30: No module named 'zoneinfo'
2025-08-14 22:05:13,430:INFO:__init__:list_protocols@48: Error in module pi30m044: No module named 'zoneinfo'
2025-08-14 22:05:13,469:INFO:__init__:list_protocols@48: Error in module pi30m045: No module named 'zoneinfo'
2025-08-14 22:05:13,505:INFO:__init__:list_protocols@48: Error in module pi30max: No module named 'zoneinfo'
2025-08-14 22:05:13,542:INFO:__init__:list_protocols@48: Error in module pi30mst: No module named 'zoneinfo'
2025-08-14 22:05:13,574:INFO:__init__:list_protocols@48: Error in module pi30revo: No module named 'zoneinfo'
2025-08-14 22:05:13,608:INFO:__init__:list_protocols@48: Error in module pi41: No module named 'zoneinfo'
2025-08-14 22:05:13,617:INFO:__init__:list_protocols@52: Error in module protocol_helpers: module 'mppsolar.protocols.protocol_helpers' has no attribute 'protocol_helpers'
2025-08-14 22:05:13,647:INFO:__init__:list_protocols@48: Error in module ved: No module named 'zoneinfo'
2025-08-14 22:05:13,650:INFO:screen:output@23: Using output processor: screen
2025-08-14 22:05:13,651:DEBUG:screen:output@24: kwargs {'data': {'_command': 'protocols help', '_command_description': 'List available protocol modules'}}
2025-08-14 22:05:13,652:DEBUG:screen:output@68: displayData: {}
Command: protocols help - List available protocol modules
--------------------------------------------------------------------------------
Parameter Value          Unit
--------------------------------------------------------------------------------

```

jblance, 2025-08-14:
> 2025-08-14 ***@***.***: Error in module abstractprotocol: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module daly: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module daly40: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module jk02: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module jk02_32: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module jk04: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module jk232: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module jk485: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module jkabstractprotocol: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module jkpb: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module jkserial: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module jkv11: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module pi16: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module pi17: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module pi17infini: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module pi17m058: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module pi18: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module pi18lvx: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module pi18sv: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module pi30: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module pi30m044: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: Error in module pi30m045: No module named 'zoneinfo'
> [... cut ...]
> 2025-08-14 ***@***.***: Error in module protocol_helpers: module 'mppsolar.protocols.protocol_helpers' has no attribute 'protocol_helpers'
> 2025-08-14 ***@***.***: Error in module ved: No module named 'zoneinfo'
> 2025-08-14 ***@***.***: kwargs {'data': {'_command': 'protocols help', '_command_description': 'List available protocol modules'}}

that seems like the problem - ie the missing module 'zoneinfo'

jblance, 2025-08-14:
which should be part of the python install...
try starting python on the command line and importing zoneinfo to see what result you get
```
$ python
>>> import zoneinfo
>>> zoneinfo.__all__
['ZoneInfo', 'reset_tzpath', 'available_timezones', 'TZPATH', 'ZoneInfoNotFoundError', 'InvalidTZPathWarning']
```

JiMcNulty, 2025-08-14:
```
On Venus OS, anything outside Victron’s core stack is usually stripped out. That means:

zoneinfo isn’t missing because it’s broken — it was never compiled in.
You won’t be able to just apt install it, because Venus OS doesn’t have a package manager with Python stdlib extras.
Use pytz instead of zoneinfo if you just need time zones — pytz is pure Python and easier to drop into /data.
```

jblance, 2025-08-14:
oh - I'm guessing `datetime` uses zoneinfo
does Venus OS support datetime module? is there an alternative

jblance, 2025-08-14:
Thats weird, I'm not sure where `zoneinfo` is getting imported then - seems like it must be abstractprotocol?
can you try
`from mppsolar.protocols.abstractprotocol import AbstractProtocol`

jblance, 2025-08-14:
As an aside, what is your use case here?

JiMcNulty, 2025-08-14:
```
root@raspberrypi2:~# python
Python 3.12.9 (main, Feb  4 2025, 14:38:38) [GCC 13.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from mppsolar.protocols.abstractprotocol import AbstractProtocol
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.12/site-packages/mppsolar/protocols/abstractprotocol.py", line 7, in <module>
    from pydantic import BaseModel
  File "/usr/lib/python3.12/site-packages/pydantic/__init__.py", line 435, in __getattr__
    module = import_module(module_name, package=package)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/pydantic/main.py", line 36, in <module>
    from ._internal import (
  File "/usr/lib/python3.12/site-packages/pydantic/_internal/_model_construction.py", line 26, in <module>
    from ._generate_schema import GenerateSchema, InvalidSchemaError
  File "/usr/lib/python3.12/site-packages/pydantic/_internal/_generate_schema.py", line 41, in <module>
    from zoneinfo import ZoneInfo
ModuleNotFoundError: No module named 'zoneinfo'
>>>

```

jblance, 2025-08-14:
looks like you could have a local mppsolar tree and have the code use that (ie a mppsolar source tree)
tho the DTO stuff is largely unneeded, so I could try removing it at some point

jblance, 2025-08-14:
actually most of the pydantic has been removed - Ive updated abstractprotocol in v 0.16.56 - once that shows in pypi you should be able to get past the current error