jblance/mpp-solar #528: bookworm - docker issue: The requested image's platform.....

sabom2d, 2025-03-26:
Hi,

I'm trying to install mpp-solar as Docker image on a RPi 3 64bit, but get following error message when calling docker run:

The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested

Don't have a clue, what to do.
Any advice or solution?

regards

jblance, 2025-03-27:
It doesnt look like my docker build CI includes arm image support - I'll need to look into how to make this work

jblance, 2025-03-27:
I have a build with arm64 on docker hub now - dont know if the platforms will match or if a bunch of variations will be needed
Can you try pulling again and let me know the result