jblance/mpp-solar #547: Error: No module found for protocol

ahmar20, 2025-07-09:
Hi there, I am new to all this. I am trying to use mppsolar on my router running OpenWrt v24.10.1 kernel 6.6.86. I have installed mppsolar and related dependencies. However, I am getting an error as below when I try to run the mppsolar.

```
root@AP254:~# mpp-solar -p /dev/ttyUSB0 --getDeviceId
2025-07-09 23:44:23,213:ERROR:__init__:get_protocol@21: No module found for protocol pi30
2025-07-09 23:44:23,227:ERROR:device:run_command@54: Attempted to run command with no protocol defined
Command: Unknown command - No description found
--------------------------------------------------------------------------------
Parameter Value          Unit
error     Attempted to run command with no protocol defined

```

I have tries to search for the error but I could not find any information. Can someone help with this? Thanks.

jblance, 2025-07-10:
can you add the debug flag `-D` and post the output

dc1rjj, 2025-10-29:
Hi have the same problem:

pi@solaranzeige:~/mpp-solar $ jkbms -p /dev/ttyUSB1 -b 115200 -P JK485 -D
2025-10-29 09:09:38,747:INFO:__init__:main@214: Solar Device Command Utility, version: 0.15.53
2025-10-29 09:09:38,747:DEBUG:mqttbrokerc:__init__@29: mqttbroker config: {'name': 'localhost', 'port': 1883, 'user': None, 'pass': None}
2025-10-29 09:09:38,748:DEBUG:__init__:main@239: MqttBroker name: localhost, port: 1883, user: None
2025-10-29 09:09:38,749:DEBUG:__init__:main@241: udp port 5555
2025-10-29 09:09:38,749:DEBUG:__init__:main@243: Using Postgres None
2025-10-29 09:09:38,749:DEBUG:__init__:main@246: Using Mongo None with mppsolar
2025-10-29 09:09:38,749:INFO:__init__:main@348: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB1 (porttype=None)" using protocol "JK485"
2025-10-29 09:09:38,759:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2025-10-29 09:09:38,759:DEBUG:device:__init__@33: __init__ args ()
2025-10-29 09:09:38,760:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB1', 'protocol': 'JK485', 'baud': 115200, 'porttype': None, 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7602a390>, 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2025-10-29 09:09:38,761:DEBUG:__init__:get_port_type@73: port matches ttyusb
2025-10-29 09:09:38,762:INFO:__init__:get_port@121: Using serialio for communications
2025-10-29 09:09:38,789:DEBUG:__init__:get_protocol@13: Protocol JK485
2025-10-29 09:09:38,802:ERROR:__init__:get_protocol@21: No module found for protocol jk485
2025-10-29 09:09:38,804:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0x7602a490>, protocol None
2025-10-29 09:09:38,805:DEBUG:__init__:main@402: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x7602a510>, '', '', 'screen', None, None)]
2025-10-29 09:09:38,806:INFO:__init__:main@407: Looping 1 commands
2025-10-29 09:09:38,806:INFO:__init__:main@418: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x7602a490>, protocol: None for command: , tag: , outputs: screen
2025-10-29 09:09:38,807:INFO:device:run_command@50: Running command
2025-10-29 09:09:38,808:ERROR:device:run_command@53: Attempted to run command with no protocol defined
2025-10-29 09:09:38,808:DEBUG:__init__:main@421: results: {'ERROR': ['Attempted to run command with no protocol defined', '']}
2025-10-29 09:09:38,809:INFO:__init__:get_output@40: attempting to create output processor: screen
2025-10-29 09:09:38,817:DEBUG:screen:__init__@15: processor.screen __init__ kwargs {}
2025-10-29 09:09:38,817:DEBUG:__init__:main@427: Using output filter: None
2025-10-29 09:09:38,818:INFO:screen:output@21: Using output processor: screen
2025-10-29 09:09:38,819:DEBUG:screen:output@22: kwargs {'data': {'ERROR': ['Attempted to run command with no protocol defined', '']}, 'tag': '', 'name': 'unnamed', 'mqtt_broker': <mppsolar.libs.mqttbrokerc.MqttBroker object at 0x7602a390>, 'udp_port': '5555', 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'mqtt_topic': None, 'filter': None, 'excl_filter': None, 'keep_case': False}
2025-10-29 09:09:38,820:DEBUG:screen:output@74: displayData: {'error': ['Attempted to run command with no protocol defined', '']}
Command: Unknown command - No description found
--------------------------------------------------------------------------------
Parameter Value                 Unit
error     Attempted to run command with no protocol defined
2025-10-29 09:09:38,821:DEBUG:__init__:main@452: Not daemon, so not looping

any help?